subogero/omxd #26: Playlist in Loop

OMID-313, 2015-02-03:
Dear @subogero 

Thanks to your helpful suggestion here (https://github.com/popcornmix/omxplayer/issues/308), I installed omxd on my Raspberry Pi B+ which has NOOBS/Raspbian running on it.

My goal is to have several video files in a playlist, which play in loop.
Currently I have used the method from this link (http://stackoverflow.com/questions/16470795/bash-script-to-loop-through-all-videos-in-a-folder-on-raspbian-wheezy) to loop my videos in a folder.

But after installing omxd, I want to modify this method, so that besides playing the playlist automatically in loop, I could be able to jump to next/previous/specific video in the list, etc. whenever I want using my IR remote controller. (I'll be using the LIRC for this.)

Would you please help me how I should do this?
How should I tell omxplayer or omxd to start to play video files in a folder?

Thanks for your time and support.

OMID-313, 2015-02-03:
EDIT:

Is the following an optimum way to have a playlist in loop, while taking advantage of omxd's other features?

while (true) ; do
    omxd A <my_folder>
done

subogero, 2015-02-03:
No shell scripting is needed. Once you've installed omxd, the daemon is running in the background, and it loops your playlist. The omxd commands are simply for manipulating the list.

    omxd A <file>  # Appends file to the list
    omxd A <folder>  # Appends all files in the folder to the list
    omxd S all  # Prints playback status and playlist
    omxd n  # Jump to next track in playllist
    omxd N  # Jump to previous track in playlist
    omxd g 2  # Jump to 3rd item in playlist

OMID-313, 2015-02-03:
Thanks @subogero for your reply.
I tried these commands commands in the terminal (after log in), and they worked perfectly.

But there's a minor problem:

In order to play the videos automatically at startup, Previously I had created a executable script (for example "/home/pi/bin/my_script.sh") with the following contents:

    # My Script to run at start up
    
    while (true); do
        omxplayer my_video_1.mp4
        omxplayer my_video_2.mp4
        omxplayer my_video_3.mp4
    done

and I had added the following lines to the "rc.local" file (in /etc) :

    # Auto run our application
    /home/pi/bin/my_script.sh

These scripts would work perfectly at startup (after boot, before login). It would play the 3 videos in loop after boot.

Now with the omx installed, I decided to change them. So I removed that "while" loop and omxplayer commands, and modified the "my_script.sh" file as the following:

    # My Script to run at start up
    
    omxd A /home/pi/my_folder

But unfortunately, nothing happened. The videos didn't start to play, and the login screen appeared!
I changed it to:

    # My Script to run at start up
    
    while (true); do
        omxd A /home/pi/my_folder
    done

But unfortunately, again nothing happened. The screen remained black, without any videos. The only difference was that the login screen didn't come, because of the infinite while loop.

If omxplayer is enabled before login, so should omxd be.
What should I do to enable omxd before login?

subogero, 2015-02-03:
You have to know that the playlist is persistent, so you don't have to add the folder again after startup. All you have to do is starting playback. All you need in `/etc/rc.local` is this:

    omxd g 0

which will start playback at the first item.

OMID-313, 2015-02-03:
Thanks @subogero for your reply.

You mean, I change the contents of the "rc.local" file from
    # Auto run our application
    /home/pi/bin/my_script.sh
to 
    omxd g 0
??

What about "my_script.sh" file?
Where should I assign the playlist folder??

subogero, 2015-02-03:
Yes. Do you want to define a new playlist at each reboot? If not, `omxd g 0` is enough.

You can change the playlist manually whenever you want, and your changes will remain after reboot.

    omxd X  # delete entire playlist
    omxd A <otherfolder>  # add the files from another folder

Please read the README or try the `man omxd` command.

OMID-313, 2015-02-03:
Thanks @subogero for your helpful reply.

I changed the contents of the `rc.local` file to

    omxd X
    omxd A /home/pi/my_folder

and it worked successfully. Great.

But again, there are some minor problems:

1. When for example video_1 finishes, it switches automatically to video_2 with a gap of 1-2 seconds. This short gap is really perfect. But when I press the `next` button to jump to the next video, this gap is sometimes 7~8 sec. What should I do to shorten this gap between next/previous jumps?

2. Are there any options in omxd to do `mute` or `unmute` ?

3. In the past, when I used the `while` method in my_script file, The omxplayer would start playing files before the login screen. And the login screen would never appear. But now, after boot, firstly the login screen appears, and then after 3-4 seconds, the videos start to play. In addition, the login screen appears each time in the short gap between the videos. What should I do to remove this login screen? I want a black screen in the background, either in the beginning after boot, or in the gap between videos. What should I do?

Sorry to ask so many silly questions.
Ans thanks a billion for your time and support.

subogero, 2015-02-03:
1. I'll check the code, whether `omxd n` uses the preloaded next track or not.
2. Don't remember, what does README say? It definitely has volume up/down commands?
3. Never heard of `seek relative`. Is it a DBUS only control? `omxd` does not use DBUS...
4. Deleting and creating your playlist at reboot takes time. What's wrong with `omxd g 0`? It's much faster.

subogero, 2015-02-03:
Well spotted, `omxd n` destroys the preloaded omxplayer instance with the next track and starts a new one instead. I'll fix it to just use the preloaded one.

OMID-313, 2015-02-03:
Thanks @subogero for your support.

As for your answers to my questions:
1. I look forward to the fixes in `omxd n` , and probably `omxd N` . Thanks.
2. There are already volume up/down commands. But what I say is mute/unmute commands. I couldn't find them.
3. The `seek relative` (or simply `seek`) is available via DBUS. But is it limited to DBUS or not, I don't know. But you can find the command here (https://github.com/popcornmix/omxplayer/blob/master/KeyConfig.h), and here (https://raw.githubusercontent.com/popcornmix/omxplayer/master/dbuscontrol.sh).
4. Actually the problem is not with being fast or slow. As I described, I don't want to see the login screen. I want the background to be black, without login screen.

One more issue:
5. I just found out when using omxd, the dbuscontrol of the omxplayer is disabled. The dbuscontrol was really helpful for me. Is that possible to enable it?

Thanks again for your time and support.

OMID-313, 2015-02-03:
Dear @subogero ,

This post (https://github.com/popcornmix/omxplayer/issues/180) describes how to add mute/unmute function to omxplayer.

I would appreciate it if you add something similar to omxd.

subogero, 2015-02-03:
1. Already working on it
2. omxd uses the standard key bindings of omxplayer, see `omxplayer -k`. It does not have mute/unmute, that's the reason omxd does not have them either. It could be implemented, though, via converting the omxd command into a sequence of volume up/down omxplayer commands.
3. `seek relative` is DBUS only, it seems from the code.
4. The console with the login screen goes to sleep a few minutes after reboot
5. omxd does not disable omxplayer's DBUS. The problem might be that it always runs 2 instances, the 2nd paused at the beginning of the next track, and they might want to connect to the same DBUS name, which breaks. I don't see a solution without reintroducing huge gaps and login screens all the time between tracks.

subogero, 2015-02-03:
Do you realize that @popcornmix's solution for mute is not an interactive control, but a command line option? It means you can't mute a track during playback, you only can start a track muted.

If that's OK for you, omxd has something similar, see the `j` and `h` commands. Assume you only have a TV connected via HDMI, but no amplifier to the audio-jack of the Pi. If you say `omxd j`, the current track will be started again, but with its audio directed to the audio-jack. Result silence.

OMID-313, 2015-02-03:
@subogero 

Actually, it's not OK.
I want to have the mute/unmute option in my IR remote controller, so that the user can mute/unmute whenever he/she wants, not only at the beginning.

Maybe I write a script to run `volume_down` several times, so as to virtually do `mute`. But I don't know how to combine both `mute` and `unmute` functions in one button press.

subogero, 2015-02-03:
You have to store the mute state in your script if it's running continuously, or in a file if your scripts is started up upon every LIRC keypress. Then you can decide what to do. It's a STATE machine. :-)

OMID-313, 2015-02-03:
Thanks @subogero for your reply.

I realized that LIRC has an option for toggling functions with one button press. So the toggling problem is solved.

But as for the mute/unmute problem, I wrote the following executable script:

    #!/bin/bash
    # This is mute_script
    # 20 x 3 dB = 60 dB
    for i in {1..20}
    do
        omxd -
    done

And another similar script for unmute function with `moxd -` replaced with `omxd +` .
But the problem is that when I press mute/unmute, instead of decreasing/increasing the volume for 60 dB, it acts randomly!! Sometimes only 3 dB, sometimes 12, 15, 9, etc. Why?

subogero, 2015-02-03:
It's an omxplayer problem. If the +/- keyboard inputs come too fast from omxd, it ignores some of them. This should do the trick:

    for i in {1..20}; do sleep 0.3; omxd -; done

OMID-313, 2015-02-05:
Thanks @subogero for your reply.

I used the following script for the `mute` function:

    for i in {1..40}
    do
        sleep 0.05
        omxd -
    done

and it worked! Great. I also tried other numbers for `sleep` like 0.2, 0.1, 0.01, 0.001, but they were either too slow or too fast. I realized that 0.05 was an optimum number.

OMID-313, 2015-02-05:
One more question, @subogero :

`omxplayer` has many powerful options, listed in its readme file (https://github.com/popcornmix/omxplayer/blob/master/README.md). 
How can we use those options, while using `omxd`?

For example, I want to set the -- layer n option. Or maybe I want to specify the window properties by `--win 'x1 y1 x2 y2'`, or maybe `--no-osd`, . . .

How can we use these powerful options of `omxplayer`, while taking advantage of `omxd` ?

subogero, 2015-02-05:
Well, they are not available via omxd, unless you replace `/usr/bin/omxplayer` with a script that uses these options.

I'll think about a way to implement it though.

subogero, 2015-02-07:
See new issue https://github.com/subogero/omxd/issues/27

OMID-313, 2015-02-07:
Thanks @subogero for the new features.

What about the long gap between video when using `omxd n` or `omxd N` or `omxd g 3` ? Any news or updates?

subogero, 2015-02-07:
I've already fixed the long gap and just pushed the changes to GitHub. But I have not released the new version of the Debian package.

You can start using it if you clone this repo and compile from source.

    git clone <URL of this repo>
    cd omxd
    make
    sudo make install
    service omxd restart

OMID-313, 2015-02-10:
Thank you @subogero for the fixes.

So, now that I have omxd installed, in order to replace it with the newer version, can I use something like `sudo update omxd` ?

If not, do I have to remove `omxd` first !? How?

And then after removal, would the following code work?

    git clone <https://github.com/subogero/omxd>
    cd omxd
    make
    sudo make install
    service omxd restart

Thanks again.

subogero, 2015-02-10:
If you want to remove it first, it's actually

    sudo apt-get remove omxd

But it's not strictly necessary. Just the steps above. Make sure you use `sudo` with the restart command as well.

OMID-313, 2015-04-23:
Thanks for all your help and support @subogero .
I think this issue is solved, and it's now better be closed.

closed 2015-04-23 by OMID-313