subogero/omxd #27: Pass omxplayer options

subogero, 2015-02-07:
Add new command to pass arbitrary omxplayer options. Similar to `j` and `h` commands, restart playback of current track and remember options until further changes. Exampes:

    omxd O --orientation 270  # Play videos rotated by 270 deg from now on
    omxd O  # Remove additional omxplayer options

OMID-313, 2015-02-07:
Thanks @subogero for the new features.
Do we have to update or upgrade something, in order to use these features?

Does this `omxd O` include all `omxplayer` features?
That would be great if you also update the readme file for detailed information.

subogero, 2015-02-07:
No new features yet. This is just an issue that reminds me to implement them.

And yes, it will include all present and future features, as it will simply pass them on to omxplayer.

OMID-313, 2015-02-07:
OK.

I look forward to seeing these new features being fully implemented.

Thanks again for all your help and support @subogero .
Your time is highly appreciated.

subogero, 2015-02-10:
First I won't allow the options to contain any spaces, to make the implementation simple. For the same reason I submitted a patch to `omxplayer` that allows comma separators in the `--win` option:

    omxplayer --win 0,960,0,540

OMID-313, 2015-02-10:
Good to hear you're trying to implement these features.
Thanks a billion for your time @subogero .

subogero, 2015-02-10:
I actually need these options myself... :-)

subogero, 2015-02-13:
omxd 1.3 is in my Debian repo now, including the new `O` command.

closed 2015-02-13 by subogero

OMID-313, 2015-02-14:
Thanks a billion @subogero for developing this new helpful feature.

So now that I have both `omxd` and `omxplayer` installed on my Raspberry Pi B+ (with NOOS/Raspbian on it), what shall I do to update/upgrade/remove/reinstall them?

subogero, 2015-02-14:
Please see my last comment for installation instructions.

Please note that the omxplayer options are accumulated if you send multiple O commands. After the 2 examples below omxplayer will play rotated in a small window.

    omxd O --orientation 270 # rotated playback from now on
    omxd O --win '100 100 300 500' # Yes, spaces do work, so no omxplayer patches are needed

In the meantime omxplayer was updated with my patch, so if you update it, it will also allow comma separators for --win:

    omxd O --win 100,100,300,500

Clear the extra options from now:

    omxd O

You should try dbuscontrol yourself, omxd does not care about Dbus. But it might be broken due to 2 omxplayers running all the time.

OMID-313, 2015-02-14:
Thanks again @subogero for your helpful reply.

Just one more point:
What do you exactly mean by *"the omxplayer options are accumulated if you send multiple O commands."* ?
Do you mean that we cannot use them as instant controls when the video is playing?

subogero, 2015-02-14:
Accumulated means that if you send `O --orientation 270`, and then later `O --win 1,2,3,4`, then both options will be applied after the second.

Whenever you send new options the current track is *restarted* with them. These are not interactive controls, these are command line options that you can only use when starting omxplayer.

OMID-313, 2015-02-14:
OK, Got it.
Thanks a billion for your time :)

OMID-313, 2015-02-15:
Thanks @subogero for the new options and features.

I had previously installed `omxd` on my Raspberry Pi B+. Today, I installed the new version of `omxd` by the following commands:

    sudo apt-get update
    sudo apt-get install omxd

After rebooting the system, I just realized that some controls like `pause`, `volume_up`, `volume_down`, etc don't work. I tried several things, like changing TTY1 to TTY3 and back, adding `irexec -d` to `rc.local`, etc. but none worked.

Then I did the following:

    sudo apt-get remove omxd
    sudo apt-get remove omxplayer
    sudo apt-get update
    sudo apt-get install omxplayer
    sudo apt-get install omxd

And then, it worked! I don't know what the problem was, but it seems that removing the older version before installing the newer version solved the problem.

One more point:
I just realized that by default, the status information is disabled by default in the newer version of omxd. (Maybe `--no-osd` is enabled by deafult.) Because I don't see any status information anymore.
How should I enable these status information?

subogero, 2015-02-17:
Yes, I've enabled `--no-osd` by default. Because the `Paused` status info of the next track starting in the background was very annoying. OSD should be disabled by default I think. The `omxd S` command gives you everything you want to know.

subogero, 2015-02-17:
I'm just about to make a new release with a fixed segfault, which was exposed by these changes. If you very very very much want the on-screen-display back, I can remove the `--no-osd` option. :-)

OMID-313, 2015-02-17:
Thanks @subogero for your reply.

I think right now I'm OK with the osd beign disabled.
But I'd like to see an option in the future releases to be able to enable/disable osd whenever we want.

Thanks again for your time and support.

subogero, 2015-02-17:
I'll remove the default `--no-osd`, because that's the only way to preserve freedom. Users can now add it manually if necessary.

OMID-313, 2015-02-18:
Thanks @subogero for your revisions.

One question:
Right now, I'm using `omxd` in this way:

    omxd A home/pi/my_folder

If I want to enable several features at the beginning, which of the following is correct?
This way:

    omxd O --no-osd O -win 'x1,x2,y1,y2' A home/pi/my_folder

or this way:

    omxd O --no-osd
    omxd O -win 'x1,x2,y1,y2'
    omxd A home/pi/my_folder

?
Which way is correct?

subogero, 2015-02-18:
You can pass all the options in one step:

    omxd O --no-osd --win 'x1 x2 y1 y2'
    omxd A /home/pi/my_folder

OMID-313, 2015-02-18:
Thanks a billion :)