subogero/omxd #30: problems with playlist looping

WillyPinguino, 2015-04-12:
Hi guys!

first of all, a big big thanks to @subogero, your daemon is great!

now experimenting with it, I've founded a strange issue searching to play the playlist at the startup.

I've addedd 
omxd g 0
in the /etc/init.d/rc.local (in etc/rc.local where @subogero suggest to put it, unfortunately doesn't work) and it's work, but omxd result invoked from root and in this case, to send comands at the daemon I must use the echo way with sudo permissions if I need to play an other video out from the playlist with I option
ok ok, I come at the point...
if I change the audio output with a simple 
omxd h
and call the video out from the playlist with the 
sudo echo I /home/pi/videos/parla.mp4 > /var/run/omxctl
I can see the right video and at the end the playlist start from the interrupted point, but when I reach the end of the playlist, it's don't restart!

well I suppose that my issue is rare, because I suppose that there is not common that the hdmi output is setted on group 2 in dvi mode, and no others than me need to change the audio output, but in every case I think that is a strange issue...

WillyPinguino, 2015-04-12:
hemmm.. and now I cannot use omxd from common user, I've removed lines in /etc/init.d/rc.local and /etc/rc.local and still it's start from root permissions... I've tired to uninstall and reinstall omxd and omxplayer and it's don't want work :(

now I can only use it with sudo arguments and send the comands using FIFO messages...

and using the h argument to activated hdmi audio output killl the loop function :(

subogero, 2015-04-13:
The new versions of `omxd` come with a proper init script `/etc/init.d/omxd` and if you do `make install` or even better, use the deb package, you don't have to create any files manually. It just works.

Originally `make install` added an entry to `/etc/rc.local`, but now it creates `/etc/init.d/omxd` and registers it to start automatically at boot.

I also don't get why you need root to write into the FIFO, `make install` creates it with permissions 466, so everybody can write into it.

WillyPinguino, 2015-04-13:
I used normal 
sudo apt-get install omxd

maybe was my fault to explicate my situation. I need that the playing of the playlist start at boot, and to do it I've added 
omxd g 0
omxd O --no-osd
in the /etc/rc.local . at the first attempt it's won't work, and I've added it on the /etc/init.d/rc.local and from that moment omxd listen comands only from superuser .
I've tired to uninstall omxd and omxplayer and reinstall it using apt-get but without behavior change's

subogero, 2015-04-13:
I see. Please remove `/etc/init.d/rc.local`. On the other hand, I've reproduced your problem, investigating.

WillyPinguino, 2015-04-13:
I've removed the lines that invoke omxd from /etc/init.d/rc.local but nothing happens... do you mean that I need remove all the rc.local file?

subogero, 2015-04-13:
Put your stuff into `/etc/rc.local`, not `/etc/init.d/rc.local`. But don't remove the latter file.

WillyPinguino, 2015-04-13:
ok I've do it from yesterday but still I need to use superuser privileges to invoke omxd comands and it's work only if I use the echo way

subogero, 2015-04-13:
Please tell me the output of

    ls -l /var/run/omxctl

WillyPinguino, 2015-04-13:
lol.. was my mistake!  probably I've edited it with root privileges...
ls -l /var/run/omxctl
return
prw--w--w- 1 root root 0 Apr 12 22:19 /var/run/omxctl

now how I can change the permissions of the file?

subogero, 2015-04-13:
Those permissions are actually correct. What happens when you say `omxd n` ?

WillyPinguino, 2015-04-13:
now I remember... I've read it using 
sudo nano /var/run/omxctl
and probably I'm exit from nano making some change and saved it... damn me and my ignorance... :(

subogero, 2015-04-13:
It's a named pipe. Editing it results in undefined behaviour. :-)

WillyPinguino, 2015-04-13:
this is the result of this comand without superuser privileges

pi@raspberrypi ~ $ omxd n
Opening FIFO 'omxctl' timed out.
The omxd daemon is dead or listening on another one.
An unprivileged omxd listens on omxctl in its current dir.
One started as root listens on '/var/run/omxctl'.

subogero, 2015-04-13:
Do you have a file called `omxctl` in the current directory? If so, delete it.

WillyPinguino, 2015-04-13:
using sudo the result is the same...

pi@raspberrypi ~ $ sudo omxd n
Opening FIFO 'omxctl' timed out.
The omxd daemon is dead or listening on another one.
An unprivileged omxd listens on omxctl in its current dir.
One started as root listens on '/var/run/omxctl'.

no I got no files called omxctl in my home directory, but I got omxd.pid like you can see here 
pi@raspberrypi ~ $ ls
autostart.sh  ocr_pi.png  omxlog        snapshots           videos
Desktop       omxctl      python_games  startfullscreen.sh
fermo.png     omxd.pid    README        startvideos.sh

subogero, 2015-04-13:
If you call `omxd` without arguments as an unprivileged user, the daemon starts, but all the files it uses are created in the current directory: `omxctl`, `omxstat`, `omxlog`, etc. You need to clean that up.

subogero, 2015-04-13:
Actually all the files are there in your list.

WillyPinguino, 2015-04-13:
ok, but if I call omxd with g 0 in the /etc/rc.local maybe I call it before a login and I invoke it with superuser privileges, is it possible?

WillyPinguino, 2015-04-13:
lol... it's in yellow and I've not seen it  :)

subogero, 2015-04-13:
You might have called `omxd` ONCE in the past as non-root. The files remain there until you clean them up.

WillyPinguino, 2015-04-13:
ok I've removed it and now the call from an unprivileged user works fine...

WillyPinguino, 2015-04-13:
now I try to reboot the pi

subogero, 2015-04-13:
It won't work. I've already reproduced your original problem. As I've mentioned, I'm investigating it.

WillyPinguino, 2015-04-13:
now finally it's work like it must do ...
pi@raspberrypi ~ $ omxd n
n

hummm.. I think I need to investigate another lil thing

WillyPinguino, 2015-04-13:
ok, now I've investigated my experiments... I've added a webcam in my configuration to use the motion detection software, and my need was invoke the external video from the playlist only from motion detection, before the solution that you give me 10 minutes ago, I invoke the extra video using sudo and echo... can maybe it's provoke the omxctl file presence in my home directory?

WillyPinguino, 2015-04-13:
hummm.. no, the h option still provoke the playlist looping exit... and with a 
omxd n
or
omxd p (2 times)
I can only play one file of my playlist and after that it's stop...

subogero, 2015-04-13:
Please post the contents of the `/var/log/omxlog` file.

WillyPinguino, 2015-04-13:
this is the content

1428879563 daemonize: omxd started, SID 1817
1428879569 main: g 0
1428879569 player: start /home/pi/videos/scrive.mp4
1428879569 main: O --no-osd
1428879569 main: .
1428879569 player: start /home/pi/videos/scrive.mp4
1428879569 main:
1428950661 main: I /home/pi/videos/parla.mp4
1428950661 player: start /home/pi/videos/parla.mp4
1428950661 main:
1428950849 quit_callback: start /home/pi/videos/scrive.mp4
1428950886 quit_callback: start /home/pi/videos/scrive.mp4
1428950924 quit_callback: start /home/pi/videos/scrive.mp4
1428950948 main: h
1428950948 player: start /home/pi/videos/scrive.mp4
1428950948 main:
1428950961 main: I /home/pi/videos/parla.mp4
1428950961 player: start /home/pi/videos/parla.mp4
1428950961 main:
1428951147 quit_callback: start /home/pi/videos/scrive.mp4
1428951297 main: I /home/pi/videos/parla.mp4
1428951297 player: start /home/pi/videos/parla.mp4
1428951297 main:
1428951483 quit_callback: start /home/pi/videos/scrive.mp4
1428951605 main: n
1428951605 player: start /home/pi/videos/scrive.mp4
1428951605 main:
1428951716 main: p
1428951716 player: play/pause
1428951745 main: p
1428951745 player: play/pause
1428951745 main:
1428951745 quit_callback: start /home/pi/videos/scrive.mp4

subogero, 2015-04-13:
Let me guess. Your playlist contains a single file, `scrive.mp4`.  How long is it? And who issues the `I parla.mp4` commands? Are you aware that `parla.mp4` never becomes part of the playlist?

WillyPinguino, 2015-04-13:
all true! 
only one file, scrive.mp4 (repeated 3 times) in the playlist, the duration is 37 seconds and the scrive.mp4 is a file not in playlist but in the same folder (/home/pi/videos/)
the comand to start parla.mp4 is invoked from motion (http://www.lavrsen.dk/foswiki/bin/view/Motion/WebHome) when it's recognized a movement in the usb cam connected at the raspi

subogero, 2015-04-13:
So you say that at epoch-time 1428951745, scrive.mp4 is not started? What if you say `omxd p` a third time? What's suspicious is that the past `p` and the `quit_callback` happen at the same time. Did you pause it just before `la fine`?

WillyPinguino, 2015-04-13:
when I invoke the I option (moving in front of the cam or calling it from ssh) the video padla.mp4 starts and works good with the audio through hdmi exit.

the quit_callback comes when I invoke 2 times omxd p 
the first call do nothing, the second one start scrive.mp4 only one time and after it the image stay freezed on the same frame than before

subogero, 2015-04-13:
What's funny, there's 111s between starting scrive.mp4 and the first `p`. With a 37s long video. I'll investigate the omxplayer instance management in case of a single-element playlist and the `n` command. You probably noticed, all problems started after the `n`.

WillyPinguino, 2015-04-13:
not after the     n     but after the     h

WillyPinguino, 2015-04-13:
not after the `n` but after the `h` 
if I don't invoke the `h` comand, the playlist looping without problems continuosly and after the `omxd I /home/pi/movies/parla.mp4`
the 111 seconds are the time that I've waiting to be sure that the playlist don't looping... after the end of `parla.mp4` I've seen the `scrive.mp4` start and play, but I don't know if it's play 3 times or less, when I was sure that it's don't loop, I've tired to call `omxd n` from ssh, verified that it's don't work I've tired `omxd p` 2 times. the first one do nothing, the second one play `scrive.mp4` one time only and stil freeze on the frame where the video was stopped to play `parla.mp4`

(I'm happy to had founded a solution for the backtick to corectly format the posts... well semes that italians programmers are hated from M$ programmers... without some trick for italians that use italian layout is impossible to type the backtick without the ASCII code and my laptop have not the numeric keypad...)

subogero, 2015-04-13:
OK, I'll check the `h` command.

subogero, 2015-04-14:
Get a US keyboard. With the US International layout you can have backticks and àèéìòù as well.

WillyPinguino, 2015-04-14:
mmm I hope that you found a solution, good luck for it

unfortunately retrive a US keyboard for a laptop isn't so easy or cheap... but I've found a good compromise changing the keyboard default layout with another layot, provided by microsoft without specifications but it's provide an easy shortcut for the `\``and `˜`... italian programmers are safe  :-P

WillyPinguino, 2015-04-14:
if I can bother you for a little side question, I need to make a simple script (probably a bash script will be the right solution) that when invoked from a Motion event, verify if omxd plays one certain video or another.
how can I use the `omxd S` comand result to do it? (I'm a newbbie in linux program, you can see it :) )

subogero, 2015-04-14:
video=parla.mp4
    omxd S | grep "Playing .*$video" && do something

closed 2015-04-14 by subogero