subogero/omxd #31: No Audio Output over HDMI

OMID-313, 2015-04-23:
Hi,

A few months ago, I was using my RPi B+ with a composite cable and an old TV, to play videos in loop using omxd. It would work without any problems.

Now I changed my TV to a brand new Samsung LED TV, and decided to use HDMI cable.
Unfortunately, there's no audio output!

This is the code I use:

    omxd X
    omxd h
    omxd A <my folder>

I also checked the '<config.txt>' file in BOOT partition, and enabled/checked the following:

    hdmi_force_hotplug=1
    hdmi_drive=2
    hdmi_safe=1

But I don't know why there's no audio output!

Also, I tried '<omxplayer>' itself, and it worked !!
Using '<omxplayer>' in terminal like

    omxplayer file.mp4

has very nice audio output.

But unfortunately using '<omxd>' doesn't have audio!

What's the problem !?

subogero, 2015-04-23:
I can't reproduce the problem. I can switch between the HDMI and Jack audio outputs with the `omxd h` and omxd j` commands. Please check your omxplayer processes started by omxd:

    ps -AF | grep omx

Thus you can inspect the actual arguments omxd uses to start omxplayer.

OMID-313, 2015-04-23:
Thanks @subogero for your reply.

Actually, I use the following codes :

    omxd X
    omxd h
    omxd A <my folder>

inside `root/etc/rc.local` , so I don't see the main terminal.

Where should I type your suggested code?

    ps -AF | grep omx

subogero, 2015-04-23:
Type it in a shell. BTW, I reproduced your problem, but it's a temporary pause in sound, caused by the 2 omxplayer instances racing for the HDMI sound. There is an omxplayer issue about it somewhere.

I'll have to disable low-gap playback for videos.

OMID-313, 2015-04-23:
Thanks @subogero for your reply.

A few questions:

1. Would it be OK if I use the following codes:
    omxd X
    omxd h
    omxd A <my folder>
    ps -AF | grep omx
inside `root/etc/rc.local` !?

2. What does this `ps -AF | grep omx` do exactly !?
Does it save the logs and reports in a file, so that I can copy them here !?

3. If you disable (or let's say sacrifice) "low-gap playback" to solve this issue, what features will we lose?

subogero, 2015-04-23:
1. Putting the `ps` stuff into `rc.local` is pointless
2. Did you try it? The answer `no` and further questions are not accepted. :-)
3. In case of a playlist if videos, the gap between them will increase

OMID-313, 2015-04-24:
Thanks again @subogero for your helpful reply.

1,2. Well, I haven't tried it yet. Is the following OK in `rc.local` ?

    omxd X
    omxd h
    omxd A <my folder>
    test_script

and in `test_script` :

    ps -AF | grep omx

and then, before restarting, do:

    chmod 755 test_screen

Will this all work well !?

3. How much the gap will increase !? 7-8 seconds !?!? (Now the gap between playlist videos is less than 1-2 seconds which is really good.)

subogero, 2015-04-24:
[1-2] PLEASE PLEASE PLEASE! TRY IT! I'm absolutely not willing to teach you basic Linux, if you're not willing to learn by doing. Sorry.
[3] Yes, it may increase a lot. But the omxplayer issue indicated that fixing the sound for 2 instances requires a big redesign, so they won't do it. But the bigger gap is still better than no sound, right?

OMID-313, 2015-04-27:
Thanks @subogero for your reply, and sorry for my late answer.

I tried `ps -AF | grep omx` which gave the following info:

    root 1935    1 0 424 364 0 15:21 ? 00:00:00 omxd
    root 2216 1935 0 766 1272 0 15:21 ? 00:00:00 /bin/bash /usr/bin/omxplayer -I --no-osd -olocal /home/pi/my_video_1.mp4
    root 2217 1935 0 766 1236 0 15:21 ? 00:00:00 /bin/bash /usr/bin/omxplayer -I --no-osd -olocal /home/pi/my_video_2.mp4
    root 2219 2207 0 506 656 0 15:21 ? 00:00:00 grep omx

OMID-313, 2015-04-27:
Dear @subogero , I got some interesting info:

Instead of running `omxd` on `rc.local` , I waited for the actual system to boot, then user/pass login,so the main terminal appeared.

In the main terminal, I typed:

    omxplayer my_video_1.mp4

and it started to play with beautiful sound.
Then I typed:

    omxd X

to terminate all the saved playlists, and then:

    omxd A my_video_1.mp4

but unfortunately, it didn't have any sound :(

It seems that the **no sound** problem is not related to **several instances** nor **low gap**. As you see, there's no sound even when trying to play a single video file using `omxd` .

I wonder why this problem is happening to me!! For several years, you and many people have used `omxd` over hdmi without problem. Now I don't know why it doesn't work for me :( It's really weird!

Maybe the problem is something else. What do you think @subogero ?

OMID-313, 2015-04-27:
Is it possible that the problem is related to `tty` !?

Because in the info that `ps -AF | grep omx` gave, there are **?** marks instead of terms like **tty1** or **tty3**.

What do you think @subogero ?

subogero, 2015-04-27:
The ps command shows that omxd starts the omxplayer instances with the `-olocal` option which is not  HDMI.

Please repeat the ps-grep check after creating the single-track playlist. You'll probably see 2 omxplayer instances with the same video. That's why there's no sound I guess.

OMID-313, 2015-04-27:
Dear @subogero ,

After creating a single-video playlist (`omxd A video_1.mp4`), I typed `ps -AF | grep omx` (while the video was playing), and then stopped the video to see the output info:

    . . . .
    pi 2216 1935 0 766 1272 0 15:21 ? 00:00:00 /bin/bash /usr/bin/omxplayer -I --no-osd -olocal /home/pi/my_video_1.mp4
    pi 2216 1935 0 766 1272 0 15:21 ? 00:00:00 /usr/bin/omxplayer.bin -I --no-osd -olocal /home/pi/my_video_1.mp4
    . . . .

(The numbers are not exact.)
As you see, there are two instances.

subogero, 2015-04-27:
As you can see, omxd starts the omxplayer instances with the -olocal option. Did you issue `omxd h` ?

OMID-313, 2015-04-27:
I have some interesting info dear @subogero :

I did the following:

    omxd X
    omxd
    omxd A my_video_1.mp4

The video started to play without sound.
Then I typed the following (while the video was playing) :

    omxd h

Yessss! The sound came back! Beautiful sound :D 

then I typed the following (while the video was still playing) :

    ps -AF | grep omx

and the nstopped the video to see the output info:

    . . .
    pi 2216 1935 0 766 1272 0 15:21 ? 00:00:00 /bin/bash /usr/bin/omxplayer -I --no-osd -ohdmi /home/pi/my_video_1.mp4
    pi 2216 1935 0 766 1272 0 15:21 ? 00:00:00 /usr/bin/omxplayer.bin -I --no-osd -ohdmi /home/pi/my_video_1.mp4
    . . .

As you see, the `-olocal` has benn changed to `-ohdmi`; And of course there is sound, despite having two instances.
This probably means that the **no sound** problem is related to **-ohdmi** settings rather than **low gap** or **two instances** .So we have to find a way to enable hdmi audio output.

Because the following doesn't help:

    omxd h
    omxd A my_folder

Is it possible to write something like this:

    omxd h A my_folder

!?

OMID-313, 2015-04-27:
@subogero 

Or as an alternative solution, is it possible to pass the `-ohdmi` option to `omxplayer` using `omxd O ...` that you developed a few months ago !?

subogero, 2015-04-27:
Just swap the 2 commands in rc.local

    omxd A xxx
    omxd h

OMID-313, 2015-04-27:
Thanks @subogero for all your support.

I tried the following code:

    omxd A my_folder
    omxd h

but unfortunately, 2 things happened: (1) there was no sound, and (2) video_0 and video_1 (of my_folder) started to play at the same time! It was really strange, because it was playing both videos' frames simultaneously !! when I paused (`omxd p`) the first video paused but the second video continued to play. And I didn't have any control on it.

So as you see, we have to find a way to pass the `-ohdmi` option to `omxplayer` **at the same time**, not before, and not after. Something like `omxd h A my_folder` would be nice.

It is really strange that omxd's default have been set/changed to `-olocal`.
Is there any way to change this setting, so that we don't have to run `omxd h` each time !!?

subogero, 2015-04-27:
What you describe are clearly bugs. I'll try to reproduce and fix them.

OMID-313, 2015-04-27:
By the way, @subogero , while I was trying many ways, I found a funny and crazy way to enable the sound:

    omxd
    omxd X
    omxd A my_folder
    omxd P
    omxd g 0
    omxd h

After the above sequence, the 1st video of `my_folder` starts to play with beautiful sound.
But for the next videos of the playlist, we have to do:

    omxd p
    omxd p

each time in order to enable its sound!

Can you suggest a stable solution based on this funny method !?

subogero, 2015-04-27:
Why is the first `omxd` there??? You probably start a second omxd instance, which might screw up everything. If omxd is installed properly, its init script `/etc/init.d/omxd` is registered for automatic start at boot, and you don't have to start it it `/etc/rc.local`.

OMID-313, 2015-04-27:
Well, you're right @subogero . It is not needed.
But some times I had to do it!

For example, After login to pi, I typed `omxd X` , but I received the following error/warning:

    Opening FIFO 'omxctl' timed out.
    The omxd daemon is dead or listening on another one.
    An unprivileged omxd listens on omxctl in its current dir.
    One started as root listens on '/var/run/omxctl'.

This error/warning happened to me several times later too, at different situations.
And when this error came, none of the `omxd` commands worked.
And the only way to get rid of this situation, was to type:

    omxd

to start another instance with a new PID.
So that's why I inserted `omxd` at the beginning, to avoid this error.

subogero, 2015-04-27:
Never ever start `omxd` as a non-root user! It will start omxd in test mode, and clutter your current directory with the control and log files. That's also the reason for the error message above. If you've done it accidentally, you must delete the following files from your current directory:` omxlog`, `omxstat`, `omxctl`, `omxplay`.

OMID-313, 2015-04-27:
Thanks @subogero for your point.
I will log in as root (username:root / pass:raspberry !!?), or try `sudo su` after normal login.
Thanks for the point.

Anyway, I look forward to hearing good news regarding these problems.

(Personally, I think that permanently changing the default sound output from `-olocal` to `-ohdmi` is enough. How can I do it!?)

subogero, 2015-04-27:
Again. You don't need to log in as root. If you boot your Pi, omxd is started automatically as root. Just never start `omxd` without arguments. OK?

In the meantime I've found the bug that caused omxd to ignore the `h` command. Fix coming soon.

OMID-313, 2015-04-27:
Thanks again @subogero for all your help and support.
OK, OK, I got it ;)
Looking forward to the fixes. Thanks.

subogero, 2015-04-27:
Can you proceed with the new commits, or do you need a new release of the deb package?

OMID-313, 2015-04-28:
If you help me, @subogero , yes I can.
What shall I do exactly?

subogero, 2015-04-28:
Go into your omxd repo's directory and

    git fetch
    git merge origin/master
    make
    sudo make install
    sudo service omxd start

OMID-313, 2015-04-28:
Thanks for your helpful instructions @subogero .

Sorry, just a silly question:
How should I go to omxd repo's directory!? Where is it?
After I login to RPi, if I type the above codes, is it OK !?

(Sorry I'm a beginner in linux and git!!)

subogero, 2015-04-28:
In case you installed omxd from the Deb package, you need to clone the git repo.

    git clone git@github.com:subogero/omxd.git
    cd omxd 
    make
    sudo make install
    sudo service omxd start

But of course, you need git and the build tools before that. If you don't have them, you need to install them beforehand:

    sudo apt-get install build-essential git

OMID-313, 2015-04-28:
Thanks again @subogero .

I had git installed, so there's no need to install it again.

After I typed 

    git clone git@github.com:subogero/omxd.git

the following error/warnings came:

    Cloning into 'omxd' ...
    The authenticity of host 'github.com (192.30.252.130)' can't be established.
    RSA key fingerprint is 4656646446...
    Are you sure you want to continue connecting (yes/no)?
    Warning: Permanently added 'github.com (192.30.252.130)' (RSA) to the list of known hosts.
    Permission denied (publickey).
    fatal: The remote end hung up unexpectedly.

and then when I type

    cd omxd

it says:

    bash: cd: omxd: No such file or directory.

I tried these codes with both Pi and root users, but both give these errors.
What shall I do !?

subogero, 2015-04-28:
Sorry, I gave you the read-write SSH URL, which you can't use. Use the read-only HTTP URL:

    git clone https://github.com/subogero/omxd.git

OMID-313, 2015-04-28:
Thanks @subogero for your quick reply.
I do really appreciate your great support.

I cloned the git repo successfully, following your detailed instructions.
Then I rebooted the system.

Then I typed the following in `/root/etc/rc.local`:

    oxmd X
    omxd A my_folder
    omxd h

and rebooted the system, to start again.
But unfortunately, strange things happened:

1. After system boot (about 20 seconds), when it was time to play the videos, a **pause** notification appeared at the bottom of the screen, and then the **No Signal** error appeared on TV screen, asking to check the HDMI source and cable.
2. I disconnected the HDMI cable, and connected it again. But nothing changed. Still the same error on TV screen. I waited for 2-3 minutes (the length of video_1). Then Suddenly the error went off and the last 10 seconds of video_1 started to play. It finished, and then video_2 started to play.
3. The video_2 had sound when started; but after 4 seconds, the **pause** notification appeared again at the bottom of the screen and the sound got mute, but the video was still there playing.
4. The only way to bring its sound back was to do `omxd p` twice!
5. Points 3 and 4 happened to all of the other videos in the playlist too.
6. Some of my videos (video_4,5,6) have short lengths (22, 9, 9 seconds respectively). When these short videos finishes, they freeze on their latest frames, and don't switch automatically to the next video. I have to do either `omxd n` or `omxd p` (twice) to go to the next video.

OMID-313, 2015-04-28:
Dear @subogero ,
Do you have any idea/solution to solve these problems/bugs !?

subogero, 2015-04-28:
These are exactly the symptoms of the 2 omxplayer instances.

OMID-313, 2015-04-28:
That's true, @subogero .

But if we find a way to enable `-ohdmi` by default (instead of `-olocal`), there will be no need to run `omxd h` after each playback, and these problems probably will not occur.

Hundreds of people are using `omxd` over `hdmi` without any problem. Still I wonder why these problems are happening to me! Maybe that's because I was using AV component before. Maybe default settings have been changed to `-olocal`.

What to do now !!?

subogero, 2015-04-28:
Have you checked `ps -AF | grep omxplayer` again? I'm sure you'll see `-ohdmi` everywhere, as the `omxd h` command sets the audio output to HDMI forever.

OMID-313, 2015-04-28:
Well, I checked `ps -AF | grep omx` , but it had different results:

1. After login, I typed `omxd X` and then `omxd A my_folder`. It didn't have sound, and `ps -AF | grep omx` reported that it was `-olocal`. I stopped the playback. Typed `omxd h`. Now it was `-ohdmi` and had sound. But again in some videos I had to do `omxd p` twice to enable the sound.
2. In rc.local, I typed 

omxd X
omxd A my_folder

and removed `omxd h` from the end of it (as you said `omxd h` sets the audio to HDMI **forever**). After that, while playing the videos, I tried `ps -AF | grep omx`, and it said `-olocal` again!!

So it seems that `omxd h` does not set the audio to HDMI forever, and we have to enable it each time.
Is there any other way to change the default from `-olocal` to `-ohdmi` forever!?

OMID-313, 2015-04-28:
Some interesting info:

After login (as root), if I type something like `omxd A my_folder`, the ps-grep reports `-ohdmi`.
But if start another omxd daemon by typing `omxd` , and then `omxd A my_folder`, the ps-grep reports `-olocal`.

This means that the default audio settings (when creating a daemon) is `-olocal`.
But the initializations change it to `-ohdmi` probably.
This is probably the reason why the codes in `rc.local` have `-olocal` (because it has not been initialized yet), and the codes after login have `-ohdmi`.

I think it's better to change the settings in a way, so that whenever an omxd daemon is created (either by user, or automatically after boot or whenever) it becomes `-ohdmi` by default. This might solve a part of the problems.

subogero, 2015-04-28:
Do not, ever, start a second omxd daemon instance! I do not give support for that kind of crap.

OMID-313, 2015-04-28:
I won't start a second omxd daemon. That's OK.

I just wanted to diagnose the problem.
And you see the result:

Currently, on boot, omxd starts with `-olocal`, runs codes in `rc.local` with `-olocal`, and then it gets initialized and changes to `-ohdmi` and runs he codes after login with `-ohdmi`.

But we want omxd to start with `-ohdmi` , not `-olocal`.

subogero, 2015-04-28:
That's exactly what `omxd h` is for.

OMID-313, 2015-04-28:
OK.

I changed the codes in `rc.local` to the following:

omxd X
omxd h
omxd A my_folder

Now, every time after boot, the following scenario happens:

1. The first video starts to play with beautiful sound, without any problem.
2. For the next videos, after 3-4 initial seconds with sound, a **pause** notification comes and the sound gets mute.
3. If I do `omxd p` twice (pause and then play again ), the sound comes back. Also the **freeze** problem of the short-length movies gets solved. (It seems that the freeze problem was happening because I was not doing double-pause/play.)

So I think there are two solutions for this problem:

1. Preventing `omxd` from starting 2 `omxplayer` instances.
2. Running double-pause (`omxd p` twice) automatically after the video starts to play.

The 2nd solution is crazy, and not technical. But after all, it solves the problem!!
BTW, How can we implement the 2nd solution automatically !?

What do you think !?

subogero, 2015-04-28:
I'll disable the 2 omxplayer instances for videos. The gap will be larger but the sound will be ok.

OMID-313, 2015-04-28:
Thanks @subogero for your support.

Well, it's good to hear that the sound will be ok; but it's bad to hear that the gap will be longer.
Are there any other ways to reduce the gap !?

BTW, the 2nd solution above (double pause/play after video starts) can solve the sound's problem, while still keeping the low gap. What do you think about that !?

subogero, 2015-04-28:
Good idea. I'll try it, at least fir videos only. If it works, you deserve a beer. :-)

OMID-313, 2015-04-28:
Well, it works for me.

After the video starts, I do

    omxd p
    omxd p

and then the sound comes back :D 

Actually, the beer is yours, for your great support and help ;)

OMID-313, 2015-04-28:
One point @subogero :

Remember I told you about the *pause notification* that appears after 3-4 seconds of the video start, and mutes the video!?

I just noticed that we must use the **double-pause** method **AFTER** that pause notification.
Because when I do double-pause right at the beginning, after 2-3 seconds that pause notification comes and mutes the video again!

Maybe it would be better to run several double-pauses along the video!!

subogero, 2015-04-28:
That Pause noticication comes from the next video prepared in the background. It's enough to wait for that.

OMID-313, 2015-04-28:
That's good.
But maybe it would be better to do 2 double-pauses: one double-pause at the beginning, and one double-pause after that notification.

Anyway, I look forward to the fixes.
Thanks for your time and support.

subogero, 2015-04-28:
It seems to work, but it's a terribly ugly hack :-)

OMID-313, 2015-04-28:
Beautiful sound and low gap via ugly hack :D 
That seems a reasonable deal, until a stable solution is found ;)

I look forward to updating repository or installing deb package.

OMID-313, 2015-04-29:
Dear @subogero ,

I just reviewed the `omxd.c` commit that you referenced above, and noticed a few things:

1. You have set the program to wait for 2.5 seconds at the beginning. But I have noticed that the *pause notification* appear in different times at different videos. For example, 3 seconds, 4 seconds, 7 seconds, 1 second, ... . How did you result in 2.5 seconds !?
2. The time between two pauses has been set to 0.6 second (`usleep 600000`). Isn't this too long!? This might cause a noticeably small lag in video playback. I know that `omxplayer` skips fast commands, but 600000 microseconds is too much. Something like 50000 microseconds or at most 100000 microseconds shall be enough. (I've tested these numbers before, for creating mute/unmute commands using `omxd -` and `omxd +` in loop, and as I mentioned a few months ago [here] (https://github.com/subogero/omxd/issues/26#issuecomment-73029360) , the 50000 microseconds is the optimum number for the delay between `omxplayer` commands.)
3. I really appreciate your time, help, and great support. But your comments at the beginning of that commit about this ugly solution related to a specific country seem a little bit racial in this open-source society. Right!? ;) I'd be really thankful if you revise those comments.
4. Can I now update `omxd` via the following commands?

        git clone https://github.com/subogero/omxd.git
        cd omxd 
        make
        sudo make install
        sudo service omxd start

subogero, 2015-04-29:
1. That Pause notification is not seen by omxd, it just goes to the screen. Its time depends on the length of the video I guess. But I'm thinking about a terribly dirty Hungarian hack... :-)
2. I've tried shorter periods between the 2 `p`s, but it did not seem to work every time.
3. It's actually a compliment. Search for quotes by Ken Thompson, the greatest programmer of all time. But I'll remove it if you wish. :-)
4. No need to clone again

    cd omxd
    git fetch
    git checkout -t origin/hdmi
    make
    sudo make install
    sudo service omxd start

subogero, 2015-04-29:
"Racist" remark removed. :-)

OMID-313, 2015-04-29:
Dear @subogero ,

Thanks for all your help and support.

I just fetched the new changes, and the following are some points:

1. For 2/6 of  my videos (lengths of these videos: 2 minutes, 3 minutes) this method works well, though with an **inevitable** lag of ~1 second; but for 4/6 of them (lengths of these videos: 5 minutes, 22 seconds, 9 seconds, 10 seconds) it doesn't work! The reason is that our *double-pause* solution occurs **BEFORE** the *pause-notification* in the background. Hence, that *pause-notification* causes the video to mute.

2. Because the *double-pause* solution occurs **BEFORE** the *background pause notification*, that unusual *freeze* problem happens at the end of some short videos again.

3. I said this method works well for 2/6 of my videos, because our *double pause* solution occurs **AFTER** that *background pause notification*. But ... when I use `omxd n` (next) to switch from previous *bad* videos to these *good* videos, the timing of *background pause notification* and *double pause* solution mix together, in a way that sound gets mute!

It seems that this solution causes many more problems compared to a few it solves :D 
Even if all these errors and problems get solved, that *inevitable ~1 second lag* doesn't feel good.

I think we need a truly stable and reliable solution.
Any ideas !?

subogero, 2015-04-29:
My dirty Hungarian hack would involve looking at the length of the next video from its own `omxplayer.log.xxxx` file, and wait a proportional amount of time until the double-p. But I just found out that the necessary time to wait also depends on processor load. 

I think I should just disable preparing the next track in the background if
* it's a video
* current audio output is HDMI

That will increase the gap between videos, but only in case of HDMI audio. Jack audio is fine, I've tried.

OMID-313, 2015-04-29:
Is it possible to send video over hdmi, and audio over av jack !?

If yes, and if we do it,
what problems will be solved!? And which problems will remain again!?

subogero, 2015-04-29:
Yes, it's possible: `omxd j`. If you mean the audio and video streams of the same track.

OMID-313, 2015-04-29:
Well, 
(1) if using `omxd j` sends video to HDMI and audio to AV Jack,
and 
(2) if using Jack audio doesn't have any problem with low-gap
Then
It means that we can disable low-gap, and still benefit from low-gap :D Right!?

It would be reasonable to give it a try.
How can I test it !?

[23 more replies]

closed 2015-05-02 by subogero