subogero/omxd #33: Too long gap when using `omxd g num`

OMID-313, 2015-05-23:
Dear @subogero,

I remember that you had found a way to decrease the long gap (7-8 seconds) between video playbacks.
This method works perfectly when automatically switching to the next video, or even when doing `omxd n` or `omxd N` to go to the next/previous videos.

But unfortunately, the long gap still happens when using `omxd g num` to go to nth track, which is really annoying.

Is it possible to reduce this long gap as well !?

subogero, 2015-05-23:
There is absolutely no way, unless you tell me how to predict the future. :)

OMID-313, 2015-05-23:
Well, Is it possible to apply the same method that you used to reduce the gap between video playbacks or on `omxd n` !?

They must be similar. Right @subogero ?

subogero, 2015-05-23:
Oh yes, they are nearly identical. So if I have a 100 element playlist, I just start a paused omxplayer instance for each at the beginning, and whenever someone hits `omxd g x`, I just un-pause the right one. Easy.

OMID-313, 2015-05-23:
OK, then.
I'm waiting for your update, to test and see what happens ;)

subogero, 2015-05-23:
Good luck waiting.

closed 2015-05-23 by subogero

OMID-313, 2015-05-23:
Why did you close this issue @subogero?
Is it ready to update?

subogero, 2015-05-23:
I assume from your remarks that the irony of "predicting the future" and "starting up 100 paused omxplayer instances" did not achieve its original goal.

So I repeat it in plain English:

    There is no way in hell I'll implement this,
    and I absolutely and positively refuse to even think about it.

Just buy a faster Raspberry Pi 2.

OMID-313, 2015-05-23:
Oh, ok.
So you were just kidding, and I didn't get your points!

Well, buying a faster RPi 2 can reduce this gap from 8-9 seconds to 6-7 seconds. (900 MHz vs. 700 MHz) So still the problem would exist.

There must be some kind of solution.
Nothing is impossible ;)

OMID-313, 2015-05-26:
Dear @subogero ,

This problem (too long gap) also happens when using `omxd N` to go to previous video.
Is there any solution for this one!?

subogero, 2015-05-26:
Yes. There are two solutions:

* starting up paused omxplayer instances for ALL tracks in the list. But what if there are 100 tracks. Or 1000? Or 10000?
* knowing the future, i.e. where will the user jump next? Just start up a paused omxplayer instance for that track beforehand

I refuse to implement the first and, as for the second, do you know a method for time travel? Otherwise in 95% of the cases the *next* track will follow, so I only start a paused omxplayer for that.

OMID-313, 2015-05-26:
I don't understand it @subogero .

Solving the problem of `omxd g num` is difficult. That's ok. End of story.
But the `omxd N` (PREVIOUS) must have a solution very similar to `omxd n` (NEXT).
Right?

subogero, 2015-05-26:
Look at the code, implement it and send me a pull request.

OMID-313, 2015-05-27:
Ok, @subogero .

Which file should I look for? `omxd.c` !? Or other files?