subogero/omxd #35: Possible to use with forked repos of omxplayer?

OMID-313, 2015-06-20:
Hi @subogero,

I'm currently using `omxd` (thanks for this nice repo) to control `omxplayer` on raspbian over RPi B+.

But since we have decided to play individual synchronized videos on several RPis (https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=113498), we might have to use `omxplayer-sync` (https://github.com/turingmachine/omxplayer-sync) or even `pwomxplayer` ( https://github.com/JeffCost/pwomxplayer and http://www.piwall.co.uk/ ), which both are forked repositories from `omxplayer`.

Now a few questions:

1. Can we still use `omxd` as a playlist daemon to control `omxplayer-sync` or `pwomxplayer` !?
2. If not, how can we enable `omxd` to get to work with them!? Is it possible to build new repos (e.g. pwomxd, or omxd-sync) for them!?
3. Do you have any other suggestions to solve this synchronization issue (https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=113498)!? Is it possible to implement it with bare `omxplayer` and `omxd` !?

Your time and support is highly appreciated.

subogero, 2015-06-20:
I've been thinking about this for a long time, to support multiple configurable player backends.

Right now, you can use it with any other player that has the same key bindings and command line options. Since omxd always uses `/usr/bin/omxplayer`, you can simply replace that with the other player.

First save the original omxplayer, renaming it (all operations as root):
    cd /usr/bin
    mv omxplayer omxplayer~
Then create a symbolic link to e.g. `pwomxplayer`, called `omxplayer

OMID-313, 2015-06-20:
Thank you very much @subogero for your quick reply.

As for the last line (creating a symbolic link), how should I create this link?
How should I tell `omxd` to use e.g. `/usr/bin/pwomxplayer` instead of `/usr/bin/omxplayer` ?

Thanks :)

subogero, 2015-06-20:
Reload the page, I hit the `Comment` button too early, I've edited the post since.

OMID-313, 2015-06-20:
Thanks @subogero , I got it.
I will try this method as soon as I get my new RPi, and will post the results.

By the way, would you suggest some solutions (other than PiWall and omxplayer-sync) to the synchronized video playback problem!? (https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=113498)
Can I do such things with only `omxplayer` and `omxd` !!?

subogero, 2015-06-20:
You could use `remotepi` on each RPi, which gives omxd a HTTP REST API. So you can control them simultaneously over the network. See https://github.com/subogero/remotepi

OMID-313, 2015-06-20:
Thanks @subogero for your suggestion.

I read the readme file of `remotepi`, but unfortunately I couldn't understand how to use it to solve my problem.
Would you please describe some details?
If you think this is not the right place, we can create an issue in `remotepi` page, and continue the discussion there.

subogero, 2015-06-20:
It's not trivial, because you have to write a program that uses the `remotepi` REST API. But it's possible.

OMID-313, 2015-06-20:
Thanks again @subogero for your reply.

Let's continue our discussion on `remotepi` here: https://github.com/subogero/remotepi/issues/13

closed 2016-09-30 by subogero