subogero/omxd #36: Youtube playlist

edap, 2015-08-11:
Hello, I was wondering if it's possible to run a playlist of youtube videos. 
As "playlist" i do not mean a YouTube playlist, but a text file containing youtube's video URLs

subogero, 2015-08-11:
Not yet, but I'm working on it. I'll get back to you when I have something.

Originally omxplayer was not able to use the necessary YouTube cookies, so the way to play YouTube is to stream the video by curl into a temporary FIFO, and let omxplayer play the FIFO file.

subogero, 2015-08-11:
In the meantime, see my other project, u2b. It can play YouTube via omxd. But still, no playlists...

subogero, 2017-05-04:
So far, you can use `u2b` to extract YouTube streams

    u2b u0s Valentino Rossi

will print the stream URL of the first search hit for Valentino Rossi. Then you can

    omxd A '<URL>'

which will add it to the playlist. I'm working on u2b to be able to do thius directly.

closed 2017-05-04 by subogero