subogero/omxd #37: Possible to play X items on play list then pause

kaptk2, 2015-08-14:
It is possible to play like 3 items on the play list then pause then wait for a another command to play say 1 item?

subogero, 2015-08-14:
Do you mean pause automatically? No.
If you just add 3 items to the list, omxd will loop through them forever, until you pause it manually.

subogero, 2015-08-14:
Do you want a command do disable looping?

kaptk2, 2015-08-15:
The idea is for an interactive display. So I would have a playlist of 20 short videos. A user would come to the screen a push a button that would send a command to play either 1, 2, 3 or 4 videos in a row. When those are over omxd would wait for another command to play another 1,2,3 or 4 videos.

subogero, 2015-08-15:
From my side it's still about disabling looping. If it's possible, you can store you 20 vids separately, when a user wants to play 1,2,3,4, then your script can do this

    omxd X # delete playlist
    omxd l # disable looping, not yet implemented
    omxd A 1
    omxd A 2
    omxd A 3
    omxd A 4

kaptk2, 2015-08-16:
Yeah, I suppose that would work. I'm worried about the lag time between pushing the button and the first video starting.

subogero, 2015-08-16:
OK. Now I'm totally confused.

kaptk2, 2015-08-17:
What you are proposing would work but... in my testing I noticed that there
was a lag time between doing this: 'echo A /path/to/video.mp4 >
/var/run/omxctl' and the video showing on screen. However if I make a
playlist before hand, there is no lag. So the lag time is the main thing
that I am concerned about.

subogero, 2015-08-17:
The delay mainly comes from omxplayer. When you create the playlist beforehand, omxd already preloads (starts paused) the first 2 videos. If you always want to play some videos from the beginning of the playlist, you can just create a full one. When the user does something, you start and delete the unnecessary part of the list.

    omxd p # start playback
    for i in `seq 4 19`; do omxd x $i; done

subogero, 2015-08-17:
So, shall I implement `l` for toggle looping?

kaptk2, 2015-08-17:
Yeah, I think disabling looping would get me most of the way to my goal

subogero, 2015-08-21:
Can you compile and test omxd from the `no-loop` branch? Or do you need a new version and a deb package?

kaptk2, 2015-08-21:
I'll compile and test

subogero, 2015-08-22:
Two more status (S) related fixes in the meantime...

subogero, 2015-08-22:
I've just realized that no-loop mode is not compatible with shuffle. Whenever the random jump causes an overflow, the shuffled playback would just stop. We need 3 playback modes instead:

* loop (default)
* once (no loop)
* shuffle

I propose activating each mode with a dedicated command, so it's easier to follow what's going on.

* l: loop
* e: stop at End of list
* u: shuffle (Unsorted)

What do you think?

kaptk2, 2015-08-23:
That sounds perfect

subogero, 2015-08-27:
Hi, does it work for you? If so, I merge and release a new version.

kaptk2, 2015-08-31:
Sorry for the slow reply, I was out of town and didn't have a chance to
test until tonight. Should the playlist still contain previously play items
when complete?

Thanks

Hi, does it work for you? If so, I merge and release a new version.

subogero, 2015-09-10:
Sorry for the late reply. Do you propose a self-annihilating playlist in "end" mode? Currently the playlist is preserved and you can play it again manually.

kaptk2, 2015-09-10:
Yeah, I think that would be best

subogero, 2015-09-10:
OK, I'll change it that way.

subogero, 2015-09-10:
Please test.

kaptk2, 2015-09-11:
It isn't working. When you add more than one file to the playlist the
second one (and subsequent ones) fail to play.

For example:
echo A /path/to/file1 > /var/run/omxctl
echo A /path/to/file2 > /var/run/omxctl
echo A /path/to/file3 > /var/run/omxctl

only file1 plays, file2 and file3 never get played

kaptk2, 2015-09-11:
I also notice on the previous commit
(65a29f2f71c3676fe13feff14a96962a04aa8546) that this: echo O --no-osd >
/var/run/omxctl doesn't appear to disable the osd like I would expect it to.

subogero, 2015-09-11:
Have you tried using omxd as its own client?

    omxd A /path/to/file1
    omxd A /path/to/file2
    omxd A /path/to/file2

    omxd O --no-osd

kaptk2, 2015-09-11:
I haven't, I'll try that today and report back

subogero, 2015-09-17:
If there are problems during your testing, could you please have a look at the `/var/local/omxplay` file, which contains the playlist, and check whether the file names are correct?

Another possible problem are userspace filesystems (fuse) like curlftpfs, sshfs etc. Only the user can see those mounts, but `omxd` runs as root.

kaptk2, 2015-09-17:
I'm still troubleshooting getting "omxd <command>" running correctly. It is
a function of my program and not of omxd but I did use the pipe version of
my program this past weekend with great success (on the previous version of
omxd, before the deleting of the playlist was added).

kaptk2, 2016-01-26:
Just going through my open issues and noticed this. Just wanted to mage sure you know that everything is working great. Thanks for this

closed 2016-01-26 by kaptk2