subogero/omxd #38: If you add a file with the "omxd L" does not play...

GigAl, 2016-01-03:
Hello! I do not have a team, "omxd L /filename"

If you add a file with the "omxd L" does not play, even though the file exactly there!
Meanwhile, on the screen that I can differentiate amongst played two times one and the same file that was added earlier.
I tried differently
omxd L /filename and echo L /filename > /var/run/omxctl

Thank you for your program! I would be very grateful if you help with this problem.

subogero, 2016-01-03:
Bug confirmed.

What I can see is this:
L should play a track after the current item but it does not get part of the playlist.
But instead, the next playlist item is played TWICE, once instead of the temporarily inserted track, then as the next item on the playlist.

subogero, 2016-01-03:
Can you install the fix from source?

subogero, 2016-01-03:
I'll release omxd 1.6 to my deb repo soon with the fix. But it'd be nice if you could confirm the fix before.

subogero, 2016-01-03:
omxd 1.6 released into the deb repo. As root:

    apt-get update
    apt-get install omxd

GigAl, 2016-01-17:
Thank you for your work!
I am very grateful for your program.

closed 2016-01-17 by subogero