subogero/omxd #39: Video Loop Freezes Randomly

OMID-313, 2016-01-09:
Hi @subogero ,

Thank you very much for this great daemon.
I remember `omxd` on RPi 1 B+ had some problems, like long delay on `next` or `previous` commands. However, after switching to RPi 2, and also using minimal raspbian (Minibian) as the OS, it is almost flawless. the delay is less than1 second, which is great.

Anyway. I'm experiencing some unusual behavior on this RPi 2 with `omxd`. I've set the system to play videos in loop after boot. Something like this in `/etc/rc.local` :

    omxd X
    omxd j
    omxd O --no-osd
    omxd A /myfolder

But I don't know for what reason, the video playback gets weird!! Sometimes the next video begins to play simultaneously with the current video. When this happens, the system freezes after 3-4 seconds. Or sometimes when this happens, the next video wins the fight(!!) and continues the playlist.

This unusual thing happens randomly. Sometimes you watch for 5 hours, and nothing happens. Sometimes it happens after 30 minutes.

What is the problem !?

subogero, 2016-01-09:
Which omxd and omxplayer versions are you using?

OMID-313, 2016-01-09:
I haven't checked the versions.
I installed `omxd` almost 3 months ago.

subogero, 2016-01-09:
Please use version 1.6 which I've released just a few days ago. It contains lots of bugfixes.

OMID-313, 2016-01-09:
Shall I remove the older version first?
Or just an update ?

subogero, 2016-01-09:
apt-get update
    apt-get install omxd

OMID-313, 2016-01-09:
Thanks. I'll try it with the newer version.
But one question: had you ever experienced this weird problem?

subogero, 2016-01-09:
Yes, the older version had many similar weird problems.

OMID-313, 2016-01-09:
One more thing:
I don't remember if I had installed `omxd` and `omxplayer` properly or not.
So I want to remove them first, and reinstall them.
Is the following a good approach?

    sudo apt-get remove omxd
    sudo apt-get remove omxplayer
    sudo apt-get update
    sudo apt-get install omxd

Is this OK?
Do I have to install `omxplayer` too !?

subogero, 2016-01-09:
It's not necessary to remove them. `omxplayer` will be installed automatically, because `omxd` depends on it.

OMID-313, 2016-01-14:
I updated the `omxd`.
So far so good.
I will put the system under test, to see if that error has been resolved.
Thanks.

OMID-313, 2016-01-17:
Dear @subogero , this problem happened again today.
I was testing, when I pressed the next button. It happened. Two videos began to play simultaneously. 
I pressed the next button again, they stopped. Again I pressed the next button, the next video started to play normally.
What's the problem ?

subogero, 2016-01-17:
I assume it's version 1.6. But I don't know what's the problem. `omxplayer` sometimes ignores hotkeys, so there is not much I can do.

OMID-313, 2016-01-18:
Maybe the problem is the MiniBian OS (Minimal Raspbian) !?

subogero, 2016-01-18:
I seriously doubt it

OMID-313, 2016-01-20:
Minibian OS is a minimal version of Raspbian. (http://sourceforge.net/projects/minibian/)
If you think the MiniBian OS might cause this problem, then maybe it can be solved by installing dependencies or pre-requisites of `omxd` and `omxplayer`. 

Are there anything you would suggest to install !?

subogero, 2016-01-20:
I'm using Minibian myself.

OMID-313, 2016-01-20:
Well, that's good.
Have you experienced these weird problems with the newer version (omxd 1.6) on Minibian !?

subogero, 2016-01-20:
Not yet. Omxd 1.6 semed to fix it.

OMID-313, 2016-02-07:
This problem happened again today.
The system was under test, when I noticed it has happened. Two videos began to play simultaneously. 
I pressed the next button , they stopped. Again I pressed the next button, the next video started to play normally.

subogero, 2016-02-07:
Are you using omxd in a mission critical production environment?

OMID-313, 2016-02-07:
Well, it is not critical nor serious. It might happen once or twice in a 24/7 operation mode. We can live with that.
I'm just reporting the bugs, hoping to help solve them.

OMID-313, 2016-02-07:
This is the content of `rc.local` : 

    omxd X
    omxd j
    omxd O --no-osd
    omxd A /my/folder

Could this be the cause of this problem !?

subogero, 2016-02-07:
I don't think so. Omxplayer's STDIN handler is very unreliable. Sometimes it ignores commands randomly, especially if you multiple ones at the same time. I can't solve this problem.

closed 2016-02-07 by subogero

OMID-313, 2016-02-08:
You're right @subogero . `omxplayer` needs some serious improvements.

By the way, now that we cannot solve this problem, is it possible to maintain a watchdog-like method to detect when two simultaneous videos begin to play, and run some commands?

Because when this weird thing happens, I solve it by pressing the `next` command twice. Now I was wondering if we could automate this process, we would safely ignore this problem for now.

OMID-313, 2016-05-05:
Dear @subogero ,

Is there any method to establish a watchdog-like method to detect when two simultaneous videos are being played?

OMID-313, 2016-05-05:
Thanks @subogero  for opening a new issue for watchdog problem.
[https://github.com/subogero/omxd/issues/50]

OMID-313, 2016-05-05:
Useful info here :

https://github.com/popcornmix/omxplayer/issues/181

It seems that omxplayer needs serious updates.

OMID-313, 2016-05-09:
Useful info here:

https://github.com/popcornmix/omxplayer/issues/437

This issue has been created recently, and still is open. So we can post log files of failure, crash, and freeze problem of omxplayer, so they can help.

OMID-313, 2016-05-12:
Dear @subogero ,
As stated here (https://github.com/popcornmix/omxplayer/issues/451) by popcornmix and confirmed by some users, it seems that the freeze problem of omxplayer has been solved.
Would you please update the omxd based on this new build of omxplayer?

OMID-313, 2018-01-02:
Today, I was testing something, and found some useful info:

The following script causes video freeze; two videos start to play simultaneously:

```
while true; do
        for f in *.mp4
        do
                mkfifo /tmp/mycmd
                omxplayer $f --no-osd < /tmp/mycmd &
                echo . > /tmp/mycmd
                wait $!
                rm /tmp/mycmd
        done
done
```

However, the following code has no problem:

```
FILES=/home/pi/videos
while true; do
        for f in $FILES
        do
                mkfifo /tmp/mycmd
                omxplayer $f --no-osd < /tmp/mycmd &
                echo . > /tmp/mycmd
                wait $!
                rm /tmp/mycmd
        done
done
```

As the freeze/hang problem was very similar to omxd, I thought maybe this could help.

subogero, 2018-01-11:
You're playing two different sets of files in the two cases. Can you reproduce the difference with the same set of files? Because I don't see any other difference between the two scripts.

As I see `.` restarts the track. But with `echo`, you're sending an additional LF character to `omxplayer`. Have you tried `echo -n . > /tmp/mycmd`?

OMID-313, 2018-01-11:
The following code is working fine, without video freeze, and also without white flicker.

```
mkfifo /tmp/mycmd
sleep 0.5
sudo chmod 777 /tmp/mycmd

while [true]
do
        omxplayer /home/pi/videos/myvideo.mp4 --no-osd -o local --video_fifo 200 < /tmp/mycmd &
        myPID=$!
        echo . > /tmp/mycmd
        sleep 1
        wait $myPID
        wait
done
```

Also, note that firstly omxplayer waits for fifo commands, so `.` does not restart the track; it starts it.
All the commands are sent in the form of `echo -n p > /tmp/mycmd` except the initial `.` which is sent without `-n`.