subogero/omxd #40: Volume level resets for every video

OMID-313, 2016-01-14:
Hi @subogero .
Thanks for this great daemon.

I'm having a trouble using the volume commands:
The `omxd +` and `omxd -` functions increase and decrease the volume to desired level.
But after the current video has finished, and the next video is started, the volume level gets reset to default level.

Is there anyway to restore the volume level for all the videos in the playlist !?

subogero, 2016-01-14:
Unfortunately the volume commands only go to the active omxplayer instance and are not stored by the daemon.

OMID-313, 2016-01-14:
Is there anyway to do this ?
Any help/suggestions?

subogero, 2016-01-14:
Open another issue for "persistent volume" and I'll look into it.

closed 2016-01-14 by subogero