subogero/omxd #41: Persistent Volume

OMID-313, 2016-01-14:
Hi @subogero 
Thanks for this great daemon.

We increase or decrease the volume using `omxd +` or `omxd -` . But the  problem is that when the current video is over and the next video starts, the volume resets and returns to the default level. This happens also on `omxd g num` or `previous` commands. In summary, when the video changes.

Is there any method to store the volume level for all the videos in the playlist?

If possible, it would be also much better if it stores the volume level after shutdown. So on every reboot, one doesn't have to set the initial volume level.

Thanks for your support :)

OMID-313, 2016-01-16:
New news on this issue, dear @subogero !?

subogero, 2016-01-16:
It's very flattering of you to rely on omxd in such a faithful manner. On the other hand it's Saturday, also known as the Shabbat, and I'm somewhat reluctant to respond to work requests in a timely way.

To increase my motivation, you might be interested in donating for the noble cause of omxd. Please find the donate button at the bottom of the blogpost below.

http://linux.subogero.com/1467/omxd-debian-package/

OMID-313, 2016-01-20:
Some thing came to my mind as a solution:
We can define a variable or a file, with default value of 0.
Every time the commands `omxd +` or `omxd -` are executed, this variable increments or decrements.
And when we want to call `omxd` we can call it this way:
`omxd --vol myvar`
Then it can set the volume according to `myvar`.
But there is a problem.
The next videos that are going to be played, have to be started with the updated `myvar`, not the initial value.
Any suggestions?

subogero, 2016-02-03:
`omxplayer` has a command line option `--vol n`. So it's possible to set the initial volume.

Explore how `n` correlates with volume settings with the `+` `-` hotkeys.

When the `omxd` daemon receives a `+` or `-` command, not only forward it to `omxplayer`'s STDIN, but also modify a persistent variable accordingly, and pass it to the next `omxplayer` instance as a command line parameter.

Further complication: the next instance is usually preloaded already, waiting paused for the current track to finish. So when the volume is modified, the next instance needs to be destroyed and restarted with the new volume. Or as an alternative solution, the `+` `-` hotkeys could be forwarded to both the current and next instances. Let's see how a paused `omxplayer` handles that.

Anyway, it's non-trivial. So a donation would definitely boost my motivation, especially now that Iran is becoming a very rich country with the end of the sanctions.

OMID-313, 2016-02-03:
Dear @subogero ,

Thanks for your help and time.
As you mentioned, the difficult part would be handling the preloaded paused `omxd` instance.

I look forward to finding a stable solution.
Thanks.

subogero, 2016-02-03:
Of course the donation is optional. I'll look into it when I have some time.

subogero, 2016-02-07:
*omxplayer investigation results*

The `+-` hotkeys change the volume in 3 dB steps. The `--vol n` command line option, on the other hand, takes `n` in mB (millibel). So it's best to store the persistent volume variable in mB, and change it 300 steps if a `+-` command comes in.

Paused omxplayer instances seem to obey the `+-` commands, so the preloaded instances don't need to be destroyed and reloaded upon volume changes.

OMID-313, 2016-02-07:
Thanks @subogero for your time in this issue.

subogero, 2016-02-07:
Can you compile from source and test?

OMID-313, 2016-02-07:
Shall I use the following commands?

    apt-get update
    apt-get install omxd

subogero, 2016-02-07:
No. There is no new deb package. Pull the new commits into your git repo and then

    make
    sudo make install
    service omxd restart

OMID-313, 2016-02-07:
So, are the following codes correct?

    git clone <https://github.com/subogero/omxd>
    cd omxd
    make
    sudo make install
    service omxd restart

subogero, 2016-02-07:
Yes

subogero, 2016-02-07:
Yes, but it's

    sudo service omxd restart

at the end.

OMID-313, 2016-02-07:
Thanks @subogero for your help.

I tried the following codes:

    sudo apt-get install build-essential git

and then:

    git clone https://github.com/subogero/omxd.git
    cd omxd
    make
    sudo make install
    sudo service omxd restart

It works great. When I decrease/increase the volume, the changes are made to the next video too. This is really good.
But the problem is that it only applies to the current and the next video. For example, if you have 20 videos in your playlist, the rest of the videos do not take the volume. Likewise, when I go to previous video (`omxd N`) it doesn't set the volume. Or, when I go to video number 10 for example, by `omxd g 10` it doesn't take the volume.

subogero, 2016-02-07:
Sorry, I can't reproduce the error. If I do `omxd -`, the volume is reduced forever for all tracks. Are you using the `omxd O <option>` command as well? If so, when?

OMID-313, 2016-02-07:
Oh, sorry.
I was using this code in `rc.local` : 
    omxd O --no-osd --vol -1500
I changed it to : 
    omxd O --no-osd

And now it works great. It keeps the volume for all the videos.
I will test the system again for reliability of this solution.

Thanks a lot for your time and help.

closed 2016-02-07 by subogero

OMID-313, 2016-02-07:
Everything seems fine.
Just one problem:
When I shut down the RPi, and then start it later, the volume level has been set to default.
I mean, I have to set the volume level after every boot of the RPi.
Is it possible to store the volume level, so that it can set the volume level even after shut down !?

subogero, 2016-02-07:
At the moment omxd is not able to store any runtime parameters, not even the position in the playlist. Just use your rc.local

    omxd O --no-osd
    omxd -
    omxd -

OMID-313, 2016-02-07:
Can I use it this way:

    omxd O --no-osd
    omxd O --vol -1500

?
Or do I have to use the following code:

    omxd O --no-osd
    sleep 0.05
    omxd -
    sleep 0.05
    omxd -
    sleep 0.05
    omxd -

?

subogero, 2016-02-07:
You just realized an hour ago that putting `--vol n` into `rc.local` breaks the persistent volume feature.

OMID-313, 2016-02-07:
You're right.
But I thought maybe splitting `--no-osd` and `--vol n` might help solve the problem!
Any way.

OMID-313, 2016-02-07:
Dear @subogero ,
I tried the second method:

    omxd O --no-osd
    sleep 0.1
    omxd -
    sleep 0.1
    omxd -
    sleep 0.1
    omxd -
    sleep 0.1
    omxd -
    sleep 0.1
    omxd -

But unfortunately, after every boot, the videos number 1 and 2 of the playlist begin to play simultaneously.

subogero, 2016-02-07:
Try to sleep some more before the first volume command. E.g. `sleep 2`. If the 2nd omxplayer intance gets the `-` command at the same time that it receives the `p` to pause it, then it ignores both. omxplayer is bloody unreliable in this respect.

OMID-313, 2016-02-08:
I tried `sleep 1` and `sleep 1.5` and `sleep 2` .
1.5 and 2 seems to solve the problem.
I chose 1.5 for now. It is under test.

Thanks a billion for your time and help.