subogero/omxd #42: omxd install script not creating /usr/share/doc/omxd/ directory

priiduonu, 2016-01-19:
root@raspberrypi:~/omxd# make install
./preinst
cp omxd /usr/bin
IFS=''; while read i; do [ "$i" = HELP ] && sed -n '/rpyt/,$p' README; echo "$i"; done <rpyt >/usr/bin/rpyt
chmod +x /usr/bin/rpyt
cp omxd.1 /usr/share/man/man1/
cp rpyt.1 /usr/share/man/man1/
cp init      /usr/share/doc/omxd/
cp: cannot create regular file ‘/usr/share/doc/omxd/’: Not a directory
Makefile:24: recipe for target 'install' failed
make: *** [install] Error 1

root@raspberrypi:~/omxd# grep cp Makefile 
	cp omxd $(DESTDIR)/usr/bin
	cp omxd.1 $(DESTDIR)/usr/share/man/man1/
	cp rpyt.1 $(DESTDIR)/usr/share/man/man1/
	cp init      $(DESTDIR)/usr/share/doc/omxd/
	cp logrotate $(DESTDIR)/usr/share/doc/omxd/
	cp -rt $(REL)/omxd-$(TAG) *; \
	cp -f $(REL)/omxd_$(TAG).tar.gz $(REL)/omxd_$(TAG).orig.tar.gz
	cp preinst postinst prerm postrm $(DEB)
root@raspberrypi:~/omxd# 

because /usr/share/doc/omxd directory has not been created by install script, also the postinst script fails

(omxd cloned from github)

regards,
priiduonu

subogero, 2016-01-19:
Is this the newest Raspbian?

priiduonu, 2016-01-19:
yes, jessie

subogero, 2016-01-19:
The bastards at Debian broke all my projects, because they insisted on changing everything...

subogero, 2016-01-19:
Does the directory `/usr/share/doc/` exist before or after `make install`?

priiduonu, 2016-01-19:
/usr/share/doc exists (it has several subdirectories) but /usr/share/doc/omxd/ doest not exist. I think you should create it yourself.

subogero, 2016-01-19:
OK. But why did it work with Wheezy?

subogero, 2016-01-19:
Or why did it work *ever*? The Makefile is clearly wrong... Anyway, I'll fix it.

priiduonu, 2016-01-19:
I have never used the /etc/init.d/omxd script for starting the player, I only discovered this bug today.

subogero, 2016-01-19:
I realized what the problem is. If you install from the deb package, its `omxd.dirs` file specifies `/usr/share/doc/omxd`, so it's created automatically before the deb package runs `make install`.

closed 2016-01-19 by subogero

priiduonu, 2016-01-19:
thanks and keep up the good work!