subogero/omxd #46: OMXD audio -> '-I' arg

jbriggsnh, 2016-03-14:
Really appreciate omxd. I noticed that on my pi that omxplayer would not send audio through hdmi unless both "-r -o hdmi" flags were set. The '-r' seemed to conflict with '-I'. So I hacked your code to replace the defalts of '-I' with '-r' and it works find.

subogero, 2016-03-14:
Hi. What is `-r`? Unfortunately we need `-I ` because it prints some track info at the beginning. `omxd S` needs it for track length.

jbriggsnh, 2016-03-14:
Its kind of strange - the '-r' forces frame-rate synchronization.  But if
its not there, I can not get audio out of my hdmi-connected TV.

I tried to just hack the option and add "-r -I" - which executes OK but it
causes the command to repeat constantly.

I can just use my local version - just thought I would pass it on as I had
seen some discussion in 2014 about hdmi audio that did not seem to be
resolved.

Thanks again for creating this.

BTW - separate topic, but I am using this for VOD. Its configure to
power-up, mount a nfs drive, and start an apache web server. I have a
simple bash script that does basic navigation and vcr control (thanks to
omxd!). Its intended to be run by the browser in your phone. So it comes
up, displays its IP address to the TV, and you plug that address into your
phone's browser. Everything works OK, but I would like to put a image to
the screen with the ip address in large font. I can use fbi for the image -
do you know what can be used for text? It looks like directFB is no longer
being supported or available.

subogero, 2016-03-17:
You could use figlet or toilet: huge ASCII art numbers:

    nslookup `hostname` | sed -rn 's/Address: //p' | figlet
     _  ___ ____    _  __    ___   _   ____   ___   __   
    / |/ _ \___ \  / |/ /_  ( _ ) / | |___ \ / _ \ / /_  
    | | (_) |__) | | | '_ \ / _ \ | |   __) | | | | '_ \ 
    | |\__, / __/ _| | (_) | (_) || |_ / __/| |_| | (_) |
    |_|  /_/_____(_)_|\___/ \___(_)_(_)_____|\___/ \___/

subogero, 2016-03-17:
BTW what version of the OS and omxplayer are you using?

subogero, 2016-09-30:
No feedback, closed.

closed 2016-09-30 by subogero