subogero/omxd #47: Playing a single file breaks omxd

cacyss0807, 2016-03-31:
Apparently, when playing a single file (5 min video), after looping through it for ~2 times, the screen turns out black.

cacyss0807, 2016-03-31:
It seems that this problem is due to an specific file. I am guessing that omxd is getting the wrong video-length information.

subogero, 2016-04-01:
What does turns out black mean?

cacyss0807, 2016-04-01:
Sorry, I think I was not specific enough. When adding some a `omxd` playlist, after a couple of loops, the videos appear to stop playing, and if I see what `omxd` is currently playing it shows `X/5 <file path>` something like that. Note that, `X` surpasses 5 which shouldn't happen (I guess). When this happens the screen turns black as the video playback has finished, but still `omxd` does not play the next video on the playlist.

If I skip to the next file (`omxd N`), then suddenly the next video starts playing.

subogero, 2016-04-01:
If you have the source, could you post me the output of `make ps` while the screen is black? It prints the entire omxd/omxplayer process hierarchy.

cacyss0807, 2016-04-03:
```
pi@raspberrypi ~/omxd $ make ps
pstree -pu | grep omx
        |-omxd(2409)---omxplayer(7944)---omxplayer.bin(7954)-+-{omxplayer.bin}(7955)
        |                                                    |-{omxplayer.bin}(7956)
        |                                                    |-{omxplayer.bin}(7957)
        |                                                    |-{omxplayer.bin}(7958)
        |                                                    |-{omxplayer.bin}(7959)
        |                                                    |-{omxplayer.bin}(7960)
        |                                                    |-{omxplayer.bin}(7961)
        |                                                    `-{omxplayer.bin}(7962)
```
Also

```
pi@raspberrypi ~/omxd $ omxd S
Playing 1773/3 /home/pi/Downloads/07f0629ab193f78d7dded4849f5f01adda09f71a
```
I am looping throguh a playlist of 2 files. One is 5 seconds long, and the other is 3 seconds long. `omxd` correctly loops thorugh this playlist 3 times. Then, the black screen appears. As you can see from the output of `omxd S`,  `omxd` keeps going even after the end of the video.

subogero, 2016-04-03:
If you're using HDMI audio, then it's `omxplayer` which failed to exit at the end of the video. There is not much I can do.

Unfortunately this does happen a few times, so I might implement a watchdog in `omxd`.

cacyss0807, 2016-04-03:
I see. Would this be corrected if I re-direct the audio to the audio-jack? 
Thank you for your help :+1:

subogero, 2016-04-03:
No. I just asked because in case of audio jack, the omxplayer instance for the next video is already started paused at the beginning. But with HDMI, all we see is the stuck instance for the current track.

cacyss0807, 2016-04-03:
I don't know if it is usefull but

closed 2016-04-03 by cacyss0807

cacyss0807, 2016-04-03:
I was thinking, could it be possible to kill the current instance of `omxplayer` to avoid this? Since, we know the length of the video.

subogero, 2016-04-03:
That's exactly what I mean by watchdog. Currently omxd wakes up only upon user input from the FIFO and the SIGCHLD from exited omxplayers. We'll need to introduce a 3rd time based event to look around and kill stale omxplayers.