subogero/omxd #49: Why I cannot control omxplayer through DBUS when I run omxd as a init service?

tzraeq, 2016-04-03:
When I run a omxd service ,the omxplayer generate the dbus connection but I cannot connect to the session. Is there any other solutions?

subogero, 2016-04-03:
My knowledge of dbus is zero. I can't help you. Even if it worked, it would screw up omxd's status, because stuff would happen with the omxplayer instances outside of omxd's knowledge or control. A absolutely and positively refuse to to deal with Dbus.

tzraeq, 2016-04-03:
If the omxd control the omxplayer through by dbus but not through the the keyboard char,and give some api, this will fixed

tzraeq, 2016-04-03:
omxplayer repo has a dbuscontrol.sh,it is so easy.

subogero, 2016-04-03:
If it's easy then do it yourself. I'm looking forward to your pull request.

tzraeq, 2016-04-03:
I think only the play/pause action will screw up the omxd status.
Or , if you can tell me the method to open the second omxplayer and pause at the beginning,I think I can fix it.

subogero, 2016-04-03:
Do you realise that it's Sunday and I'm not your emloyee?

tzraeq, 2016-04-03:
Sorry , it doesn't means that I want you to reply immediatly.
Have a good weekend!

closed 2016-09-30 by subogero