subogero/omxd #50: Watchdog

subogero, 2016-05-05:
Run a periodic watchdog to detect and kill stale omxplayer instances that have ignored commands on STDIN.

OMID-313, 2016-05-05:
Thanks @subogero for opening a new issue for watchdog problem.
I look hopefully forward to the solution to this video playing abnormalities caused mainly by omxplayer.

OMID-313, 2016-05-05:
Useful info here:
https://github.com/wireload/screenly-ose/issues/21
They seem to have the same problem with omxplayer :D

OMID-313, 2016-05-09:
Hi @subogero ,
I understand that you're too busy.
I just logged in to say keep up the good work. This omxd is great. Never give up.
Thanks man.

OMID-313, 2016-05-12:
New news on this watch-dog method?
@subogero

OMID-313, 2016-06-05:
I was hoping that the freeze problem of `omxplayer` had been solved via the raspberry-pi framework upgrade. But unfortunately, today when I was testing the system, the freeze problem happened again between the videos. Also, it caused the tv to show **_No HDMI Signal_** notification.

I had to use **_Next_** (`omxd n`) to un-freeze the display.

I would really really appreciate if you design this watchdog method to avoid these `omxplayer` crazy anomalies.

subogero, 2016-06-05:
I'm currently working on the watchdog.
On the other hand, the newest few omxplayer versions are totally crap. I've installed an old version manually. As you can see it's one year old!

    apt-get update
    apt-get install --reinstall omxplayer=0.3.6~git20150627~843744e

OMID-313, 2016-06-05:
Thanks @subogero for keeping the good work.

You mean the older version works better and has less bugs (white flicker between videos, video freeze, etc.) !?

subogero, 2016-06-05:
The old version does not freeze at the end.

OMID-313, 2016-06-05:
OK.
I'll try the older version.
Do I have to update the repository or the list before installing the older version?

subogero, 2016-06-05:
The bloody instructions are 6 lines above. Come on!!!

OMID-313, 2016-06-05:
:yum: 
I meant editing the repository files **_before_** running those commands.

Anyway, the omxplayer _downdated_ (!!) to the older version.
Thanks.

Waiting enthusiastically to enable the watchdog method, so as to get rid of the omxplayer freeze problem.

subogero, 2016-06-07:
It seems that a newer version of omxplayer `0.3.6~git20160102~f544084` also does not freeze at the end of videos. But the newest one is bad.

OMID-313, 2016-06-07:
Thanks @subogero for your info. I'll give it a try.

By the way, what do you mean by _does not freeze **at the end** of videos_ ?
The `omxplayer` freezes the videos randomly whenever it likes! Beginning, middle, end, ...
Am I wrong?

subogero, 2016-06-07:
I only saw omxplayer 3.7-xxx hanging at the end of the video.

subogero, 2016-06-14:
Please test `wd` branch.

    git fetch
    git checkout wd
    make
    sudo make stop
    sudo make install
    sudo make start

OMID-313, 2016-06-14:
Thanks @subogero for this `omxwd` .
I'll try it.

Just one question:
If omxplayer freezes, what does this `wd` do exactly? Does it switch to the next video?
(when it freezes, I manually do `omxd n` twice, so it solves the freeze and switched to the next video.)

subogero, 2016-06-14:
Just click the last commit above and read the commit message.

subogero, 2016-06-14:
Please use newest commit, it fixes client/server segfault.

    git pull origin wd
    make
    ...

OMID-313, 2016-06-15:
Are the following process OK?

```
git fetch
git checkout wd
git pull origin wd
make
sudo make stop
sudo make install
sudo make start
```

OMID-313, 2016-06-15:
I did the following:

```
git fetch
git pull origin wd
make
sudo make stop
sudo make install
sudo make start

reboot
```

And then I tested the system.

In order to reproduce the freeze conditions, I do omxd n several times very fast!
It freezes, then No HDMI Input comes to the screen, then after 10-20 seconds, a video comes to the screen from its middle frames (not its beginning).

I think the watchdog would be more applicable and useful if running in the background every 1 or 2 seconds.

subogero, 2016-06-16:
Please test new code again. Watchdog period down to 5s. But I don't want to run it too often...

And a very significant bug fixed, listening to internet radios should not bring down the entire system any more.

subogero, 2016-06-16:
Please note, the watchdog does not shut down omxplayers whose track length has not elapsed. If omxplayer gets stuck in the middle, the watchdog will wait until the supposed end of the track.

`omxd` has absolutely no way to detect omxplayers stuck in the middle.

OMID-313, 2016-06-19:
Thanks for the bug fix and the update @subogero .

Is there any other way to reproduce the freeze situation?
(Else than doing `omxd n` several times very fast !!)

subogero, 2016-06-19:
I'm still working on improving the logging, reproducing your use case, and a possible race condition, indicated by leftover `omxplayer.log.xxxx` files in `/var/log`.

subogero, 2016-06-19:
Huge breakthrough! See my last commit. Could you please test it again with very frequent `omxd n` commands?

OMID-313, 2016-06-19:
WOWWWWW !!!

Thank you man for this huge breakthrough.
I tested doing `omxd n` as fast as I could :D 
But the player was unbelievably robust and reliable.
That's PERFECT. Thanks a zillion.

Right now, the system is under test to see if random freeze case occurs or not.

OMID-313, 2016-06-19:
I couldn't bother the system with the `omxd n`, even with ultra fast hits !!

But the `omxd N` causes some freezes when hit fast.

subogero, 2016-06-19:
Yes, `omxd N` still needs some work.

subogero, 2016-06-19:
The `N` command generates twice as many killed and new omxplayer instances. There was a race condition where the `omxwd` watchdog wanted to kill the `omxplayer.bin` child of `omxplayer` script with PID X, but the latter has not yet started the former. So they get missed and stay alive.

Fix: When a command wants to replace both `omxplayer` instances, like N, use `stop_all()` to start the watchdog to kill ALL `omxplayer.bin` instances.

OMID-313, 2016-06-19:
Thanks for your support @subogero .
Keep up the great work.

subogero, 2016-06-19:
See the last commit. It improved my `omxd N` loop a lot.

OMID-313, 2016-06-19:
Unfortunately, in my case, it corrupted both `omxd n` and `omxd N`
Normal fast hits corrupt both next and previous commands.

subogero, 2016-06-19:
What does "corrupted" mean? Can you post the last 50 lines from your `/var/log/omxlog`?

OMID-313, 2016-06-20:
`/var/log/omxlog` :

```
1466331916 daemonize: omxd started, SID 402
1466331916 main: X
1466331916 main: j
1466331916 main: O --blank
1466331916 main: .
1466331916 main: O --no-osd
1466331916 main: .
1466331916 main: A /usermedias/AA.mp4
1466331916 player_new: PID=412 /usermedias/AA.mp4
1466331916 main: A /usermedias/BB.mp4
1466331916 player_new: PID=413 /usermedias/BB.mp4
1466331916 main: A /usermedias/CC.mp4
1466331916 main: A /usermedias/DD.mp4
1466331916 main: A /usermedias/EE.mp4
1466331916 main: A /usermedias/FF.mp4
1466331916 main: A /usermedias/GG.mp4
1466331916 main: A /usermedias/HH.mp4
1466331916 main: A /usermedias/II.mp4
1466331916 main: A /usermedias/JJ.mp4
1466331916 main: A /usermedias/KK.mp4
1466331916 main: A /usermedias/LL.mp4
1466331916 main: A /usermedias/MM.mp4
1466331916 main: A /usermedias/NN.mp4
1466331916 main: A /usermedias/OO.mp4
1466331916 main: A /usermedias/PP.mp4
1466331932 player_quit: PID=412 with 0
1466331932 player_cmd: Send p to omxplayer PID/fd 413/4
1466331932 player_new: PID=542 /usermedias/CC.mp4
1466331960 player_quit: PID=413 with 0
1466331960 player_cmd: Send p to omxplayer PID/fd 542/1
1466331960 player_new: PID=565 /usermedias/DD.mp4
1466331994 player_quit: PID=542 with 0
1466331994 player_cmd: Send p to omxplayer PID/fd 565/3
1466331994 player_new: PID=586 /usermedias/EE.mp4
1466332034 player_quit: PID=565 with 0
1466332034 player_cmd: Send p to omxplayer PID/fd 586/1
1466332034 player_new: PID=604 /usermedias/FF.mp4
1466332072 player_quit: PID=586 with 0
1466332072 player_cmd: Send p to omxplayer PID/fd 604/3
```

OMID-313, 2016-06-20:
A new problem is occurring!
After reboot, it does not load the videos, or load them frozen!
And when I SSH into it via putty, and do `omxd n` or `omxd P`, it says: 
```
Opening FIFO '/var/run/omxctl' timed out.
The omxd daemon is dead or listening on another one.
An unprivileged omxd listens on omxctl in its current dir.
One started as root listens on '/var/run/omxctl'.
```

I think the last commit was not successful, and better to return to the _huge-breakthrough_ commit.

OMID-313, 2016-06-20:
Sorry. The above log file was the beginning lines!
The following are the last lines of `/var/log/omxlog` for the last 3-4 RPi boots/reboots:

```
1466341436 daemonize: omxd started, SID 334
1466341440 daemonize: omxd started, SID 401
1466341440 main: X
1466341440 main: j
1466341440 main: O --blank
1466341440 main: .
1466341440 main: O --no-osd
1466341440 main: .
1466341440 main: A /usermedias/AA.mp4
1466341440 player_new: PID=412 /usermedias/AA.mp4
1466341440 main: A /usermedias/BB.mp4
1466341440 player_new: PID=413 /usermedias/BB.mp4
1466341440 main: A /usermedias/CC.mp4
1466341440 main: A /usermedias/DD.mp4
1466341440 main: A /usermedias/EE.mp4
1466341440 main: A /usermedias/FF.mp4
1466341440 main: A /usermedias/GG.mp4
1466341440 main: A /usermedias/HH.mp4
1466341440 main: A /usermedias/II.mp4
1466341440 main: A /usermedias/JJ.mp4
1466341440 main: A /usermedias/KK.mp4
1466341440 main: A /usermedias/LL.mp4
1466341440 main: A /usermedias/MM.mp4
1466341440 main: A /usermedias/NN.mp4
1466341440 main: A /usermedias/OO.mp4
1466341440 main: A /usermedias/PP.mp4
1466341456 player_quit: PID=412 with 0
1466341456 player_cmd: Send p to omxplayer PID/fd 413/4
1466341456 player_new: PID=543 /usermedias/CC.mp4
1466341484 player_quit: PID=413 with 0
1466341484 player_cmd: Send p to omxplayer PID/fd 543/1
1466341484 player_new: PID=581 /usermedias/DD.mp4
1466341517 player_quit: PID=543 with 0
1466341517 player_cmd: Send p to omxplayer PID/fd 581/3
1466341517 player_new: PID=601 /usermedias/EE.mp4
1466341522 main: n
1466341522 player_off: PID 581
1466341522 omxwd: kill omxplayer.bin 591 of omxplayer 581
1466341522 player_cmd: Send p to omxplayer PID/fd 601/1
1466341522 player_new: PID=626 /usermedias/FF.mp4
1466341522 player_quit: PID=581 with 137
1466341524 main: n
1466341524 player_off: PID 601
1466341524 omxwd: kill omxplayer.bin 609 of omxplayer 601
1466341524 player_cmd: Send p to omxplayer PID/fd 626/4
1466341524 player_new: PID=649 /usermedias/GG.mp4
1466341524 player_quit: PID=601 with 137
1466341526 main: n
1466341526 player_off: PID 626
1466341526 omxwd: kill omxplayer.bin 634 of omxplayer 626
1466341526 player_cmd: Send p to omxplayer PID/fd 649/3
1466341526 player_new: PID=676 /usermedias/HH.mp4
1466341526 player_quit: PID=626 with 137
1466341526 main: n
1466341526 player_off: PID 649
1466341526 omxwd: kill omxplayer.bin 657 of omxplayer 649
1466341526 player_cmd: Send p to omxplayer PID/fd 676/4
1466341526 player_new: PID=692 /usermedias/II.mp4
1466341526 player_quit: PID=649 with 137
1466341530 main: n
1466341530 player_off: PID 676
1466341530 omxwd: kill omxplayer.bin 684 of omxplayer 676
1466341530 player_cmd: Send p to omxplayer PID/fd 692/3
1466341530 player_new: PID=726 /usermedias/JJ.mp4
1466341530 player_quit: PID=676 with 137
1466341530 main: n
[... 3614 lines cut ...]
1466395302 player_new: PID=1165 /usermedias/KK.mp4
1466395302 main: A /usermedias/LL.mp4
1466395302 player_off: PID 1165
1466395302 player_new: PID=1183 /usermedias/LL.mp4
1466395302 main: A /usermedias/MM.mp4
1466395302 player_off: PID 1183
1466395302 player_new: PID=1220 /usermedias/MM.mp4
1466395302 main: A /usermedias/NN.mp4
1466395302 player_off: PID 1220
1466395303 player_new: PID=1243 /usermedias/NN.mp4
1466395303 main: A /usermedias/OO.mp4
1466395303 player_off: PID 1243
1466395303 player_new: PID=1301 /usermedias/OO.mp4
1466395303 main: A /usermedias/PP.mp4
1466395303 player_off: PID 1301
1466395303 player_quit: PID=1118 with 134
1466395303 player_new: PID=1320 /usermedias/PP.mp4
1466395303 player_quit: PID=1141 with 134
1466395304 player_quit: PID=1165 with 134
1466395304 player_quit: PID=1183 with 134
1466395304 player_quit: PID=1243 with 134
1466395304 player_quit: PID=1220 with 134
1466395305 player_quit: PID=1301 with 134
1466395305 player_quit: PID=1320 with 134
```

OMID-313, 2016-06-20:
Right now, on each reboot, the screen freezes at the very beginning, then a `No HDMI signal` comes to the screen, then after 1 minute, the screen shows 3 or maybe 4 videos playing simultaneously, then again no hdmi signal, then 1 video in the normal condition, then screen gets blank forever. Then I did `omxd P` .

The following are the log files that are added to `/var/log/omxlog` due to this new scenario:

```
1466399198 daemonize: omxd started, SID 336
1466399202 daemonize: omxd started, SID 401
1466399202 main: X
1466399202 main: j
1466399202 player_killall
1466399202 omxwd: killing all
1466399202 main: O --blank
1466399202 main: .
1466399202 player_killall
1466399202 omxwd: killing all
1466399202 main: O --no-osd
1466399202 main: .
1466399202 player_killall
1466399202 omxwd: killing all
1466399202 main: A /usermedias/AA.mp4
1466399202 player_new: PID=429 /usermedias/AA.mp4
1466399202 main: A /usermedias/BB.mp4
1466399202 player_off: PID 429
1466399203 player_new: PID=437 /usermedias/BB.mp4
1466399203 main: A /usermedias/CC.mp4
1466399203 player_off: PID 437
1466399203 player_new: PID=460 /usermedias/CC.mp4
1466399203 main: A /usermedias/DD.mp4
1466399203 player_off: PID 460
1466399203 player_new: PID=474 /usermedias/DD.mp4
1466399203 main: A /usermedias/EE.mp4
1466399203 player_off: PID 474
1466399203 player_new: PID=490 /usermedias/EE.mp4
1466399203 main: A /usermedias/FF.mp4
1466399203 player_off: PID 490
1466399203 player_new: PID=507 /usermedias/FF.mp4
1466399203 main: A /usermedias/GG.mp4
1466399203 player_off: PID 507
1466399203 player_new: PID=527 /usermedias/GG.mp4
1466399203 main: A /usermedias/HH.mp4
1466399203 player_off: PID 527
1466399203 player_new: PID=542 /usermedias/HH.mp4
1466399203 main: A /usermedias/II.mp4
1466399203 player_off: PID 542
1466399203 player_new: PID=558 /usermedias/II.mp4
1466399203 main: A /usermedias/JJ.mp4
1466399203 player_off: PID 558
1466399203 player_new: PID=577 /usermedias/JJ.mp4
1466399203 main: A /usermedias/KK.mp4
1466399203 player_off: PID 577
1466399203 player_new: PID=590 /usermedias/KK.mp4
1466399203 main: A /usermedias/LL.mp4
1466399203 player_off: PID 590
1466399204 player_new: PID=610 /usermedias/LL.mp4
1466399204 main: A /usermedias/MM.mp4
1466399204 player_off: PID 610
1466399204 player_new: PID=629 /usermedias/MM.mp4
1466399204 main: A /usermedias/NN.mp4
1466399204 player_off: PID 629
1466399204 player_quit: PID=577 with 134
1466399204 player_new: PID=708 /usermedias/NN.mp4
1466399204 main: A /usermedias/OO.mp4
1466399204 player_off: PID 708
1466399205 player_new: PID=724 /usermedias/OO.mp4
1466399205 main: A /usermedias/PP.mp4
1466399205 player_off: PID 724
1466399205 player_quit: PID=542 with 134
1466399205 player_new: PID=753 /usermedias/PP.mp4
1466399205 player_quit: PID=590 with 134
1466399206 player_quit: PID=610 with 134
1466399206 player_quit: PID=629 with 134
1466399206 player_quit: PID=708 with 134
1466399206 player_quit: PID=724 with 134
1466399273 player_quit: PID=429 with 0
1466399303 player_quit: PID=437 with 0
1466399345 player_quit: PID=460 with 0
1466399354 player_quit: PID=490 with 0
1466399355 player_quit: PID=558 with 0
1466399361 player_quit: PID=474 with 0
1466399367 player_quit: PID=507 with 0
1466399384 player_quit: PID=527 with 0
1466399683 main: P
1466399683 player_killall
1466399683 omxwd: killing all
1466399684 player_quit: PID=753 with 134
```

subogero, 2016-06-20:
Yes, this is more like it. In my experience on a RPi 2 board, if I wait 0.2s between the `omxd N` commands, there is no problem any more. If there is no delay between the commands, it still gets messed up, BUT a few `P` and `n` commands completely clean up the zombies.

Do you think these rapid-fire n/N commands are a realistic use case? You can go to any track directly with something like `omxd g 8`...

OMID-313, 2016-06-20:
Actually, I think the previous _huge-breakthrough_ commit was perfect.
And I think up to 0.5 s delays (or maybe even 1 s if needed) between `n` or `N` commands are really OK in realistic use cases.

subogero, 2016-06-20:
Yes, I went back to that commit too and deleted the last one. Let's leave it there for now.

OMID-313, 2016-06-20:
Can I get back to the previous commit with the following codes?

```
git fetch
git pull origin wd
make
sudo make stop
sudo make install
sudo make start

reboot
```

subogero, 2016-06-20:
Yes, but replace the second step with

    git reset --hard origin/wd

OMID-313, 2016-06-20:
I did the following:

```
git fetch
git reset --hard origin/wd
make
sudo make stop
sudo make install
sudo make start

reboot
```

But unfortunately, again it freezes at the beginning.
Do I make any mistakes?

subogero, 2016-06-20:
No. It seems systemd does not start it on boot after the manual install. Just start it manually for now after the reboot with

    systemctl start omxd

I'll try to sort out the Makefile to support testing better.

OMID-313, 2016-06-20:
But I start it manually on boot every time.

See my codes here:
https://github.com/subogero/omxd/issues/54#issue-158541670

OMID-313, 2016-06-20:
These are the content added to `/var/log/omxlog` after reboot:

```
1466426744 daemonize: omxd started, SID 333
1466426748 daemonize: omxd started, SID 400
1466426748 main: X
1466426748 main: j
1466426748 main: O --blank
1466426748 main: .
1466426748 main: O --no-osd
1466426748 main: .
1466426748 main: A /usermedias/AA.mp4
1466426748 player_new: PID=408 /usermedias/AA.mp4
1466426748 main: A /usermedias/BB.mp4
1466426748 player_off: PID 408
1466426748 player_new: PID=422 /usermedias/BB.mp4
1466426748 main: A /usermedias/CC.mp4
1466426748 player_off: PID 422
1466426749 player_new: PID=438 /usermedias/CC.mp4
1466426749 main: A /usermedias/DD.mp4
1466426749 player_off: PID 438
1466426749 player_new: PID=455 /usermedias/DD.mp4
1466426749 main: A /usermedias/EE.mp4
1466426749 player_off: PID 455
1466426749 player_new: PID=472 /usermedias/EE.mp4
1466426749 main: A /usermedias/FF.mp4
1466426749 player_off: PID 472
1466426749 player_new: PID=493 /usermedias/FF.mp4
1466426749 main: A /usermedias/GG.mp4
1466426749 player_off: PID 493
1466426749 player_new: PID=505 /usermedias/GG.mp4
1466426749 main: A /usermedias/HH.mp4
1466426749 player_off: PID 505
1466426749 player_new: PID=522 /usermedias/HH.mp4
1466426749 main: A /usermedias/II.mp4
1466426749 player_off: PID 522
1466426749 player_new: PID=540 /usermedias/II.mp4
1466426749 main: A /usermedias/JJ.mp4
1466426749 player_off: PID 540
1466426749 player_new: PID=557 /usermedias/JJ.mp4
1466426749 main: A /usermedias/KK.mp4
1466426749 player_off: PID 557
1466426749 player_new: PID=577 /usermedias/KK.mp4
1466426749 main: A /usermedias/LL.mp4
1466426749 player_off: PID 577
1466426750 player_new: PID=590 /usermedias/LL.mp4
1466426750 main: A /usermedias/MM.mp4
1466426750 player_off: PID 590
1466426750 player_new: PID=612 /usermedias/MM.mp4
1466426750 main: A /usermedias/NN.mp4
1466426750 player_off: PID 612
1466426750 player_new: PID=688 /usermedias/NN.mp4
1466426750 main: A /usermedias/OO.mp4
1466426750 player_off: PID 688
1466426751 player_quit: PID=493 with 134
1466426751 player_new: PID=719 /usermedias/OO.mp4
1466426751 main: A /usermedias/PP.mp4
1466426751 player_off: PID 719
1466426751 player_quit: PID=557 with 134
1466426751 player_new: PID=732 /usermedias/PP.mp4
1466426752 player_quit: PID=577 with 134
1466426752 player_quit: PID=590 with 134
1466426752 player_quit: PID=612 with 134
1466426752 player_quit: PID=688 with 134
1466426752 player_quit: PID=719 with 134
1466426819 player_quit: PID=408 with 0
1466426848 player_quit: PID=422 with 0
1466426869 player_quit: PID=522 with 0
1466426886 player_quit: PID=438 with 0
1466426895 player_quit: PID=472 with 0
1466426895 player_quit: PID=540 with 0
1466426899 player_quit: PID=455 with 0
1466426922 player_quit: PID=505 with 0
```

subogero, 2016-06-20:
I don understand. The log clearly shows that systemd starts it on reboot automatically, and also that you restart it 4s later unnecessarily. And then it's running. What is the problem?

OMID-313, 2016-06-20:
This is what happens now:
Right now, on each reboot, the screen freezes at the very beginning, then a No HDMI signal comes to the screen, then after 1 minute, the screen shows 3 or maybe 4 videos playing simultaneously, then again no hdmi signal, then 1 video in the normal condition, then screen gets blank forever.

OMID-313, 2016-06-20:
This time, after boot, the screen just went black for several minutes.
Then I did SSH into it via putty, and typed `omxd n` .It said:
```
root@raspberrypi:~# omxd n

Opening FIFO '/var/run/omxctl' timed out.
The omxd daemon is dead or listening on another one.
An unprivileged omxd listens on omxctl in its current dir.
One started as root listens on '/var/run/omxctl'.
```

I think the installation process, and also the process of returning to the previous commit had some mistakes.

OMID-313, 2016-06-20:
Just a few seconds after boot (when again the screen is black), I typed `service omxd status` , it says:

```
● omxd.service - omxd media player daemon
   Loaded: loaded (/usr/share/doc/omxd/omxd.service; enabled)
   Active: failed (Result: signal) since Mon 2016-06-20 14:38:30 BST; 16s ago
  Process: 401 ExecStart=/usr/bin/omxd (code=exited, status=0/SUCCESS)
 Main PID: 402 (code=killed, signal=FPE)

Jun 20 14:38:25 raspberrypi omxd[401]: omxd daemon started, PID 402
Jun 20 14:38:29 raspberrypi systemd[1]: omxd.service: main process exited, code=killed, status=8/FPE
Jun 20 14:38:30 raspberrypi systemd[1]: Unit omxd.service entered failed state.
```

subogero, 2016-06-20:
This seems to be a RPi2. And now I'm quite sure I'm doing a division by zero somewhere, because I don't use floating point numbers at all. I'll have a look in the evening.

OMID-313, 2016-06-20:
Actually, this is RPi 3 with the latest framework, and I'm using Minibian upgraded to Jessie.

Thanks for your support.

subogero, 2016-06-21:
I've started testing with videos instead of mp3s, it's much more sensitive, I guess the videos cause much higher processor load.

I've reproduced the flickering screen when more videos are played at once. But the last commit improved it in a big way. The watchdog retries killing the target `omxplayer.bin` if not found the first time.

On the other hand, I could not reproduce the SIGFTE.

OMID-313, 2016-06-22:
I updated to the latest commit via:

```
git fetch
git pull origin wd
make
sudo make stop
sudo make install
sudo make start

reboot
```

After reboot, one of the videos (KK.mp4) which is the **_last_**(!!!) video in the playlist directory begins to play, then a white flicker, then the screen gets black, and it doesn't play any other videos.

The following are the status of omxd service, both during playing that video, and also after finishing that video and getting to the black screen:

```
root@raspberrypi:~# service omxd status

● omxd.service - omxd media player daemon
   Loaded: loaded (/usr/share/doc/omxd/omxd.service; enabled)
   Active: active (running) since Wed 2016-06-22 10:57:10 BST; 26s ago
  Process: 401 ExecStart=/usr/bin/omxd (code=exited, status=0/SUCCESS)
 Main PID: 402 (omxd)
   CGroup: /system.slice/omxd.service
           ├─402 /usr/bin/omxd
           ├─664 /bin/bash /usr/bin/omxplayer -I --vol 0 --blank --no-osd -olocal /usermedias/KK.mp4
           └─674 /usr/bin/omxplayer.bin -I --vol 0 --blank --no-osd -olocal /usermedias/KK.mp4

Jun 22 10:57:10 raspberrypi omxd[401]: omxd daemon started, PID 402
```

and

```
root@raspberrypi:~# service omxd status

● omxd.service - omxd media player daemon
   Loaded: loaded (/usr/share/doc/omxd/omxd.service; enabled)
   Active: failed (Result: signal) since Wed 2016-06-22 10:59:08 BST; 4s ago
  Process: 401 ExecStart=/usr/bin/omxd (code=exited, status=0/SUCCESS)
 Main PID: 402 (code=killed, signal=FPE)

Jun 22 10:57:10 raspberrypi omxd[401]: omxd daemon started, PID 402
Jun 22 10:59:08 raspberrypi systemd[1]: omxd.service: main process exited, code=killed, status=8/FPE
Jun 22 10:59:08 raspberrypi systemd[1]: Unit omxd.service entered failed state.
```

and the following are the last lines of `/var/log/omxlog` : 

```
...
...
1466589433 main: A /usermedias/JJ.mp4
1466589433 player_off: PID 622
1466589433 omxwd: kill omxplayer.bin 639 of omxplayer 622
1466589433 player_quit: PID=600 with 137
1466589433 player_new: PID=643 /usermedias/JJ.mp4
1466589433 main: A /usermedias/KK.mp4
1466589433 player_off: PID 643
1466589433 omxwd: kill omxplayer.bin 660 of omxplayer 643
1466589433 player_quit: PID=622 with 137
1466589433 player_new: PID=664 /usermedias/KK.mp4
1466589433 player_quit: PID=643 with 137
1466589548 player_quit: PID=664 with 0
```

OMID-313, 2016-06-22:
The same scenario happens when I run the script manually:

`. /path/to/myscript.sh`

subogero, 2016-06-22:
Please post the log from the last `daemonize` line.

OMID-313, 2016-06-23:
The following are the logs added to `/var/log/omxlog` after reboot:

```
1466590933 daemonize: omxd started, SID 334
1466590937 daemonize: omxd started, SID 400
1466590937 main: X
1466590937 main: j
1466590937 main: O --blank
1466590937 main: .
1466590937 main: O --no-osd
1466590937 main: .
1466590937 main: A /usermedias/AA.mp4
1466590937 player_new: PID=408 /usermedias/AA.mp4
1466590937 main: A /usermedias/AAB.mp4
1466590937 player_off: PID 408
1466590937 omxwd: kill omxplayer.bin 426 of omxplayer 408
1466590938 player_new: PID=430 /usermedias/AAB.mp4
1466590938 main: A /usermedias/AAC.mp4
1466590938 player_off: PID 430
1466590938 omxwd: kill omxplayer.bin 447 of omxplayer 430
1466590938 player_quit: PID=408 with 137
1466590938 player_new: PID=451 /usermedias/AAC.mp4
1466590938 main: A /usermedias/AAD.mp4
1466590938 player_off: PID 451
1466590938 omxwd: kill omxplayer.bin 468 of omxplayer 451
1466590938 player_quit: PID=430 with 137
1466590938 player_new: PID=472 /usermedias/AAD.mp4
1466590938 main: A /usermedias/BB.mp4
1466590938 player_off: PID 472
1466590938 omxwd: kill omxplayer.bin 489 of omxplayer 472
1466590938 player_quit: PID=451 with 137
1466590938 player_new: PID=493 /usermedias/BB.mp4
1466590938 main: A /usermedias/DD.mp4
1466590938 player_off: PID 493
1466590938 omxwd: kill omxplayer.bin 510 of omxplayer 493
1466590938 player_quit: PID=472 with 137
1466590938 player_new: PID=514 /usermedias/DD.mp4
1466590938 main: A /usermedias/EE.mp4
1466590938 player_off: PID 514
1466590938 omxwd: kill omxplayer.bin 531 of omxplayer 514
1466590939 player_quit: PID=493 with 137
1466590939 player_new: PID=535 /usermedias/EE.mp4
1466590939 main: A /usermedias/FF.mp4
1466590939 player_off: PID 535
1466590939 omxwd: kill omxplayer.bin 552 of omxplayer 535
1466590939 player_quit: PID=514 with 137
1466590939 player_new: PID=556 /usermedias/FF.mp4
1466590939 main: A /usermedias/GG.mp4
1466590939 player_off: PID 556
1466590939 omxwd: kill omxplayer.bin 573 of omxplayer 556
1466590939 player_quit: PID=535 with 137
1466590939 player_new: PID=577 /usermedias/GG.mp4
1466590939 main: A /usermedias/HH.mp4
1466590939 player_off: PID 577
1466590939 omxwd: kill omxplayer.bin 594 of omxplayer 577
1466590939 player_quit: PID=556 with 137
1466590939 player_new: PID=598 /usermedias/HH.mp4
1466590939 main: A /usermedias/II.mp4
1466590939 player_off: PID 598
1466590939 omxwd: kill omxplayer.bin 616 of omxplayer 598
1466590939 player_quit: PID=577 with 137
1466590939 player_new: PID=620 /usermedias/II.mp4
1466590939 main: A /usermedias/JJ.mp4
1466590939 player_off: PID 620
1466590939 omxwd: kill omxplayer.bin 637 of omxplayer 620
1466590940 player_quit: PID=598 with 137
1466590940 player_new: PID=641 /usermedias/JJ.mp4
1466590940 main: A /usermedias/KK.mp4
1466590940 player_off: PID 641
1466590940 omxwd: kill omxplayer.bin 658 of omxplayer 641
1466590940 player_quit: PID=620 with 137
1466590940 player_new: PID=662 /usermedias/KK.mp4
1466590940 player_quit: PID=641 with 137
1466657982 player_quit: PID=662 with 0
```

subogero, 2016-06-23:
Your `omxd A` works differently than mine. Yours always destroys the current track as if it was doing an implicit `n`.  But it should not. The very first `A` should start the first track, the second one should start the next track paused, and from then on they should just append to the playlist and leave the omxplayers alone. This is what mine does.

Are you sure you're running the same code? What are the outputs of the following three commands?

    make ps
    git status
    git rev-parse HEAD

OMID-313, 2016-06-23:
```
root@raspberrypi:~# cd omxd

root@raspberrypi:~/omxd# make ps

pstree -pu | grep omx
bash: pstree: command not found
Makefile:57: recipe for target 'ps' failed
make: *** [ps] Error 1

root@raspberrypi:~/omxd# git status

On branch wd
Your branch is up-to-date with 'origin/wd'.
nothing to commit, working directory clean

root@raspberrypi:~/omxd# git rev-parse HEAD

496cbec88982e1ab4276a9702f602c0f9f9a9529
```

I think I'd better re-write my microSD card with my stable backup, then update omxd to the latest commit.

OMID-313, 2016-06-23:
I wrote the latest backup to microSD, and then updated the omxd to the latest commit as the following:

```
root@raspberrypi:~# cd omxd

root@raspberrypi:~/omxd# ls

client.c  init       Makefile  omxd.1  omxd_help.h  postrm   README         utils.c
COPYING   INSTALL    m_list.c  omxd.c  player.c     preinst  test_m_list.c  version.h
HACKING   logrotate  omxd      omxd.h  postinst     prerm    test_utils.c   version.txt

root@raspberrypi:~/omxd# git fetch

remote: Counting objects: 72, done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 72 (delta 15), reused 15 (delta 15), pack-reused 54
Unpacking objects: 100% (72/72), done.
From https://github.com/subogero/omxd
   420ea4f..a32d806  master     -> origin/master
 * [new branch]      wd         -> origin/wd
 * [new tag]         1.7        -> 1.7

root@raspberrypi:~/omxd# git pull origin wd

From https://github.com/subogero/omxd
 * branch            wd         -> FETCH_HEAD
Updating 420ea4f..496cbec
Fast-forward
 Makefile     |   6 +++++-
 client.c     |  73 +++++++++++++++++++++++++++++++++++++++++++++++----------------------
 m_list.c     |  10 ++++++----
 omxd.c       |  27 +++++++++++++++++---------
 omxd.h       |   2 ++
 omxd.service |  11 +++++++++++
 omxwd        |  45 +++++++++++++++++++++++++++++++++++++++++++
 player.c     | 110 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++----------------------------
 postinst     |  15 +++++++++++----
 postrm       |  13 ++++++++++---
 preinst      |   2 ++
 version.txt  |   2 +-
 12 files changed, 242 insertions(+), 74 deletions(-)
 create mode 100644 omxd.service
 create mode 100755 omxwd

root@raspberrypi:~/omxd# make

sed -rn '1,/^\.$/ s/^(.*)$/"\1\\n"/p' README >omxd_help.h
gcc -g -o omxd omxd.c client.c player.c utils.c m_list.c

root@raspberrypi:~/omxd# make stop

./prerm
./postrm
Running omxd postrm
572
Synchronizing state for omxd.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d omxd defaults
insserv: warning: current start runlevel(s) (empty) of script `omxd' overrides LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `omxd' overrides LSB defaults (0 1 6).
Executing /usr/sbin/update-rc.d omxd disable
insserv: warning: current start runlevel(s) (empty) of script `omxd' overrides LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `omxd' overrides LSB defaults (0 1 6).
Removed symlink /etc/systemd/system/omxd.service.
killall omxd
bash: killall: command not found
Makefile:41: recipe for target 'stop' failed
make: [stop] Error 127 (ignored)
killall omxplayer.bin
bash: killall: command not found
Makefile:41: recipe for target 'stop' failed
make: [stop] Error 127 (ignored)

root@raspberrypi:~/omxd# make install

./preinst
Running omxd preinst
Unknown operation 'omxd'.
cp omxd /usr/bin
cp omxd.1 /usr/share/man/man1/
mkdir -p /usr/share/doc/omxd
cp init /usr/share/doc/omxd/
cp omxd.service /usr/share/doc/omxd/
cp logrotate /usr/share/doc/omxd/
cp omxwd /usr/bin
perl -lne 'print unless /^omxd$/' -i /etc/rc.local # Auto migrate from rc.local

root@raspberrypi:~/omxd# make start

./postinst
Running omxd postinst
572
Created symlink from /etc/systemd/system/multi-user.target.wants/omxd.service to /usr/share/doc/omxd/omxd.service.
Created symlink from /etc/systemd/system/omxd.service to /usr/share/doc/omxd/omxd.service.
```

The last command (`make start`) took almost 5 minutes!!
Anyway, are these warnings and errors normal during the `make` process?

[17 more replies]

closed 2016-09-30 by subogero