subogero/omxd #51: systemd config

subogero, 2016-05-15:
Add systemd unit file and install.

subogero, 2016-05-15:
Released in version 1.7

closed 2016-05-15 by subogero

OMID-313, 2016-05-16:
Thanks @subogero for adding systemd compatibility.
Can we get the new version of omxd via `apt-get` !?

subogero, 2016-05-16:
Yes, version 1.7 is in my deb repo.

OMID-313, 2016-05-16:
I installed the new version of omxd. But it seems that it has a problem. Because putting omxd in rc.local to start playing videos after boot, results in simultaneous playing of video number 1 and 2.
Can you reproduce the bug @subogero ?

OMID-313, 2016-05-16:
I tried running the code outside of rc.local . The same problem happens. Videos 1 and 2 start to play simultaneously, and after a few seconds, the screen freezes.
This is my code:

```
omxd X
omxd j
omxd O --no-osd
omxd A /mfolder
sleep 1.5
omxd -
sleep 0.1
omxd -
sleep 0.1
omxd -
sleep 0.1
omxd -
sleep 0.1
omxd -
```

OMID-313, 2016-05-16:
Dear @subogero ,
The problem outside rc.local solved by changing `sleep 1.5` to `sleep 2` in the above code.
But the `rc.local` problem still is happening.
I think the rc.local is acting differently in jessie than wheezy. Because the video starts to play (though simultaneously), and then gets stopped by the login command.

OMID-313, 2016-05-16:
The rc.local problem solved by changing `sleep 2` to `sleep 3` :D