subogero/omxd #52: Screen Flicker between videos

OMID-313, 2016-05-16:
There is a white flicker between the videos. When a video finishes and the next video starts (automatically), a flicker occurs on the screen.

Interesting point is that this flicker happens only when the videos switch automatically.
When I run commands like `omxd n` , `omxd N` , `omxd g 5` , etc , this flicker does not happen.

OMID-313, 2016-05-16:
It seems that the flicker occurs only after short videos.

OMID-313, 2016-05-16:
Actually, it is difficult to say if the flicker is relevant to video duration.
The following is the list of video order and duration:

1. 75 s    flicker at the end
2. 40 s
3. 45 s
4. 16 s    flicker at the end
5. 10 s    flicker at the end
6. 10 s
7. 14 s    flicker at the end
8. 20 s

subogero, 2016-05-16:
This is an omxplayer problem. I can't fix it.

OMID-313, 2016-05-16:
You're right @subogero .
It seems they have solved the problem here:
https://github.com/popcornmix/omxplayer/issues/449#issuecomment-217900054

Would you please update omxd based on the latest omxplayer release?
https://github.com/popcornmix/omxplayer

Thanks for your time and help.

subogero, 2016-05-17:
Yet another misunderstanding. `omxd` does not determine which `omxplayer` version is used. If you happen to install the latter from my package repo, you get the `omxplayer_0.3.6~git20160102~f544084_armhf.deb` version, after that I've stopped updating it from the build machine of http://omxplayer.sconde.net, because the newer versions were broken.

If you can confirm that the newest one is working, I can reinstate the updates. But I do not build `omxplayer` personally.

OMID-313, 2016-05-18:
Thanks @subogero for your comment.
How can I confirm whether the newest one is working or is broken !?
Do I have to uninstall omxd and install omxplayer with the following commands?
```
sudo apt-get remove omxd
sudo apt-get update
sudo apt-get install omxplayer
```
??

subogero, 2016-05-18:
Ask popcornmix the omxplayer maintainer.

OMID-313, 2016-05-18:
@popcornmix says that there is a fix for this problem, and he and other users have tested and verified the fix.
https://github.com/popcornmix/omxplayer/issues/449#issuecomment-217900054

closed 2016-09-30 by subogero