subogero/omxd #53: omxd service does not start on boot

OMID-313, 2016-06-01:
I'm using `omxd` on RPi 3, with MiniBian OS (upgraded to Jessie), and I'm using `/etc/rc.local` to start omxd to play videos on loop.

The problem is that sometimes the omxd service does not start on boot, hence, I have to log in and start the omxd daemon by typing `omxd` in command line. Then I execute the `rc.local` code manually.

I added a few seconds of delay at the beginning of `rc.local`, but it didn't help.

So, how to make sure the omxd service starts on boot, and before the codes in rc.local are executed!?

subogero, 2016-06-01:
On Jessie,  the best way is to create a systemd unit file for creating your playlist and start playback. The unit shall depend on the omxd unit. Look up details on the internet.

OMID-313, 2016-06-01:
Is the following process correct to create this systemd unit file?

**_First:_**

`sudo nano /lib/systemd/system/myservice.service`

**_Then:_**

```
[Unit]
Description=Play a playlist in loop automatically on boot
After=omxd.service
[Service]
Type=simple
ExecStart=/path/to/myscript.sh
[Install]
WantedBy=multi-user.target
```

**_And finally:_**

```
sudo chmod 644 /lib/systemd/system/myservice.service
sudo systemctl daemon-reload
sudo systemctl enable myservice.service
sudo reboot
```

Are these correct !?

subogero, 2016-06-01:
It looks good.

OMID-313, 2016-06-01:
It didn't work.

**_The status of omxd.service:_**

```
root@raspberrypi:/# systemctl status omxd.service
● omxd.service - omxd media player daemon
   Loaded: loaded (/usr/share/doc/omxd/omxd.service; enabled)
   Active: inactive (dead) since Wed 2016-06-01 13:33:30 BST; 6min ago
  Process: 314 ExecStart=/usr/bin/omxd (code=exited, status=0/SUCCESS)
 Main PID: 317 (code=killed, signal=TERM)

Jun 01 13:33:29 raspberrypi omxd[314]: omxd daemon started, PID 317
```

**_The status of myboot.service:_**

```
root@raspberrypi:/# systemctl status myboot.service
● myboot.service - Play a playlist in loop automatically on boot
   Loaded: loaded (/lib/systemd/system/myboot.service; enabled)
   Active: failed (Result: exit-code) since Wed 2016-06-01 13:33:38 BST; 7min ago
  Process: 358 ExecStart=/scripts/init.sh (code=exited, status=1/FAILURE)
 Main PID: 358 (code=exited, status=1/FAILURE)

Jun 01 13:33:36 raspberrypi init.sh[358]: One started as root listens on '/var/run/omxctl'.
Jun 01 13:33:37 raspberrypi init.sh[358]: Opening FIFO '/var/run/omxctl' timed out.
Jun 01 13:33:37 raspberrypi init.sh[358]: The omxd daemon is dead or listening on another one.
Jun 01 13:33:37 raspberrypi init.sh[358]: An unprivileged omxd listens on omxctl in its current dir.
Jun 01 13:33:37 raspberrypi init.sh[358]: One started as root listens on '/var/run/omxctl'.
Jun 01 13:33:38 raspberrypi init.sh[358]: Opening FIFO '/var/run/omxctl' timed out.
Jun 01 13:33:38 raspberrypi init.sh[358]: The omxd daemon is dead or listening on another one.
Jun 01 13:33:38 raspberrypi init.sh[358]: An unprivileged omxd listens on omxctl in its current dir.
Jun 01 13:33:38 raspberrypi init.sh[358]: One started as root listens on '/var/run/omxctl'.
Jun 01 13:33:38 raspberrypi systemd[1]: myboot.service: main process exited, code=exited, status=1/FAILURE
Jun 01 13:33:38 raspberrypi systemd[1]: Unit myboot.service entered failed state.
Jun 01 13:33:38 raspberrypi init.sh[358]: TERM environment variable not set.
```

It seems that for some reason, the `omxd.service` is dead.
Why !?

OMID-313, 2016-06-01:
And the following is the content of `myscript.sh` which shall start to execute with myboot.service :

```
clear

omxd X
omxd j
omxd O --blank
omxd O --no-osd
omxd A /usermedias

clear

sleep 3
omxd -
sleep 0.2
omxd -
sleep 0.2
omxd -
sleep 0.2
omxd -
sleep 0.2
omxd -

clear
```

subogero, 2016-06-01:
Does the `ps -A` or `pstree` command show a running `omxd` process? Try waiting a few seconds at the beginning of your script.

OMID-313, 2016-06-01:
These are the logs:

```
root@raspberrypi:/# ps -A
  PID TTY          TIME CMD
    1 ?        00:00:03 systemd
    2 ?        00:00:00 kthreadd
    3 ?        00:00:00 ksoftirqd/0
    4 ?        00:00:00 kworker/0:0
    5 ?        00:00:00 kworker/0:0H
    6 ?        00:00:00 kworker/u8:0
    7 ?        00:00:00 rcu_sched
    8 ?        00:00:00 rcu_bh
    9 ?        00:00:00 migration/0
   10 ?        00:00:00 migration/1
   11 ?        00:00:00 ksoftirqd/1
   12 ?        00:00:00 kworker/1:0
   13 ?        00:00:00 kworker/1:0H
   14 ?        00:00:00 migration/2
   15 ?        00:00:00 ksoftirqd/2
   16 ?        00:00:00 kworker/2:0
   17 ?        00:00:00 kworker/2:0H
   18 ?        00:00:00 migration/3
   19 ?        00:00:00 ksoftirqd/3
   20 ?        00:00:00 kworker/3:0
   21 ?        00:00:00 kworker/3:0H
   22 ?        00:00:00 kdevtmpfs
   23 ?        00:00:00 netns
   24 ?        00:00:00 perf
   25 ?        00:00:00 khungtaskd
   26 ?        00:00:00 writeback
   27 ?        00:00:00 crypto
   28 ?        00:00:00 bioset
   29 ?        00:00:00 kblockd
   30 ?        00:00:00 kworker/0:1
   31 ?        00:00:00 rpciod
   32 ?        00:00:00 kswapd0
   33 ?        00:00:00 vmstat
   34 ?        00:00:00 fsnotify_mark
   35 ?        00:00:00 nfsiod
   44 ?        00:00:00 kthrotld
   45 ?        00:00:00 kworker/1:1
   46 ?        00:00:00 kworker/2:1
   47 ?        00:00:00 bioset
   48 ?        00:00:00 bioset
   49 ?        00:00:00 bioset
   50 ?        00:00:00 bioset
   51 ?        00:00:00 bioset
   52 ?        00:00:00 bioset
   53 ?        00:00:00 bioset
   54 ?        00:00:00 bioset
   55 ?        00:00:00 bioset
   56 ?        00:00:00 bioset
   57 ?        00:00:00 bioset
   58 ?        00:00:00 bioset
   59 ?        00:00:00 bioset
   60 ?        00:00:00 bioset
   61 ?        00:00:00 bioset
   62 ?        00:00:00 bioset
   63 ?        00:00:00 bioset
   64 ?        00:00:00 bioset
   65 ?        00:00:00 bioset
   66 ?        00:00:00 bioset
   67 ?        00:00:00 bioset
   68 ?        00:00:00 bioset
   69 ?        00:00:00 bioset
   70 ?        00:00:00 bioset
   71 ?        00:00:00 VCHIQ-0
   72 ?        00:00:00 VCHIQr-0
   73 ?        00:00:00 VCHIQs-0
   74 ?        00:00:00 iscsi_eh
   75 ?        00:00:00 dwc_otg
   76 ?        00:00:00 kworker/3:1
   77 ?        00:00:00 DWC Notificatio
   78 ?        00:00:00 kworker/u8:1
   79 ?        00:00:00 irq/92-mmc1
   80 ?        00:00:00 VCHIQka-0
   81 ?        00:00:00 SMIO
   82 ?        00:00:00 deferwq
   83 ?        00:00:00 kworker/u8:2
   84 ?        00:00:00 bioset
   85 ?        00:00:00 mmcqd/0
   86 ?        00:00:00 jbd2/mmcblk0p2-
   87 ?        00:00:00 ext4-rsv-conver
   88 ?        00:00:00 kworker/3:2
   89 ?        00:00:00 ipv6_addrconf
  109 ?        00:00:00 kworker/2:1H
  112 ?        00:00:00 systemd-journal
  115 ?        00:00:00 kworker/0:2
  123 ?        00:00:00 kworker/1:2
  129 ?        00:00:00 kworker/2:2
  133 ?        00:00:00 systemd-udevd
  200 ?        00:00:00 cfg80211
  238 ?        00:00:00 kworker/2:3
  310 ?        00:00:00 cron
  312 ?        00:00:00 sshd
  318 ?        00:00:00 systemd-logind
  327 ?        00:00:00 dbus-daemon
  346 ?        00:00:00 lircd
  348 ?        00:00:00 ntpd
  351 ?        00:00:00 irexec
  357 tty1     00:00:00 login
  409 ?        00:00:00 systemd
  412 ?        00:00:00 (sd-pam)
  417 tty1     00:00:00 bash
  438 ?        00:00:00 networking
  445 ?        00:00:00 ifup
  480 ?        00:00:00 dhclient
  481 ?        00:00:00 sh
  482 ?        00:00:00 run-parts
  486 ?        00:00:00 openssh-server
  489 ?        00:00:00 invoke-rc.d
  508 ?        00:00:00 systemctl
  509 ?        00:00:00 sshd
  515 pts/0    00:00:00 bash
  520 pts/0    00:00:00 ps
root@raspberrypi:/# pstree
-bash: pstree: command not found
root@raspberrypi:/# ps -A | grep omx
root@raspberrypi:/#
```

As you can see, there is no omxd process.
I also tried adding delays (5 and 30 sec.) to the beginning os `myscript.sh`. But it didn't help.

One thing I noticed:
After boot, I started the omxd service manually (`service omxd start`). The checked its status by systemctl, it said active. Then started `myboot.service` manually, it worked fine.

I think the problem is that, sometimes, the login prompt happens earlier than omxd service, hence, it doesn't allow the omxd service to start. Because the same thing had happened several times in rc.local too: sometimes omxd started to play very early after boot, but then suddenly stopped and killed by login prompt.

OMID-313, 2016-06-02:
Dear @subogero ,
Can you reproduce this bug?

subogero, 2016-06-02:
No.

OMID-313, 2016-06-02:
OK.
Is it safe to manually start the omxd service in myscript.sh file?

```
service omxd start

clear

omxd X
omxd j
omxd O --blank
omxd O --no-osd
omxd A /usermedias

clear

sleep 3
omxd -
sleep 0.2
omxd -
sleep 0.2
omxd -
sleep 0.2
omxd -
sleep 0.2
omxd -

clear
```

Is it possible that it conflicts with another omxd process?

subogero, 2016-06-02:
First try to find how `omxd` dies. Look at `systemd`'s own logs first:

    journalctl -u omxd

And `omxd`'s own logs here:

    /var/log/omxlog

OMID-313, 2016-06-02:
Here are the logs:

**_journalctl -u omxd :_**

```
root@raspberrypi:/# journalctl -u omxd
-- Logs begin at Thu 1970-01-01 01:00:02 BST, end at Thu 2016-06-02 07:24:22 BST. --
Jun 01 15:05:39 raspberrypi omxd[314]: omxd daemon started, PID 316
```

**_/var/log/omxlog :_**

```
1464789939 daemonize: omxd started, SID 316
1464789940 main: X
1464789940 main: j
1464789940 main: O --blank
1464789940 main: .
1464789940 main: O --no-osd
1464789940 main: .
1464789940 main: A /usermedias/321particles_sound.mp4
1464789940 player: start /usermedias/321particles_sound.mp4
1464789940 main: A /usermedias/Asus_1faced.mp4
```

**_systemctl status omxd :_**

```
root@raspberrypi:/# systemctl status omxd
● omxd.service - omxd media player daemon
   Loaded: loaded (/usr/share/doc/omxd/omxd.service; enabled)
   Active: inactive (dead) since Wed 2016-06-01 15:05:40 BST; 16h ago
  Process: 314 ExecStart=/usr/bin/omxd (code=exited, status=0/SUCCESS)
 Main PID: 316 (code=killed, signal=TERM)

Jun 01 15:05:39 raspberrypi omxd[314]: omxd daemon started, PID 316
```

subogero, 2016-06-02:
Everything is normal, except that someone sends a TERM signal to the `omxd` daemon. See status above:

    Main PID: 316 (code=killed, signal=TERM)

OMID-313, 2016-06-02:
Well, how do I know who has sent this `TERM` signal?
And how do I solve this?

subogero, 2016-06-02:
And yes, I think it's OK to restart it in your script. But wait a few seconds before

    sleep X
    ststemctl start omxd
    ...

OMID-313, 2016-06-02:
Then I think it's better to remove `After=omxd.service` from my service unit, and change `Type=simple` to `Type=idle` to ensure my script starts to execute after all processes:

```
[Unit]
Description=Play a playlist in loop automatically on boot

[Service]
Type=idle
ExecStart=/path/to/myscript.sh

[Install]
WantedBy=multi-user.target
```

Is that correct !?

subogero, 2016-06-02:
Looks good.

subogero, 2016-06-02:
Please also try if `omxd` is killed when you disable your special unit.

OMID-313, 2016-06-02:
The above method worked fine.

----

I disabled the unit by typing `omxd P` , and then by `service myboot stop` . Then it says:

```
root@raspberrypi:~# systemctl status omxd
● omxd.service - omxd media player daemon
   Loaded: loaded (/usr/share/doc/omxd/omxd.service; enabled)
   Active: active (running) since Thu 2016-06-02 07:51:20 BST; 5min ago
  Process: 386 ExecStart=/usr/bin/omxd (code=exited, status=0/SUCCESS)
 Main PID: 387 (omxd)
   CGroup: /system.slice/omxd.service
           ├─387 /usr/bin/omxd
           └─484 dbus-daemon --fork --print-address 5 --print-pid 6 --session

Jun 02 07:51:20 raspberrypi omxd[386]: omxd daemon started, PID 387
```

So the omxd service doesn't get killed any more.
I seriously suspect that the login prompt was killing the omxd process. (Because it wouldn't get killed if started after login.)

OMID-313, 2016-06-02:
Everything works fine.
The following are the final files I used:

**_/lib/systemd/system/myboot.service :_**

```
[Unit]
Description=Play a playlist in loop automatically on boot

[Service]
Type=idle
ExecStart=/path/to/myscript.sh

[Install]
WantedBy=multi-user.target
```

**_/path/to/myscript.sh :_**

```
clear

sleep 0.3

clear

systemctl stop omxd
sleep 0.1
systemctl start omxd
sleep 0.1

clear

omxd X
omxd j
omxd O --blank
omxd O --no-osd
omxd A /usermedias

clear

sleep 3
omxd -
sleep 0.2
omxd -
sleep 0.2
omxd -
sleep 0.2
omxd -
sleep 0.2
omxd -

clear
```

Thanks @subogero for your support.
Your help and time is highly appreciated.

subogero, 2016-06-02:
Nice.

closed 2016-06-05 by OMID-313