subogero/omxd #54: Videos freeze when updating the playlist directory

OMID-313, 2016-06-05:
I have set the omxd to play all the videos in a folder.
But, every time I delete all those files and replace them with new video files and reboot the system, the 1st and 2nd videos begin to play simultaneously, and then freeze.
This happens only once after replacing new files in the directory and rebooting the system. It doesn't happen any more other times.

I didn't have this problem when using wheezy and `rc.local`. But after updating to Jessie and using systemd, this problem happens.

The following are the new codes I use (instead of `rc.local`):

**_/lib/systemd/system/myboot.service :_**

```
[Unit]
Description=Play a playlist in loop automatically on boot

[Service]
Type=idle
ExecStart=/path/to/myscript.sh

[Install]
WantedBy=multi-user.target
```

**_/path/to/myscript.sh :_**

```
clear

sleep 0.3

clear

systemctl stop omxd
sleep 0.1
systemctl start omxd
sleep 0.1

clear

omxd X
omxd j
omxd O --blank
omxd O --no-osd
omxd A /usermedias

clear

sleep 3
omxd -
sleep 0.2
omxd -
sleep 0.2
omxd -
sleep 0.2
omxd -
sleep 0.2
omxd -

clear
```

OMID-313, 2016-06-05:
Problem solved by commenting the last 10 lines (decreasing the volume)
```
...
clear

#sleep 3
#omxd -
#sleep 0.2
#omxd -
#sleep 0.2
#omxd -
#sleep 0.2
#omxd -
#sleep 0.2
#omxd -

clear
...
```

subogero, 2016-06-05:
I'll try to reproduce it.

closed 2017-05-03 by subogero