subogero/omxd #55: Clear play

joelhaasnoot, 2016-06-12:
I'm using omxd and have added a script to do some (remote) management functions. 
To update the current playlist when new content arrives, I'm doing this set of commands:
````
X
A <file1>
A <file2>
A <file3>
l
n
````
(I've also tried without l and n and either one of them)

This seems to work when the device has booted, but the second time round this plays just the first file and never continues to the next item in the playlist. After the first file, that status command `omxd S` returns strange results like `Playing 133/46 /opt/media/file1.mp4` and the log file contains several blank lines like this `1465745702 main: `

For now, I'm restarting the omxd service, after which it works again (this is something I would rather not do).

Which commands should be used to get it to continue playing after clearing the playlist? (I realise there are are other ways to synchronise a playlist, this was just the simplest/KISS way.)

subogero, 2016-06-12:
How do you send these commands to `omxd`? Write them into `/var/run/omxctl` directly, or use `omxd`as its own client like this?

    omxd X
    omxd A <file1>
    ...

Could you please post the entire log file just after reproducing the problem? It's `/var/log/omxlog`.

subogero, 2016-06-14:
Any news? Have you seen issue #56 in the meantime?

joelhaasnoot, 2016-07-30:
Sorry, just got round to this again. Upgraded to the latest version from the deb repo just now (1.8), but now for some reason no longer have ```/var/log/omxlog```. The other two logfiles are created, unfortunately. I write commands into /var/run/omxctl directly from Python.  With 1.8 it also seems like the 'X' command doesn't work and this issue is still present.

subogero, 2016-07-30:
Are you sure you flush the filehandle to `/var/run/omxctl` every time you write a command into it? It's a pipe, so libc will usually buffer your commands.

subogero, 2016-07-30:
And each command needs a `\n` linefeed at the end.

joelhaasnoot, 2016-07-30:
Got it all working again (my own systemd service, the init.d version wasn't working).
This now works correctly so I'll close the issue. Looks like the watchdog now shuts an Omxplayer in this case, that might have been the culprit?

```
1469883911 daemonize: omxd started, SID 383
1469883913 main: O --no-osd
1469883913 main: O -b
1469883913 main: X
1469883913 main: A /opt/media/a.mp4
1469883913 player_new: PID=583 /opt/media/a.mp4
1469883913 main: A /opt/media/b.mp4
1469883913 player_new: PID=584 /opt/media/b.mp4
1469883925 player_quit: PID=583 with 0
1469883925 player_cmd: Send p to omxplayer PID/fd 584/4
1469883925 player_new: PID=848 /opt/media/a.mp4
1469883939 watchdog: t_play = 14, t_len = 10
1469883939 omxwd: killing 584
1469883939 omxwd: kill omxplayer.bin 602 of omxplayer 584
1469883939 player_quit: PID=584 with 137
1469883939 player_cmd: Send p to omxplayer PID/fd 848/1
1469883939 player_new: PID=880 /opt/media/b.mp4
1469883950 player_quit: PID=848 with 0
1469883950 player_cmd: Send p to omxplayer PID/fd 880/3
1469883950 player_new: PID=922 /opt/media/a.mp4
1469883960 player_quit: PID=880 with 0
[looping]
1469884504 main: O --no-osd
1469884504 main: O -b
1469884504 main: X
1469884504 main: A /opt/media//c.mp4
1469884504 player_off: PID 1959
1469884504 omxwd: killing 1959
1469884504 omxwd: kill omxplayer.bin 1967 of omxplayer 1959
1469884504 player_quit: PID=1959 with 137
1469884504 player_new: PID=2045 /opt/media//c.mp4
1469884504 main: A /opt/media//d.mp4
1469884504 player_off: PID 1977
1469884504 omxwd: killing 1977
1469884504 omxwd: kill omxplayer.bin 1985 of omxplayer 1977
1469884505 player_quit: PID=1977 with 137
1469884505 player_new: PID=2073 /opt/media//d.mp4
1469884527 player_quit: PID=2045 with 0
1469884527 player_cmd: Send p to omxplayer PID/fd 2073/3
1469884527 player_new: PID=2097 /opt/media//c.mp4
[looping]
```

closed 2016-07-30 by joelhaasnoot