subogero/omxd #57: Sometimes on changing to the next playlist item via N or n both videos try to display at the same time

gk1, 2016-06-23:
I am running omxd on rpi 3 Linux raspberrypi 4.1.19-v7+, omxplayer version f544084

When i have a channel playing, and either run omxd n from command line, or exec same command from irexec, the videos will play at the same time, sometimes overlapping on itself, and fading/phasing between both channels. If i change the channel a couple of more times, it sometimes goes away.

Let me know if you need more information.

subogero, 2016-06-24:
omxplayer 3.7-xxx seems to be broken and sometimes ignores `q` on the standard input. This is how the newest `omxd` release stops playback. Try to install the older 3.6-xxx version of omxplayer. They are all available in my deb package repo. 

    deb http://linux.subogero.com/deb/ /

What is irexec? Please note that `omxd` assumes complete and exclusive control of all omxplayer instances. If you start/influence running/new ones in other ways (DBus, IR, gamma ray bursts in distant galaxies), it's your fault. `omxd` does not know about these.

In the meantime I'm working on branch `wd` towards a new release withy much more robust omxplayer process management. A new `omxd` release might be available in a few day's time.

subogero, 2016-06-24:
As an alternative to infrared, you might want to try my other project https://github.com/subogero/remotepi which is a single page webapp controlling `omxd`.  It can browse the RPi's folders, play internet radios and YouTube. Any device on the same LAN as your RPi can use it.

OMID-313, 2016-06-24:
I'm using LIRC (Linux Infrared Remote Controller) too.
There is no difference in running commands via `irexec` or the command line terminal.

The problem of playing both videos at the same time is a confirmed problem/bug of `omxplayer`, which has been discussed in both `omxplayer` and `omxd` issues.

gk1, 2016-07-15:
Thanks. I switched to using an older version and it is working fine.

closed 2016-07-15 by gk1