subogero/omxd #58: Seemingly random startup video from playlist

gk1, 2016-07-15:
When i reboot my pi, i seem to get a random video playing out of the playlist. Is there somewhere where the startup video is set? I would love to have a default video stream set at startup. 

Unless i am just missing something.

Thanks

gk1, 2016-07-15:
I guess as an adjunct to that, is there a way to start a specific stream if the one you are trying to change to fails to start. IE can the the player be forced to a specific stream from playlist.  I get random results when trying to go in either direction via omxd N or n, sometimes will go back to the previous stream, sometimes not.

subogero, 2016-07-15:
Your omxd seems to be in shuffle (u = unsorted) mode. There are 3 commands to set this:

* l (loop) default mode
* e (end) play list to end and stop
* u (unsorted) shuffle

gk1, 2016-07-15:
I am starting it from a script as the pi user, with this:
omxd -d
omxd h
omxd O --no-osd
omxd O --aspect-mode stretch
omxd O -b

So i would think it would be taking the default of loop, and if i just send omxd n it would seem to loop to the end and wrap around to the beginning. Its if one of the streams in the playlist is not working that throws it off i think. As i can send omxd n again and not get the broken stream, nor the next one in the list, and sometimes it just shows nothing, sometimes it puts me back the the previous track in the list. 

My playlist is a list of live rtmp streams, and we are using this sort of like a settop box interface. So channel up/down needs to be flawless for us. I am not sure if that make a difference in your response, the rtmps streams i mean.

subogero, 2016-07-15:
If omxd is properly installed your init system has already started an instance of the omxd daemon. With `omxd -d` you might start a 2nd instance, resulting in utter chaos. Check with `ps` or `pstree`.

subogero, 2016-09-30:
No more feedback, closed

closed 2016-09-30 by subogero