subogero/omxd #59: Feature request: disable 'seamless' mode

joelhaasnoot, 2016-08-29:
Choosing HDMI as the output disables the 'seamless' mode where two omxplayer instances are started to be able to interleave videos as good as possible. I'm running into an issue however, when combining this with the another openvg module and am running into the limits of Raspberry Pi RAM/GPU split options. (Running on 1920x1080 with 896 MB GPU running omxplayer + openvg-app fails, 1440x900 is fine however. Any less RAM and Raspbian won't boot.) I however do need audio to be output to the jack. Therefore, it would be great to have an option to disable seamless mode.

I may have a chance to write a pull-request later

subogero, 2016-08-29:
Good idea.

subogero, 2016-08-30:
What would be your preference, a command line option or a command?

joelhaasnoot, 2016-08-30:
Either is fine, a command keeps it consistent withother commands and allows you to change it 'on the fly', like the audio output choice.

subogero, 2016-08-30:
I now think it's less of an on-the-fly thing. I started implementing both command line and config options.

    omxd -g     # g for gap

or a line in `/etc/omxd.conf`:

    gap=1

subogero, 2016-08-30:
It works now. See on `gap-opt` branch.

subogero, 2016-08-30:
BTW the first commit makes the low-gap mode even more seamless.

subogero, 2016-08-30:
Just rewrote branch history to fix a typo in a commit message...

subogero, 2016-09-10:
Is my solution OK? I ask because I'm planning a new release...

joelhaasnoot, 2016-09-30:
Sorry, thought I had commented, this is perfect. A release would be great, allowing me to go back to prebuilt packages, currently built from source. Currently ramping up deployment, but tested this at a large scale event and it handled massive playlists and a ticker tape well.

closed 2016-09-30 by joelhaasnoot

subogero, 2016-09-30:
Ok, I'll do a release.
Happy to hear about large scale events.  :-)

subogero, 2016-09-30:
1.9 released in my deb repo.