subogero/omxd #60: omxd fails to start when /var/log/omxlog doesn't already exist

thomas536, 2016-09-10:
Thanks for writing this code! It's working great!

But sometimes I see an error that the daemon doesn't start.  The child process gets forked, but it immediately exits (return code 4) when omxlog doesn't already exist.  Not sure if this is intended or not.  Maybe these log files should be opened w/ O_CREAT?

Thanks!

```
[root@pi-mediacenter run]# strace -f omxd 
<snip>
clone(child_stack=0, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0xb6efc068) = 2585
creat("/var/run/omxd.pid", 0644)        = 3
write(3, "2585", 4)                     = 4
close(3)                                = 0
write(1, "omxd daemon started, PID ", 25omxd daemon started, PID ) = 25
write(1, "2585", 42585)                     = 4
write(1, "\n", 1
)                       = 1
exit_group(0)                           = ?
+++ exited with 0 +++
Process 2585 attached
setsid()                                = 2585
close(2)                                = 0
open("/var/log/omxlog", O_WRONLY|O_APPEND) = -1 ENOENT (No such file or directory)
close(0)                                = 0
close(1)                                = 0
exit_group(4)                           = ?
+++ exited with 4 +++
[root@pi-mediacenter run]# ps axf | grep 2585
 2593 pts/0    S+     0:00                          \_ grep 2585
[root@pi-mediacenter run]# ls /var/log/omxlog
ls: cannot access /var/log/omxlog: No such file or directory
```

subogero, 2016-09-10:
It's a privilege to have users who report bugs with an strace. :-)

subogero, 2016-09-10:
This bug resulted from the latest release, from commit 19b71a37 on the `wd` branch, which ended up as a major rewrite of subprocess management. I had many problems and I wanted to preserve logs from previous runs.

I'll check whether the file exists and use `open` or `creat` accordingly.

subogero, 2016-09-10:
Do you compile from source, or need a new version of the deb package?

thomas536, 2016-09-10:
I'm using the deb, but no rush.  It's pretty easy to work around. =)

subogero, 2016-09-30:
1.9 released with the fix.

closed 2016-09-30 by subogero