subogero/omxd #61: /etc/init.d/omxd matching its own pid

thomas536, 2016-09-11:
Sorry for another bug report... =)

It's also looking like /etc/init.d/omxd is matching its own pid during omxd_pid (as evidenced by logging the process list in the script).  (After much fiddling) I finally settled on using the full path to omxd and using the output of `pidof /usr/bin/omxd` in omxd_pid().  But I have no idea what sort of compatibility you need/which binaries you expect to have on disk...

Thanks again!

```
[root@pi-mediacenter pi]# cat /etc/init.d/omxd
#!/bin/sh
### BEGIN INIT INFO
# Provides:          omxd
# Required-Start:    $local_fs
# Required-Stop:     $local_fs
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Description:       Raspberry Pi omxplayer playlist daemon and YouTube player
### END INIT INFO

omxd_pid() {
  ps -C omxd -o pid=,ppid= | while read PID PPID; do
    [ "$PPID" = 1 ] && echo $PID && break
  done
}

start() {
  PID=`omxd_pid`
  ps axf
  if [ `omxd_pid` ]; then 
    echo "[info] Service omxd already running (${PID})" >&2
    return 0
  fi
  echo -n '[....] Starting omxd playlist daemon... ' >&2
  /usr/bin/omxd >/dev/null
  echo "done.\r[ ok ]" >&2
}
<snip>
[root@pi-mediacenter pi]# systemctl status omxd
* omxd.service - LSB: Raspberry Pi omxplayer playlist daemon and YouTube player
   Loaded: loaded (/etc/init.d/omxd)
   Active: active (exited) since Sun 2016-09-11 04:02:14 UTC; 53s ago
  Process: 318 ExecStart=/etc/init.d/omxd start (code=exited, status=0/SUCCESS)

Sep 11 04:02:15 pi-mediacenter omxd[318]: 317 ?        Ss     0:00 /lib/systemd/systemd-logind
Sep 11 04:02:15 pi-mediacenter omxd[318]: 318 ?        Ss     0:00 /bin/sh /etc/init.d/omxd start
Sep 11 04:02:15 pi-mediacenter omxd[318]: 340 ?        R      0:00  \_ ps axf
Sep 11 04:02:15 pi-mediacenter omxd[318]: 319 ?        Ss     0:00 /bin/sh /etc/init.d/dphys-swapfile start
Sep 11 04:02:15 pi-mediacenter omxd[318]: 335 ?        S      0:00  \_ /bin/sh /sbin/dphys-swapfile setup
Sep 11 04:02:15 pi-mediacenter omxd[318]: 323 ?        Zs     0:00 [triggerhappy] <defunct>
Sep 11 04:02:15 pi-mediacenter omxd[318]: 328 ?        Ss     0:00 avahi-daemon: starting up
Sep 11 04:02:15 pi-mediacenter omxd[318]: 339 ?        S      0:00  \_ avahi-daemon: chroot helper
Sep 11 04:02:15 pi-mediacenter omxd[318]: 333 ?        Ds     0:00 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation
Sep 11 04:02:15 pi-mediacenter omxd[318]: [info] Service omxd already running (318)
```

subogero, 2016-09-12:
You seem to mix up the SysV and systemd config files of omxd. You should use one or the other, but not both. 

You start the SysV init script `/etc/init.d/omxd` from systemd instead of the actual daemon `/usr/bin/omxd`.

My original systemd config file for omxd contains this:

    [Service]              
    ExecStart=/usr/bin/omxd
    Type=forking

subogero, 2016-09-12:
The deb package copies the systemd config file `omxd.service` to `/usr/share/doc/omxd/omxd.service`, then its `postinst` script adds it to `systemd` via

    systemctl enable /usr/share/doc/omxd/omxd.service

resulting in a symlink in `/etc/systemd/system/` pointing to the config file.

subogero, 2016-09-12:
The above happens if `postinst` detects `systemd` as your init system. Otherwise it installs the SysV init script `/etc/init.d/omxd` and registers it for automatic startup.

thomas536, 2016-09-14:
Ah, hmm yeah I don't usually use Systemd, so I very well could be confused...  I also switched from non-systemd to systemd on this system, which might have mucked things up.  I'll go ahead and close this and if I'm still seeing the issue, I'll re-open.  Thanks for the thorough explanation!

closed 2016-09-14 by thomas536