subogero/omxd #62: Gap between videos

arkoe, 2016-10-20:
Loving the daemon!

The only thing I struggle with is that there is a gap of like 0.5 seconds between the videos. Is there any way to have this seamless?

subogero, 2016-10-20:
There is a low-gap feature, which starts the omxplayer with next video beforehand, paused in the background. But it only works if you use the jack for audio output and not HDMI.

In case of HDMI sound, there is interference between the sound of the 2 omxplayer instances, so this feature is disabled.

On the other hand, using jack sound, the gap is just a tenth of a second, barely noticeable.

arkoe, 2016-10-20:
How do I enable this low-gap feature?

subogero, 2016-10-20:
Command `j` sets the jack as the audio output. Then it's low gap.

mvasin, 2016-12-01:
Is there a way to get rid of gaps completely? It's OK if there will be no sound for my application.

subogero, 2016-12-01:
No. There will always be a few tenths of a second gap.

closed 2016-12-09 by subogero