subogero/omxd #63: Version 1.9 doesn't handle clearing playlist correctly.

joelhaasnoot, 2016-10-29:
I have version 1.9 installed from the dep repo and am running into an issue changing the playlist. To change the playlist I use the "X" command to clear and then add new files. This results in the currently playing video and the new video both being played at the same time (with a lovely flashing effect).
The logfile also has some strange loglines:
```
1477757765 quit_callback: start /opt/media/TiG0XO6_lhvyntc.mp4
1477757765 player_new: PID=1695 /opt/media/TiG0XO6_lhvyntc.mp4
1477757783 player_quit: PID=1695 with 0
1477757783 quit_callback: start /opt/media/HGzzhZN_6u69tlg.mp4
1477757783 player_new: PID=1715 /opt/media/HGzzhZN_6u69tlg.mp4
1477757799 main: X
14777577991477757799  main: main:  p/ieoigmda/ot_heeas-Buste()m4A/otpxlpstn/ei/TosTilmn__leet 1.p

1477757799 main: A /opt/media/Funky_Kingston_Full_Album (1).mp4
1477757799 player_off: PID 1715
1477757799 player_new: PID=1737 1.p
1477757799 main: A /opt/media/Shape_Of_My_Heart_1993 (1).mp4
1477757799 omxwd: killing 1715
1477757799 omxwd: kill omxplayer.bin 1726 of omxplayer 1715
1477757800 player_quit: PID=1715 with 137
1477757800 player_new: PID=1760 /opt/media/Funky_Kingston_Full_Album (1).mp4
1477757800 player_quit: PID=1737 with 1
1477757800 quit_callback: start /opt/media/Shape_Of_My_Heart_1993 (1).mp4
1477757800 player_new: PID=1771 /opt/media/Shape_Of_My_Heart_1993 (1).mp4
```
Playing two video's at once crashes the GPU unfortunately.

joelhaasnoot, 2016-10-30:
Oops: accidentally was running two instances of omxd: one was listening to commands, the other was running the saved playlist.

closed 2016-10-30 by joelhaasnoot

subogero, 2016-10-31:
Thank goodness. I thought all the work refactoring the subprocess control was in vain. Please refrain from scaring me stiff in the future. :-)

subogero, 2016-11-01:
Do you think it would make sense for omxd to check for an already running instance of the daemon and exit?

joelhaasnoot, 2017-02-11:
very delayed comment, yes that would make sense in daemon mode.