subogero/omxd #65: effects on video transitions

rifotu, 2017-01-26:
Hello,

Can omxd do effects like fade out during video transitions?

subogero, 2017-01-26:
Fading out the audio is supported by `omxplayer` via hotkey. But I don't know about fading out the video.

Theoretically it's possible, but `omxd` at the moment only reacts on user inputs and exiting `omxplayer` child processes. So it does not anticipate the end of the track.

So this feature would necessitate a periodic timer event source in omxd, which is quite complicated.

rifotu, 2017-01-26:
hello,

thanks for the answer. Is it then possible to have the fade out effect happening on a user input? For example when a user presses a button, a seperate video starts playing but the transition is not instantaneous and instead fade out effect happens.

subogero, 2017-01-26:
Do you want video or audio fadeout?

rifotu, 2017-01-27:
I would like to have video fadeout if possible. Audio fadeout would be nice but not really required.

subogero, 2017-01-27:
Could you please investigate whether `omxplayer` supports this? If not, no chance. Please note that `omxd` does not play back anything, it merely manages `omxplayer` instances.

rifotu, 2017-02-01:
so sad :(

the guys at omxplayer repository said that they don't support fadeout effects

subogero, 2017-02-04:
Great sadness for you,
Great relief for me...

closed 2017-02-04 by subogero