subogero/omxd #66: Monitor the log file in real time, and do something. How?

OMID-313, 2017-02-15:
Hi,

I want to monitor the omxd log file in real time, and execute some commands when the video changes.
I searched this site (and many other sites) and these are what I tried:

`tail -f /var/log/omxlog | awk '/player_new/ { "echo hello" }'`
or
`stdbuf -o0 tail -F /var/log/omxlog | awk '/player_new/ { "echo hello" }'`

But they don't work. Whenever I run these commands, it begins to wait, but although I'm sure the log file changes, it doesn't print hello; actually it doesn't do anything. Just waiting :D

Maybe the problem is that I'm accessing the RPi via SSH. I don't know.
(System: Raspberry Pi 3. OS: Minibian)

So, what should I do!?

P.S. Is there any other method of knowing when the video is changed !?

subogero, 2017-02-15:
It's your awk script that's broken, but I don't know awk. A few alternatives:

    tail -f /var/log/omxlog | grep player_new

And if you insist on printing `hello` instead of the line:

    tail -f /var/log/omxlog | sed -n 's/.*player_new.*/hello/p'
    tail -f /var/log/omxlog | perl -lne 'print if s/.*player_new.*/hello/'

OMID-313, 2017-02-15:
Thanks for your answer.
Actually, I want to send the name of the video file via SPI. (So, which one do you recommend personally!?)
I'll try these suggestions

subogero, 2017-02-15:
This will list only the new filename:

    tail -f /var/log/omxlog | sed -rn 's/.*player_new: [^ ]+ //p'

But there is a catch! In low gap mode you don't see the current file but the next one!

subogero, 2017-02-15:
To print the coorect file, this would do:

    tail -f /var/log/omxlog | stdbuf -o0 grep player_new | while read i; do omxd S; done

OMID-313, 2017-02-16:
Thank you very much @subogero for your answers.
These two codes both work perfectly.
Thanks

closed 2017-02-16 by subogero