subogero/omxd #69: Add possibility of looping one video until user sends omxd n

tmolberg, 2017-05-02:
Add possibility off looping one video indefinantly, similar to to --loop function within omxplayer. Tried utilizing omxd O --loop without any great success. 

Typically usecase is when you want the pi to view one video until you give it the command to skip to next video.

subogero, 2017-05-02:
Have you checked whether your omxplayer supports the --loop option?

Define "without any great success".

subogero, 2017-05-02:
My omxplayer seems to support `--loop` for both audio and video.

Investigating omxd...

subogero, 2017-05-02:
Found the problem. The `omxwd` watchdog is killing `omxplayer.bin` a few seconds after the track length expires. It ignores to `--loop` option.

Fixing...

subogero, 2017-05-02:
Your use-case is running like a breeze on my RPi...

tmolberg, 2017-05-02:
Could you elaborate on how you got it to work? I've been trying all the
parameters I can see fit my usecase without any working outcome.

Current "startup" script is looking like this. Reworked a little bit from
looking at previous git issues. Also tried typing in everything manually
incase it was a problem of commands being sent to fast.

#!/bin/bash
sudo service omxd stop
sleep 1s
sudo service omxd start

omxd e
omxd X
omxd O --blank --no-osd --loop

VIDEOPATH='/home/pi/Desktop/UploadedMovies'

for entry in $VIDEOPATH/*
do
    clear
    echo A $entry > /var/run/omxctl
    sleep 0.2s
done

subogero, 2017-05-02:
I modified the C code of omxd. It won't work for you until I publish the fix.

tmolberg, 2017-05-02:
Ah! Makes sence.

Thanks

subogero, 2017-05-02:
Just realized that

    omxd O

to clear all the omxplayer options does not work...

subogero, 2017-05-02:
Can you compile from source and test?

subogero, 2017-05-02:
Clone the repo or pull from here, then

    apt install build-essential
    ./configure
    make
    sudo make restart

Good night...

tmolberg, 2017-05-02:
Will clone the repo tomorrow. Thanks alot for the effort

Good night.

tmolberg, 2017-05-03:
Working...

Will conduct a few more tests today to see how stable it is.

Thanks yet again.

subogero, 2017-05-03:
If it's OK, I'll release the 2.0 version soon.

tmolberg, 2017-05-03:
Seems disabling the watchdog was not ideal. I managed to get into a state where its not currently running alot of omxplayers and I see the screen flickering / loosing sync. 

running omxd with the follow arguments

omxd X
omxd O --blank --no-osd --loop

See ps aux 
`
root     10430  0.0  0.5   3120  2612 ?        S    12:32   0:00 /bin/bash /usr/bin/omxplayer -I --vol 0 --blank --no-osd --loop --blank --no-osd --loop --blank --no-osd
root     10438  0.2  0.0      0     0 ?        Zl   12:32   0:00 [omxplayer.bin] <defunct>
root     10459  0.0  0.5   3120  2584 ?        S    12:32   0:00 /bin/bash /usr/bin/omxplayer -I --vol 0 --blank --no-osd --loop --blank --no-osd --loop --blank --no-osd
root     10467  100  3.7 105476 18720 ?        R    12:32   2:53 /usr/bin/omxplayer.bin -I --vol 0 --blank --no-osd --loop --blank --no-osd --loop --blank --no-osd --loop
root     10488  0.0  0.5   3120  2548 ?        S    12:32   0:00 /bin/bash /usr/bin/omxplayer -I --vol 0 --blank --no-osd --loop --blank --no-osd --loop --blank --no-osd
root     10496  0.0  0.0      0     0 ?        Zl   12:32   0:00 [omxplayer.bin] <defunct>
root     10515  0.0  0.5   3120  2612 ?        S    12:32   0:00 /bin/bash /usr/bin/omxplayer -I --vol 0 --blank --no-osd --loop --blank --no-osd --loop --blank --no-osd
root     10523  0.0  0.0      0     0 ?        D    12:32   0:00 [omxplayer.bin]
root     10537  0.0  0.5   3120  2548 ?        S    12:32   0:00 /bin/bash /usr/bin/omxplayer -I --vol 0 --blank --no-osd --loop --blank --no-osd --loop --blank --no-osd
root     10545  0.0  0.0      0     0 ?        D    12:32   0:00 [omxplayer.bin]
root     10559  0.0  0.5   3120  2540 ?        S    12:32   0:00 /bin/bash /usr/bin/omxplayer -I --vol 0 --blank --no-osd --loop --blank --no-osd --loop --blank --no-osd
root     10567  0.0  0.0      0     0 ?        D    12:32   0:00 [omxplayer.bin]
root     10593  0.0  0.5   3120  2548 ?        S    12:34   0:00 /bin/bash /usr/bin/omxplayer -I --vol 0 --blank --no-osd --loop --blank --no-osd --loop --blank --no-osd
root     10601  0.1  0.0      0     0 ?        D    12:34   0:00 [omxplayer.bin]
root     10619  0.1  0.5   3120  2612 ?        S    12:35   0:00 /bin/bash /usr/bin/omxplayer -I --vol 0 --blank --no-osd --loop --blank --no-osd --loop --blank --no-osd
root     10638  0.3  0.0      0     0 ?        D    12:35   0:00 [omxplayer.bin]
root     10643  0.1  0.5   3120  2520 ?        S    12:35   0:00 /bin/bash /usr/bin/omxplayer -I --vol 0 --blank --no-osd --loop --blank --no-osd --loop --blank --no-osd
root     10658  0.3  0.0      0     0 ?        D    12:35   0:00 [omxplayer.bin]
root     10663  0.1  0.5   3120  2548 ?        S    12:35   0:00 /bin/bash /usr/bin/omxplayer -I --vol 0 --blank --no-osd --loop --blank --no-osd --loop --blank --no-osd
root     10678  0.3  0.8  37140  4092 ?        Sl   12:35   0:00 /usr/bin/omxplayer.bin -I --vol 0 --blank --no-osd --loop --blank --no-osd --loop --blank --no-osd --loop
root     10683  0.1  0.5   3120  2548 ?        S    12:35   0:00 /bin/bash /usr/bin/omxplayer -I --vol 0 --blank --no-osd --loop --blank --no-osd --loop --blank --no-osd
root     10691  0.3  0.8  37140  4192 ?        Sl   12:35   0:00 /usr/bin/omxplayer.bin -I --vol 0 --blank --no-osd --loop --blank --no-osd --loop --blank --no-osd --loop
pi       10694  0.0  0.4   4488  2128 pts/0    R+   12:35   0:00 ps aux
`

subogero, 2017-05-03:
As you can see, you've added the omxplayer options multiple times. They accumulate! Please clear them first:

    omxd O
    omxd O --blank --no-osd --loop

Please retry this way.

tmolberg, 2017-05-03:
I'll retry it. Thanks.

tmolberg, 2017-05-03:
Seems to be working. However when hammering it (extreme usecase I know). It seems like the watchdog was unable to keep track thereby spawning to many instances of omxplayer which lead to the rpi3 crashing. 

Attached are logfiles. Was unable to get a dump of ps

[omxdlog.zip](https://github.com/subogero/omxd/files/973231/omxdlog.zip)

subogero, 2017-05-03:
Define "hammering".

tmolberg, 2017-05-03:
Switching movie (omxd n) with 0.5s-ish interval.

subogero, 2017-05-03:
Sorry could not reproduce your problem. Tried "hammering" every 0.5s for a few minutes. Absolutely no runaway processes, once I stopped the hammering, the current video ran smoothly. But nothing appeared on the screen, omxplayer had not enough time to initialize.

Hammering every 1s, similar result, but I saw a fraction of a second from the beginning of each video.

I'm running an RPi 2, though...

subogero, 2017-05-04:
Looking at you `omxlog`, you're still adding the options multiple times.

    1493813370 main: X
    1493813370 main: O --blank
    1493813370 main: O --no-osd
    1493813370 main: O --loop
    1493813370 main: .

This happens many times. Again: Added options ACCUMULATE. You end up calling omxplayer like this:

    /usr/bin/omxplayer --blank --no-osed --loop --blank --no-osed --loop --blank --no-osed --loop --blank --no-osed --loop -ohdmi FILE

The right sequence to redefine the omxplayer options:

    omxd O # Clear current options
    omxd O --blank --no-osd --loop # ADD options. Not replace. See man page

subogero, 2017-05-04:
BTW, it seems I managed to reproduce a runaway omxplayer. I think the key is `omxd X` which clears the playlist, but leaves the omxplayer instances running. I also managed to crash the daemon.

Investigating...

tmolberg, 2017-05-05:
Hi.

Sorry for not including my start scriping. Before I sent the logs I had altered my start script to include. 

`omxd O #clear current omx options`

What you are seing in the logs are probably remenants of my previous mistake before I had put in 
`omxd O`

subogero, 2017-05-05:
Anyway, I'm on track. Thanks.

subogero, 2017-05-05:
Fixed a bug in the watchdog's `--loop` detector, which sometimes crashed the daemon with SIGSEGV. Then hammered it with something like your script for ages, no crash, no stale omxplayers.

BTW, now command X stops playback completely. Is it OK? See #64.

subogero, 2017-05-07:
omxd 2.0.0 available in by deb repo.

closed 2017-05-07 by subogero