subogero/omxd #72: next song won't play

kpalczewski, 2017-05-20:
Hi,

I was able to use omxd and create a playlist, 1st song is playing but it won't jump to next one.

xx@raspberrypi:~ $ omxd S
Playing 365/277 /mnt/HTPC/Music/somesong.mp3

If I tell it manually to play next song - it will be played without any problems:
xx@raspberrypi:~ $ omxd n

Do you have any idea where to look for a solution?

subogero, 2017-05-20:
Your `omxd S` output suggests that omxplayer does not exit on its own: play time over track length! The `n` command actually kills that instance.

What `omxd` version are you using?

Could you please play the first track directly with `omxplayer` and post the results? Does it exit at the end of the track?

kpalczewski, 2017-05-22:
omxd 1.11

omxplayer finished a song and exited when I played it directly with omxplayer.

Those are processes that play a song (when I use omxd as I normally do):
1946 ?        Ss     0:00 omxd
 1966 ?        S      0:00 /bin/bash /usr/bin/omxplayer -I --vol 0 -oalsa /mnt/HTPC/Music/VA Dance/03 - David Guetta - Titanium (Feat. Sia).mp3
 1976 ?        Sl     0:03 /usr/bin/omxplayer.bin -I --vol 0 -oalsa /mnt/HTPC/Music/VA Dance/03 - David Guetta - Titanium (Feat. Sia).mp3

and when the song finishes:
 1946 ?        Ss     0:00 omxd
 1966 ?        Z      0:00 [omxplayer] <defunct>

subogero, 2017-05-22:
How did you achieve `-oalsa` via omxd? It can only do `-ohdmi` and `-olocal`.

Did you run omxplayer with the exact same options?

Is that a zombie after the song has finished?

Anyway, I think it's David Guetta's fault.

kpalczewski, 2017-05-22:
I have modified omxd.c to to achieve -oalsa:
        enum e_outputs           { OUT_JACK,  OUT_ALSA };
        static char *outputs[] = { "-olocal", "-oalsa" };
        static enum e_outputs output = OUT_ALSA;
I wasn't sure, if another output would cause any issues, so I have changed out_hdmi to alsa.
I'm using RPI to stream music through bluetooth so omxd was missing an output option for me.

I'm using python script (running root I think) to create a playlist and start playing, so there might be some issues there as well.

I would believe that David Guetta is to blame, if Foo Fighters' playlist would play without any issues. :D

I'll update to 2.0 and check, if it helps.

subogero, 2017-05-22:
If you modified the code, your problem should be stated as a pull request. You should have told me about the changed code in the first place.

Audio output affects more things than you think.

closed 2017-07-28 by subogero