subogero/omxd #73: systemd service fails to start with latest raspbian image

tmolberg, 2017-08-23:
When installing from your repo the systemd service refuses to start. Seems that its finding the omxd service before it expects?

subogero, 2017-08-23:
I need details. What OS version, what error message, etc...

tmolberg, 2017-08-23:
First thing I tried was to compile it from git. Strangely enough I get the same types of errors when I compile it or by using your repo. 

Raspbian stretch - https://downloads.raspberrypi.org/raspbian_lite_latest

sudo systemctl start omxd.service
Job for omxd.service failed because the control process exited with error code.
See "systemctl status omxd.service" and "journalctl -xe" for details.

Aug 23 12:20:23 Pi systemd[1]: Starting omxd media player daemon...
Aug 23 12:20:23 Pi omxd[5326]:  5326     1     0 /usr/bin/omxd
Aug 23 12:20:23 Pi omxd[5326]: Daemon as uid 0 already running
Aug 23 12:20:23 Pi systemd[1]: omxd.service: Control process exited, code=exited status=255
Aug 23 12:20:23 Pi systemd[1]: Failed to start omxd media player daemon.
Aug 23 12:20:23 Pi systemd[1]: omxd.service: Unit entered failed state.
Aug 23 12:20:23 Pi systemd[1]: omxd.service: Failed with result 'exit-code'.

— Unit omxd.service has begun starting up.
Aug 23 12:20:23 Pi omxd[5326]:  5326     1     0 /usr/bin/omxd
Aug 23 12:20:23 Pi omxd[5326]: Daemon as uid 0 already running
Aug 23 12:20:23 Pi systemd[1]: omxd.service: Control process exited, code=exited status=255
Aug 23 12:20:23 Pi systemd[1]: Failed to start omxd media player daemon.
— Subject: Unit omxd.service has failed
— Defined-By: systemd
— Support: https://www.debian.org/support
—
— Unit omxd.service has failed.
—
— The result is failed.
Aug 23 12:20:23 Pi systemd[1]: omxd.service: Unit entered failed state.
Aug 23 12:20:23 Pi systemd[1]: omxd.service: Failed with result 'exit-code'.
Aug 23 12:20:23 Pi sudo[5320]: pam_unix(sudo:session): session closed for user root

subogero, 2017-08-23:
Please post the contents of  `/var/log/omxlog`

tmolberg, 2017-08-23:
1503489206 daemonize: omxd started, SID 2806
1503490049 daemonize: omxd started, SID 775
1503490935 daemonize: omxd started, SID 5400
1503491334 daemonize: omxd started, SID 6034
1503491516 daemonize: omxd started, SID 6124
1503492710 daemonize: omxd started, SID 6810
1503492904 daemonize: omxd started, SID 6858
15034946221503494625 daemonize: omxd started, SID 2545
1503494679 daemonize: omxd started, SID 2590
1503494696 daemonize: omxd started, SID 2607
1503494704 daemonize: omxd started, SID 2624
15034947301503494898 daemonize: omxd started, SID 750
1503495533

I had a play around with the systemd service file and got it to run but the "fix" was not pretty. What I did was to rename the omxd bin to something random and pointing the systemd service file to start that file instead which seems to make it work (however not stable)

subogero, 2017-08-23:
Daemon as uid 0 already running.

You seem to start 2 instances of `omxd` somehow.

subogero, 2017-08-23:
What is strange in getting the same results when compiling form source or using from repo? I would be worried if it wasn't the case.

Did you use this same `omxd` version on your older OS? Or was that an older version of `omxd` too?

tmolberg, 2017-08-23:
The last time I was using omxd i was running an older version of raspbian, i believe it was 14.04. I also believe it was before systemd was implemented in raspbian which is probably why I didnt have this problem last time. 

That being said I dont think the version (the last time I used omxd) was at 2.0. 

Have you had any luck recreating the problem? 

Basic procedure to replicate the problem is basicly just adding your repo to the sources list then apt-get install omxd which then fails. Same error when compiling aswell.

subogero, 2017-08-23:
Check your `/etc/rc.local` and remove everything omxd related. Same for the `/etc/init/d` directory.

tmolberg, 2017-08-23:
There is nothing related to omxd in rc.local or init.d

tmolberg, 2017-08-25:
Tried using the init.d script

● omxd.service
   Loaded: loaded (/usr/share/doc/omxd/omxd.service; bad; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2017-08-25 17:05:17 UTC; 15s ago
  Process: 815 ExecStart=/usr/bin/omxd (code=exited, status=255)

Aug 25 17:05:17 Pi systemd[1]: Starting omxd.service...
Aug 25 17:05:17 Pi omxd[815]:   815     1     0 /usr/bin/omxd
Aug 25 17:05:17 Pi omxd[815]: Daemon as uid 0 already running
Aug 25 17:05:17 Pi systemd[1]: omxd.service: Control process exited, code=exited status=255
Aug 25 17:05:17 Pi systemd[1]: Failed to start omxd.service.
Aug 25 17:05:17 Pi systemd[1]: omxd.service: Unit entered failed state.
Aug 25 17:05:17 Pi systemd[1]: omxd.service: Failed with result 'exit-code'.

subogero, 2017-08-25:
Exit status 255 means an instance of the daemon is already running at startup.

tmolberg, 2017-08-25:
What I find curious is that it says "Daemon as UID 0 already running" however now omxd is running when checking with ps aux | grep omxd

subogero, 2017-08-25:
Grep recursively your `/etc` folder for anything `omxd`.

tmolberg, 2017-08-25:
root@Cyviz-Pi:/etc# find | grep -r omxd
init.d/.depend.start:TARGETS = console-setup.sh rsyslog triggerhappy dhcpcd omxd apache2 dbus paxctld raspi-config cron dphys-swapfile rsync ssh avahi-daemon lightdm bluetooth plymouth
init.d/.depend.start:plymouth: avahi-daemon dbus rsyslog lightdm triggerhappy bluetooth dhcpcd paxctld apache2 console-setup.sh raspi-config cron dphys-swapfile rsync ssh omxd
init.d/omxd:# Provides:          omxd
init.d/omxd:omxd_pid() {
init.d/omxd:  ps -C omxd -o pid=,ppid= | while read PID PPID; do
init.d/omxd:  if [ `omxd_pid` ]; then
init.d/omxd:    echo '[info] Service omxd already running' >&2
init.d/omxd:  echo -n '[....] Starting omxd playlist daemon... ' >&2
init.d/omxd:  omxd >/dev/null
init.d/omxd:  PID=`omxd_pid`
init.d/omxd:    echo '[info] Service omxd not running' >&2
init.d/omxd:  echo -n '[....] Stopping omxd playlist daemon... ' >&2
init.d/.depend.stop:TARGETS = nfs-common rpcbind apache-htcacheclean triggerhappy dhcpcd omxd apache2 alsa-utils fake-hwclock paxctld avahi-daemon lightdm bluetooth plymouth rsyslog networking hwclock.sh

The init.d stuff you are seeing was just to get started with the project im working on currently. When I can get systemd up and running I will convert it back to using systemd.

subogero, 2017-08-25:
The last thing I want to do is getting involved in your custom setup.

closed 2017-08-25 by subogero

tmolberg, 2017-08-25:
Understandable,

I can always revert everything back to normal -> plain install with only omxplayer and omxd if you are up for further debugging.

subogero, 2017-08-30:
There seem to be omxd/systemd problems on Raspbian Jesse Lite. I don't have this OS so far, until then see this issue: #74 , it contains a workaround.

subogero, 2017-09-02:
See #74 

I apologise, this is a bug in the newest `omxd` which was installed along with the new OS.

tmolberg, 2017-09-02:
Hi,

Do you have any details in what is causing it? Im happy to try any new versions you might have, even donate a pi3 with raspbian stretch.

Regards,
Tor Martin Ølberg
+4799521108

subogero, 2017-09-02:
I know what the issue is, fix coming this weekend.

subogero, 2017-09-02:
Released `omxd 2.0.1`.

Please remove all your local omxd related hacks, and then

    apt remove omxd
    apt update
    apt install omxd