subogero/omxd #74: Can't install omxd properly

OMID-313, 2017-08-29:
Hi.

I have installed and used `omxd` several times without problem.
This time, I tried installing it on RPi 3 with Raspbian Jessie Lite.
(Note: I've installed `fail2ban` and `ssh` with secure keys.)

But it gives errors at the last steps of installation.

Here are some logs:

```
~ $ sudo systemctl status omxd

● omxd.service
   Loaded: not-found (Reason: No such file or directory)
   Active: failed (Result: exit-code) since Tue 2017-08-29 15:17:04 IRDT; 37min ago

omxd[555]: 555     1     0 /usr/bin/omxd
omxd[555]: Daemon as uid 0 already running
systemd[1]: omxd.service: control process exited, code=exited status=255
systemd[1]: Failed to start omxd media player daemon.
systemd[1]: Unit omxd.service entered failed state.
systemd[1]: Stopped omxd media player daemon.
systemd[1]: Stopped omxd media player daemon.
systemd[1]: Stopped omxd.service.
systemd[1]: Stopped omxd.service.
```

and

```
~ $ sudo systemctl start omxd.service

Failed to start omxd.service: Unit omxd.service failed to load: No such file or directory.
```

and

```
~ $ cd /etc/
/etc $ sudo find | grep -r omxd

grep: gshadow: Permission denied
grep: group-: Permission denied
grep: security/opasswd: Permission denied
grep: wpa_supplicant/wpa_supplicant.conf: Permission denied
grep: polkit-1/localauthority: Permission denied
grep: .pwd.lock: Permission denied
grep: chatscripts: Permission denied
grep: sudoers: Permission denied
grep: subgid-: Permission denied
grep: shadow-: Permission denied
grep: ssl/private: Permission denied
grep: gshadow-: Permission denied
grep: passwd-: Permission denied
grep: ssh/ssh_host_dsa_key: Permission denied
grep: ssh/ssh_host_rsa_key: Permission denied
grep: ssh/ssh_host_key: Permission denied
grep: ssh/ssh_host_ecdsa_key: Permission denied
grep: ssh/ssh_host_ed25519_key: Permission denied
grep: shadow: Permission denied
grep: subuid-: Permission denied
grep: sudoers.d/010_pi-nopasswd: Permission denied
grep: sudoers.d/README: Permission denied
grep: ppp/pap-secrets: Permission denied
grep: ppp/peers: Permission denied
grep: ppp/chap-secrets: Permission denied
```

Also, there is no `/var/log/omxlog` file !!

I tried installing both from deb (`sudo apt-get install omxd`) and cloning from github. But both of them have the same results.

So, what's wrong?
How can I solve this !?

OMID-313, 2017-08-29:
New log:

```
~ $ sudo apt-get remove omxd

. . .
Package 'omxd' is not installed, so not removed.
. . .
```

Also, I can confirm that `omxplayer` has been installed successfully with `omxd`, because calling `omxplayer myfile.mp4` starts the video.

OMID-313, 2017-08-29:
New log:

```
~ $ sudo apt-get install omxd

Reading package lists... Done
Building dependency tree
Reading state information... Done
omxd is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 76 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up omxd (2.0.0-1) ...
Running omxd postinst
748
Job for omxd.service failed. See 'systemctl status omxd.service' and 'journalctl -xn' for details.
dpkg: error processing package omxd (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 omxd
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

OMID-313, 2017-08-29:
New log:

```
~ $ sudo journalctl -xn

-- Logs begin at Tue 2017-08-29 15:17:01 IRDT, end at Tue 2017-08-29 17:59:37 IRDT. --
systemd[1]: Reloading.
systemd[1]: Starting omxd media player daemon...
-- Subject: Unit omxd.service has begun with start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- Unit omxd.service has begun starting up.
omxd[6511]: 6511     1     0 /usr/bin/omxd
omxd[6511]: Daemon as uid 0 already running
systemd[1]: omxd.service: control process exited, code=exited status=255
systemd[1]: Failed to start omxd media player daemon.
-- Subject: Unit omxd.service has failed
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- Unit omxd.service has failed.
--
-- The result is failed.
systemd[1]: Unit omxd.service entered failed state.
sudo[6479]: pam_unix(sudo:session): session closed for user root
sudo[6524]: zzzzz : TTY=pts/0 ; PWD=/home/zzzzz ; USER=root ; COMMAND=/bin/journal
sudo[6524]: pam_unix(sudo:session): session opened for user root by zzzzz(uid=0)
```

OMID-313, 2017-08-29:
I removed the omxd, and then tried to install it again:

```
~ $ sudo apt-get install omxd

Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  omxd
0 upgraded, 1 newly installed, 0 to remove and 76 not upgraded.
Need to get 0 B/19.2 kB of archives.
After this operation, 84.0 kB of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
  omxd
Install these packages without verification? [y/N] y
Selecting previously unselected package omxd.
(Reading database ... 33094 files and directories currently installed.)
Preparing to unpack .../omxd_2.0.0-1_armhf.deb ...
Running omxd preinst
Unknown operation 'omxd'.
Unpacking omxd (2.0.0-1) ...
Processing triggers for man-db (2.7.0.2-5) ...
Setting up omxd (2.0.0-1) ...
Running omxd postinst
748
Created symlink from /etc/systemd/system/multi-user.target.wants/omxd.service to /usr/share/doc/omxd/omxd.service.
Created symlink from /etc/systemd/system/omxd.service to /usr/share/doc/omxd/omxd.service.
Job for omxd.service failed. See 'systemctl status omxd.service' and 'journalctl -xn' for details.
dpkg: error processing package omxd (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 omxd
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

OMID-313, 2017-08-29:
Could this be related to this?
[https://github.com/systemd/systemd/issues/567]

subogero, 2017-08-29:
It could be the systemd issue! I don't have this newest Debian. And omxd definitely does use `/var/run`.

Unfortunately, fixing this involves me buying a new RPi, installing some new Debian on it and testing. My current one runs so much stuff in production at home, that OS upgrades are out of question.

Meanwhile you could still start omxd from `/etc/rc.local` like in the old times. :-)

subogero, 2017-08-29:
And thanks for the detailed info and the systemd investigation!

OMID-313, 2017-08-30:
When I run `omxd /myfolder/myfile.mp4` in the terminal, nothing happens!
No errors, no video playbacks, nothing.

subogero, 2017-08-30:
That is not a valid omxd command! You need a one-letter command before the filename.

    omxd a /myfolder/myfile.mp4

OMID-313, 2017-08-30:
Oh, sorry sorry. My mistake !!

The command `omxd a /myfolder/myfile.mp4` results in the following error:

`Can't open omxctl or /var/run/omxctl`

I checked `/var/run/` directory, but there's no `omxctl` in there.

OMID-313, 2017-08-30:
If it's difficult for you to get a new RPi, then I can give you remote access to my RPi with putty, so that you can work on it and fix the problem.
If so, please email me at omid.adljuy@gmail.com

subogero, 2017-08-30:
Both files are created by omxd, it is not running at all.

OMID-313, 2017-08-30:
So, any suggestions ? :(

subogero, 2017-08-30:
Stick with the old OS until I get a new RPi and fix it.

OMID-313, 2017-08-30:
:(((((((

I have configured the new Raspbian Jessie Lite, and installed many apps, firewalls, ...
Doing all of them again on the Minibian OS takes a lot of time :(((

Besides, the latest Minibian OS also had the Jessie and systemd. So why was it OK !?
I had installed omxd on Minibian (Jessie) without problem.

subogero, 2017-08-30:
Please check if the package at least managed to install the files:

    which omxd
    which omxwd

If they are found, just add the well known line to `/etc/rc;local`:

    omxd

When I've fixed the systemd config later, don't forget to remove it again.

subogero, 2017-08-30:
BTW, does this Raspbian Jesse Lite come without a GUI?

OMID-313, 2017-08-30:
Is seems they are found:

~ $ which omxd
/usr/bin/omxd
~ $ which omxwd
/usr/bin/omxwd

So, you mean adding `omxd A /myfolder/myfile.mp4` to the `rc.local` !?

P.S. Yes, Raspbian Jessie Lite doesn't have any GUI by default. It's better than Minibian I think. The Minibian's latest release is almost 18 months ago!! The Raspbian Jessie Lite gets frequesnt updates by the Raspberry Pi Foundation and it's great. And it has really fast boot time. etc etc. That's why I've chosen to migrate from Minibian to Raspbian Jessie Lite ;)

subogero, 2017-08-30:
No. Just `omxd`. You need to start the daemon, running in the background. That's what's missing at the moment.

`omxd A /somefile` is a client command manipulating the playlist. It actually writes the string `A /somefile` into the file `/var/run/omxctl`, which is a FIFO (named pipe). At the other end of the pipe, the background omxd daemon is listening. Or in your case, it's not listening, because systemd failed to start it on startup.

I'll move to Raspbian Lite soon, then.

OMID-313, 2017-08-30:
OK, done.

Added `omxd` to the last line of `/etc/rc.local`
After reboot, these are the logs:

```
~ $ omxd A /myfolder/myfile.mp4
Can't open omxctl or /var/run/omxctl

~ $ omxd
omxd daemon started, PID 3288

~ $ omxd A /myfolder/myfile.mp4
A /myfolder/myfile.mp4
```

and started to play the video in loop.
Also, `omxd P` stops the playback.

Thanks for the temporary solution.
Looking forward to the permanent one.
Thanks for your time and support.

subogero, 2017-08-30:
Cool, I'm glad to hear this. :D

OMID-313, 2017-08-30:
P.S. The version of the Raspbian Lite that I'm using now is the following:

```
Raspbian Jessie Lite (2016-09-23)
Debian Version: 8.0
Linux 4.4.50-v7+ armv71
```

P.S.S. It seems that the latest release of the Raspbian OS (and also Raspbian Lite) is based on the Debian Stretch (Debian 9.0). I hope things don't get messed up with that :D

OMID-313, 2017-09-02:
New news on this issue !?

subogero, 2017-09-02:
Please find below my SSH public key. If you append it to your `/root/.ssh/authorized_keys` file, it will give me SSH root access to your RPi. This is much safer than sharing passwords via email and also safer than sending the public key via email. Google could simply replace the private key with theirs and gain root access to your device. By doing this you have to trust me on the following things:

* I'll not screw up your RPi
* I and only I have access to the corresponding private key, so only I can login to your RPi

Please tell me the URL of your device. Are you sure your RPI is directly accessible on the internet? They are usually on a private subnet. You may have to set up some port forwarding on a router.

    ssh-rsa AAAAB3NzaC1yc2EAAAABIwAAAQEAu5iv9vvaQ6VdZd7vnF3UuSD/AQ1An4B303MEbRBq2p4ttKbQzr0Nc4rA+TYRVhFtB8atj4EB4w9jy+mkNknZF5+UPmsYS8w9c3ylyVgD67PODtrytNN8gKfYQ6cyX5/Iy7xJBCkBlKz2urabR+5687T57mg70+R5Q3bDQmqmfKQnUic+rDFB6Fjz8Iv/8P+HZO2P2peUsYbwwwgTrb00fGTQzzMRJeFLk6sTF6LKZdrnhjcgzS8F7vJWzlyachS2ZF/GxLb9PNcoYRn9DDgqK+YhsQR2O7rL5UWIV3/lIMNXQ31tWKU+C4ZiktQLHvIA1uYbRw5hMujKUXDMnd7voQ== szg@OG3

OMID-313, 2017-09-02:
Please check your gmail.

subogero, 2017-09-02:
I don't have gmail :-)

OMID-313, 2017-09-02:
:))

Ok, sent to your other email.

subogero, 2017-09-02:
So it turns out this bug has nothing to do with the new version of Raspbian, and everything to do with a bug in the newest version of `omxd` itself, which was installed alongside the new OS.

The check preventing the start of multiple `omxd` daemon is wrong, and it also prevents the first instance to start. It detects itself as an already running daemon.

Fix coming soon.

subogero, 2017-09-02:
Released `omxd 2.0.1`.

Please remove all your local hacks, e.g. the `omxd` line in `/etc/rc.local`, and then

    apt remove omxd
    apt update
    apt install omxd

OMID-313, 2017-09-03:
Thanks @subogero for the fix.
So far so good. It's working perfectly.

In the last step of installation, it gave some similar errors. But after rebooting RPi, it's working fine.
Thanks.

closed 2017-10-31 by subogero