subogero/omxd #75: Working with streaming

charsx, 2017-10-31:
Greetings. I'm working with a list of live IPTV (streaming) channels, I want to make a change in the behavior in the list jump to the previous streaming. With the command "echo N> / var / run / omxctl" I return to the previous streaming only when it is available, but if it is not available omxd returns to play the current channel. What I want is that if the previous channel is not available go to the previous one (before the fallen streaming).

What should I modify to achieve it?

subogero, 2017-10-31:
What do you mean by *available*?

charsx, 2017-10-31:
Thank you very much for answering so soon.
I mean that the link is operative, that can be played, that is not down.

charsx, 2017-10-31:
If something works, I'm initializing omxd with the parameter "omxd -g"

Gap mode, do not prepare next track (gap = 1)

subogero, 2017-10-31:
I was not looking for synonyms of *available*, I rather tried to understand your use case. :D

So you have streaming URLs in your playlist, that stop working? Do they stop forever, or only intermittently?

You have to know that `omxd` ain't got the faintest clue what those strings in the playlist are. They are just strings. It's just attempts to start `omxplayer` instances to play them, detects when they die to go to the next.

Nor do I intend to make `omxd` aware of what the playlist items are. As a workaround you could use

    omxd g 5

if you are playing number 7 and you know number 6 is broken. You can go directly the number 5 in the playlist.

I think if you want to do playlist cleanup, you should write a separate program for it, that might check the existence of files, and check the connectivity of URL with TCP half-open or similar. Then it could delete the garbage from the playlist with e.g.

    omxd x 6

charsx, 2017-10-31:
Actually I'm working with URLs that sometimes stop working and after a while they come back. For which I can not know which urls are operational and which are not.
In any case, could you tell me what instructions you use to know when the instance of omxplayer dies to move on to the next one?
I would be very helpful. Thanks for your time.

subogero, 2017-10-31:
These commands could come in handy:

    omxd S
    omxd S all

They print the currently played track and the status (stopped, paused, playing). The `all` argument adds the entire playlist to the status.

Or take a look at the FILES section of the README, you could monitor some of those files.

charsx, 2017-10-31:
I will review the files that you comment. If I have any questions, I hope you can guide me.

closed 2018-04-13 by subogero