subogero/omxd #76: Problem sending commands over SSH

OMID-313, 2017-11-04:
Recently, I've faced a weird problem with omxd.
When I try to send commands over SSH (like `omxd n`), it doesn't execute, and says:
```
pi@raspberrypi:~ $ omxd n
Opening FIFO 'omxctl' timed out.
The omxd daemon is dead or listening on another one.
An unprivileged omxd listens on omxctl in its current dir.
One started as root listens on '/var/run/omxctl'.
```
Typing `omxd n` in the raspberry pi itself works fine. This problem is with SSH.

Another problem is that after booting RPi, when I type `omxd A /home/pi/myvideos` it says the same thing:
pi@raspberrypi:~ $ omxd A /home/pi/myvideos
Opening FIFO 'omxctl' timed out.
The omxd daemon is dead or listening on another one.
An unprivileged omxd listens on omxctl in its current dir.
One started as root listens on '/var/run/omxctl'.
```
So, the only workaround to start omxd after boot, is the following:
```
sudo omxd
omxd X
omxd A /home/pi/myvideos
```

P.S. I'm not using any `rc.local` or systemd services to start omxd yet.

OMID-313, 2017-11-04:
The following are some logs immediately after boot and login:

```
pi@raspberrypi:~ $ sudo grep "omx" /var/log/boot.log
         Starting LSB: Raspberry Pi omxplayer playlist daemon and YouTube player...
[  OK  ] Started LSB: Raspberry Pi omxplayer playlist daemon and YouTube player.
```
and
```
pi@raspberrypi:~ $ sudo systemctl status omxd
● omxd.service - LSB: Raspberry Pi omxplayer playlist daemon and YouTube player
   Loaded: loaded (/etc/init.d/omxd)
   Active: active (exited) since Sat 2017-11-04 13:06:41 UTC; 3min 58s ago
  Process: 446 ExecStart=/etc/init.d/omxd start (code=exited, status=0/SUCCESS)

Nov 04 13:06:41 raspberrypi systemd[1]: Starting LSB: Raspberry Pi omxplayer playlist daemon and YouTube player...
Nov 04 13:06:41 raspberrypi omxd[446]: [info] Service omxd already running
Nov 04 13:06:41 raspberrypi systemd[1]: Started LSB: Raspberry Pi omxplayer playlist daemon and YouTube player.
```
and
```
pi@raspberrypi:~ $ cd /etc/
pi@raspberrypi:/etc $ sudo find | grep -r omxd
grep: gshadow: Permission denied
grep: security/opasswd: Permission denied
grep: wpa_supplicant/wpa_supplicant.conf: Permission denied
grep: .pwd.lock: Permission denied
init.d/omxd:# Provides:          omxd
init.d/omxd:omxd_pid() {
init.d/omxd:  ps -C omxd -o pid=,ppid= | while read PID PPID; do
init.d/omxd:  if [ `omxd_pid` ]; then
init.d/omxd:    echo '[info] Service omxd already running' >&2
init.d/omxd:  echo -n '[....] Starting omxd playlist daemon... ' >&2
init.d/omxd:  omxd >/dev/null
init.d/omxd:  PID=`omxd_pid`
init.d/omxd:    echo '[info] Service omxd not running' >&2
init.d/omxd:  echo -n '[....] Stopping omxd playlist daemon... ' >&2
init.d/.depend.start:TARGETS = rsyslog triggerhappy dhcpcd hostapd motd bootlogs killprocs omxd dbus piui-supervisor dnsmasq single dphys-swapfile ssh ntp bluetooth avahi-daemon nginx cron rsync plymouth rmnologin rc.local
init.d/.depend.start:plymouth: bluetooth rsyslog dbus piui-supervisor triggerhappy avahi-daemon dhcpcd dnsmasq nginx hostapd motd bootlogs cron dphys-swapfile ssh ntp rsync omxd
init.d/.depend.start:rmnologin: bluetooth rsyslog dbus piui-supervisor triggerhappy avahi-daemon dhcpcd dnsmasq nginx hostapd motd bootlogs cron dphys-swapfile ssh ntp rsync omxd
init.d/.depend.start:rc.local: bluetooth rsyslog dbus piui-supervisor triggerhappy avahi-daemon dhcpcd dnsmasq nginx hostapd motd bootlogs cron dphys-swapfile ssh ntp rsync omxd
init.d/.depend.stop:TARGETS = triggerhappy dhcpcd fake-hwclock hostapd omxd piui-supervisor bluetooth avahi-daemon nginx plymouth alsa-utils urandom dnsmasq sendsigs rsyslog umountnfs.sh rpcbind nfs-common hwclock.sh networking umountfs umountroot halt reboot
init.d/.depend.stop:umountfs: plymouth alsa-utils bluetooth piui-supervisor triggerhappy avahi-daemon urandom rpcbind hwclock.sh dhcpcd umountnfs.sh networking dnsmasq nginx hostapd omxd
grep: sudoers: Permission denied
grep: subgid-: Permission denied
grep: shadow-: Permission denied
grep: ssl/private: Permission denied
grep: passwd-: Permission denied
grep: ssh/ssh_host_dsa_key: Permission denied
grep: ssh/ssh_host_rsa_key: Permission denied
grep: ssh/ssh_host_key: Permission denied
grep: ssh/ssh_host_ecdsa_key: Permission denied
grep: ssh/ssh_host_ed25519_key: Permission denied
grep: shadow: Permission denied
grep: subuid-: Permission denied
grep: sudoers.d/010_pi-nopasswd: Permission denied
grep: sudoers.d/README: Permission denied
```
and
```
pi@raspberrypi:/etc $ ps aux | grep omxd
pi        2870  0.0  0.8   4276  2000 pts/1    S+   13:13   0:00 grep --color=auto omxd
```
and
```
pi@raspberrypi:~ $ which omxd
/usr/bin/omxd
```
and
```
pi@raspberrypi:~ $ which omxwd
/usr/bin/omxwd
```
```
pi@raspberrypi:/var/run $ ls -a
.             dbus                dnsmasq      log         ntpd.pid         shm       thd.pid     utmp
..            dhcpcd              hogsuspend   machine-id  plymouth         sshd      thd.socket
avahi-daemon  dhcpcd.pid          hostapd.pid  mount       resolvconf       sshd.pid  tmpfiles.d
crond.pid     dhcpcd.sock         initctl      network     rsyslogd.pid     sshwarn   udev
crond.reboot  dhcpcd.unpriv.sock  lock         nginx.pid   sendsigs.omit.d  systemd   user
```

subogero, 2017-11-04:
omxd is not running. Did you notice that?

subogero, 2017-11-04:
Which version are you using? 2.0.1 has a fix for exactly this problem.

subogero, 2017-11-04:
Please always report the version along with the bugs

OMID-313, 2017-11-04:
I'm using the 2.0.1 version.

OMID-313, 2017-11-04:
```
pi@raspberrypi:~ $ omxd --version
omxd 2.0.1

Free software, distributed under the GNU GPL v2 license
Copyright 2013-2017 SZABO Gergely <szg@subogero.com>
```

OMID-313, 2017-11-04:
These are different scenarios of the omxd behavior after boot and login:

```
pi@raspberrypi:~ $ omxd A /home/pi/myvideos
Opening FIFO 'omxctl' timed out.
The omxd daemon is dead or listening on another one.
An unprivileged omxd listens on omxctl in its current dir.
One started as root listens on '/var/run/omxctl'.

pi@raspberrypi:~ $ omxd
omxd daemon started, PID 1036

pi@raspberrypi:~ $ omxd X

pi@raspberrypi:~ $ omxd A /home/pi/myvideos
A /home/pi/myvideos/AA.mp4
A /home/pi/myvideos/AB.mp4
A /home/pi/myvideos/AC.mp4

Starts to play videos.
```
and
```
pi@raspberrypi:~ $ omxd
omxd daemon started, PID 1036

pi@raspberrypi:~ $ omxd A /home/pi/myvideos
A /home/pi/myvideos/AA.mp4
A /home/pi/myvideos/AB.mp4
A /home/pi/myvideos/AC.mp4

Does NOT play videos !!!
```
and
```
pi@raspberrypi:~ $ omxd
omxd daemon started, PID 1036

pi@raspberrypi:~ $ omxd X

pi@raspberrypi:~ $ omxd A /home/pi/myvideos
A /home/pi/myvideos/AA.mp4
A /home/pi/myvideos/AB.mp4
A /home/pi/myvideos/AC.mp4

Starts to play videos.
```

OMID-313, 2017-11-04:
Note that the above scenarios did not happen after each other.
I completely rebooted the RPi and tested different scenarios.

OMID-313, 2017-11-10:
So, any news on this issue !?

OMID-313, 2017-11-10:
I think the problem is with `user`.

When I run the script manually as the `pi` user, it works great; it starts to play, and I can send commands over SSH.
But when I start the systemd service manually as `root`, it starts to play, but I cannot send commands over SSH (logged in as pi user on SSH).

The following are the script and systemd services:

```
omxd
sleep 0.1
omxd X
sleep 0.1
omxd j
sleep 0.1
omxd O --blank
sleep 0.1
omxd O --no-osd
sleep 0.1
omxd A /home/pi/videos
```
and
```
[Unit]
Description=Play a playlist in loop automatically on boot

[Service]
Type=forking
ExecStart=/home/pi/scripts/init.sh

[Install]
WantedBy=multi-user.target
```

Any suggestions !?

subogero, 2017-11-10:
If you're starting omxd as a systemd service, why are you starting another one in your script?

subogero, 2017-11-10:
I also don't get why I have to debug your completely customized setup. You use a custom systemd service config. Which starts a custom script. In what way is this an issue with omxd?

OMID-313, 2017-11-28:
Ok, let's re-phrase the problem in other words:

I start the omxd daemon in a systemd service as the following:
```
omxd
sleep 0.1
omxd X
sleep 0.1
omxd j
sleep 0.1
omxd O --blank
sleep 0.1
omxd O --no-osd
sleep 0.1
omxd A /home/pi/videos
```
(If I don't put the `omxd` in the first line, it never starts to play. So I have to put it.)
On boot, the RPi starts to play the videos normally. However, I've got the following issue:

```
pi@raspberrypi:~ $ sudo systemctl status omxd
● omxd.service - LSB: Raspberry Pi omxplayer playlist daemon and YouTube player
   Loaded: loaded (/etc/init.d/omxd)
   Active: inactive (dead) since Mon 2017-11-27 15:58:19 UTC; 15h ago
  Process: 618 ExecStop=/etc/init.d/omxd stop (code=killed, signal=TERM)
  Process: 454 ExecStart=/etc/init.d/omxd start (code=exited, status=0/SUCCESS)

Nov 27 15:58:18 raspberrypi omxd[454]: [info] Service omxd already running
Nov 27 15:58:18 raspberrypi systemd[1]: Started LSB: Raspberry Pi omxplayer playlist daemon and YouTube player.
Nov 27 15:58:19 raspberrypi systemd[1]: Stopping LSB: Raspberry Pi omxplayer playlist daemon and YouTube player...
Nov 27 15:58:19 raspberrypi omxd[618]: [....] Stopping omxd playlist daemon...
Nov 27 15:58:19 raspberrypi systemd[1]: Stopped LSB: Raspberry Pi omxplayer playlist daemon and YouTube player.
```
and
```
pi@raspberrypi:~ $ omxd n
Opening FIFO 'omxctl' timed out.
The omxd daemon is dead or listening on another one.
An unprivileged omxd listens on omxctl in its current dir.
One started as root listens on '/var/run/omxctl'.
```

So, the problem is: how to start the omxd daemon properly on root, so that it receives commands!?

subogero, 2017-11-28:
The file you quote is not a systemd service file, it's a script. Where is it and what's its name?

Where is your actual systemd service file and what's in it?

A red flag: systemctl mentions `/etc/init.d/omxd` which is the old style init script. It should not be there. Either systemd or old style, but not both plesse.

OMID-313, 2017-11-28:
The systemd service file is `myboot.service` located in `/lib/systemd/system/` and its content is:
```
[Unit]
Description=Play a playlist in loop automatically on boot

[Service]
Type=forking
ExecStart=/home/pi/scripts/init.sh

[Install]
WantedBy=multi-user.target
```
As you can see, this service file executes a bash file named `init.sh` located in `/home/pi/scripts/` with the following content:
```
omxd
sleep 0.1
omxd X
sleep 0.1
omxd j
sleep 0.1
omxd O --blank
sleep 0.1
omxd O --no-osd
sleep 0.1
omxd A /home/pi/videos
```

How can I run the new systemd omxd file? I don't do anything about `/etc/init.d/omxd`
Please help on this. I'm totally confused.

OMID-313, 2017-11-28:
What should I do to avoid being `/etc/init.d/omxd` being there?
I want to use the new systemd style.

```
pi@raspberrypi:~ $ omxd --version
omxd 2.0.1

Free software, distributed under the GNU GPL v2 license
Copyright 2013-2017 SZABO Gergely <szg@subogero.com>
```

subogero, 2017-11-28:
Please delete the `/etc/init.d/omxd` file.

Then run `ps aux` and look for an `omxd` process. Check which user it's running as. If it's not root, that's the problem. `systemd` may run it as the user whom the script file belongs.

If you start `omxd` as non-root, it creates all the files in the current dir instead of the official locations. So it does not listen on the control FIFO `/var/run/omxctl`, but on the `omxctl` file in its current dir.

You could search your system for `omxctl` files. Also if you found the `omxd` process in the output of `ps aux`, you can check further process data in `/proc/<PID>`. Within this folder you will find a symlink `cwd` pointing to the current working directory of the process. Look for the `omxctl` and other files there.

OMID-313, 2017-11-28:
**_Before deleting_** `/etc/init.d/omxd` :
```
pi@raspberrypi:~ $ ps aux | grep omxd
root       805  0.0  0.4   1844  1060 ?        Ss   10:30   0:00 omxd
pi        1169  0.0  0.8   4276  1940 pts/0    S+   10:31   0:00 grep --color=auto omxd
```
The owner of `init.sh` script file is `pi`. I changed it to root, but no help.

Also, the contents of the current directory:
```
pi@raspberrypi:~ $ ls -l
total 2316
drwxrwxrwx 2 pi   pi      4096 Nov  6 22:49 imgs
drwxr-xr-x 2 root root    4096 Nov 10 13:52 normal
prw--w--w- 1 pi   pi         0 Nov 27 15:58 omxctl
-rw-r--r-- 1 pi   pi         4 Nov 27 15:57 omxd.pid
-rw-r--r-- 1 pi   pi   2305062 Nov 27 15:58 omxlog
-rw-r--r-- 1 pi   pi         0 Nov  6 17:02 omxplayer.log.12086
-rw-r--r-- 1 pi   pi         0 Nov  6 17:03 omxplayer.log.13375
-rw-r--r-- 1 pi   pi         0 Nov  6 16:52 omxplayer.log.2454
-rw-r--r-- 1 pi   pi         0 Nov  6 16:52 omxplayer.log.2485
-rw-r--r-- 1 pi   pi         0 Nov  6 16:55 omxplayer.log.2677
-rw-r--r-- 1 pi   pi         0 Nov  6 16:55 omxplayer.log.2703
-rw-r--r-- 1 pi   pi       808 Nov 10 14:16 omxplayer.log.2854
-rw-r--r-- 1 pi   pi       819 Nov 10 14:16 omxplayer.log.3067
-rw-r--r-- 1 pi   pi       957 Nov 10 12:22 omxplayer.log.3301
-rw-r--r-- 1 pi   pi       964 Nov 10 12:22 omxplayer.log.4632
-rw-r--r-- 1 pi   pi         0 Nov  6 17:57 omxplayer.log.6332
-rw-r--r-- 1 pi   pi         0 Nov  6 17:57 omxplayer.log.6349
-rw-r--r-- 1 pi   pi       823 Nov 10 15:05 omxplayer.log.6367
-rw-r--r-- 1 pi   pi       810 Nov 10 15:05 omxplayer.log.6386
-rw-r--r-- 1 pi   pi         0 Nov  6 17:58 omxplayer.log.6482
-rw-r--r-- 1 pi   pi         0 Nov  6 17:58 omxplayer.log.6512
-rw-r--r-- 1 pi   pi         0 Nov  6 17:00 omxplayer.log.6903
-rw-r--r-- 1 pi   pi        19 Nov 27 15:58 omxstat
drwxr-xr-x 2 pi   pi      4096 Nov 22 16:00 pic
drwxr-xr-x 8 root root    4096 Oct 28 13:51 piui
drwxr-xr-x 2 pi   pi      4096 Nov 10 10:50 scripts
-rwxr-xr-x 1 root root     285 Nov  4 16:40 test1.sh
drwxrwxrwx 2 root root    4096 Nov 10 13:52 videos
```

I will delete `/etc/init.d/omxd` and post the results.

subogero, 2017-11-28:
These files indicate that `omxd` may run as `pi` in its home folder.

subogero, 2017-11-28:
I think the correct location of you startup file is `/usr/local/bin` and it's name should indicate it's omxd related. And of course it should be owned by root.

OMID-313, 2017-11-28:
**_After deleting_** `/etc/init.d/omxd` :

```
pi@raspberrypi:~ $ sudo systemctl status omxd
● omxd.service
   Loaded: not-found (Reason: No such file or directory)
   Active: inactive (dead)

pi@raspberrypi:~ $ ps aux | grep omxd
root       791  0.0  0.4   1844  1060 ?        Ss   10:44   0:00 omxd
pi        1194  0.0  0.8   4276  1944 pts/0    S+   10:44   0:00 grep --color=auto omxd

pi@raspberrypi:~ $ omxd n
Opening FIFO 'omxctl' timed out.
The omxd daemon is dead or listening on another one.
An unprivileged omxd listens on omxctl in its current dir.
One started as root listens on '/var/run/omxctl'.
```

OMID-313, 2017-11-28:
```
pi@raspberrypi:/usr/local/bin $ ls -l
total 8
-rwxr-xr-x 1 root staff 250 Oct  4 13:20 calc-prorate
-rwxr-xr-x 1 root staff 216 Oct  4 13:20 cherryd
```

OMID-313, 2017-11-28:
Ok.
Is there any way to completely remove and uninstall all omxd-related files and folders, and then install omxd again ?
Will this solve the problem ?
How can I prevent the root / pi problem ?
How can I prevent the systemctl / old_system problem !?

OMID-313, 2017-11-28:
pi@raspberrypi:~ $ which omxd
/usr/bin/omxd

OMID-313, 2017-11-28:
I uninstalled omxd and omxplayer, and deleted all the omx-related files.
Then installed omxd again.

But again, it refers to `/etc/init.d/omxd` :
```
pi@raspberrypi:~ $ sudo systemctl status omxd
● omxd.service - LSB: Raspberry Pi omxplayer playlist daemon and YouTube player
   Loaded: loaded (/etc/init.d/omxd)
   Active: inactive (dead) since Tue 2017-11-28 12:02:35 UTC; 1min 2s ago
  Process: 617 ExecStop=/etc/init.d/omxd stop (code=killed, signal=TERM)
  Process: 453 ExecStart=/etc/init.d/omxd start (code=exited, status=0/SUCCESS)

Nov 28 12:02:34 raspberrypi systemd[1]: Starting LSB: Raspberry Pi omxplayer playlist daemon and YouTube player...
Nov 28 12:02:34 raspberrypi omxd[453]: [info] Service omxd already running
Nov 28 12:02:34 raspberrypi systemd[1]: Started LSB: Raspberry Pi omxplayer playlist daemon and YouTube player.
Nov 28 12:02:35 raspberrypi systemd[1]: Stopping LSB: Raspberry Pi omxplayer playlist daemon and YouTube player...
Nov 28 12:02:35 raspberrypi omxd[617]: [....] Stopping omxd playlist daemon...
Nov 28 12:02:35 raspberrypi systemd[1]: Stopped LSB: Raspberry Pi omxplayer playlist daemon and YouTube player.
```

OMID-313, 2017-11-28:
Ok, the problem is solved !!

I uninstalled omxd and omxplayer again, and then before re-installing it, I switched to root user by `super su`. Then installed omxd.

Also, after doing this, I removed the `omxd` line in the above script file, because there was no need for it.

OMID-313, 2017-11-28:
Although the problem is solved, I get the following :

```
pi@raspberrypi:~ $ sudo systemctl status omxd
● omxd.service - LSB: Raspberry Pi omxplayer playlist daemon and YouTube player
   Loaded: loaded (/etc/init.d/omxd)
   Active: inactive (dead) since Tue 2017-11-28 12:10:09 UTC; 46s ago
  Process: 616 ExecStop=/etc/init.d/omxd stop (code=killed, signal=TERM)
  Process: 449 ExecStart=/etc/init.d/omxd start (code=exited, status=0/SUCCESS)

Nov 28 12:10:08 raspberrypi systemd[1]: Starting LSB: Raspberry Pi omxplayer playlist daemon and YouTube player...
Nov 28 12:10:08 raspberrypi omxd[449]: [info] Service omxd already running
Nov 28 12:10:08 raspberrypi systemd[1]: Started LSB: Raspberry Pi omxplayer playlist daemon and YouTube player.
Nov 28 12:10:09 raspberrypi systemd[1]: Stopping LSB: Raspberry Pi omxplayer playlist daemon and YouTube player...
Nov 28 12:10:09 raspberrypi omxd[616]: [....] Stopping omxd playlist daemon...
Nov 28 12:10:09 raspberrypi systemd[1]: Stopped LSB: Raspberry Pi omxplayer playlist daemon and YouTube player.
```
and
```
pi@raspberrypi:~ $ ps aux | grep omxd
root       806  0.0  0.4   1844  1156 ?        Ss   12:10   0:00 omxd
pi        1286  0.0  0.8   4276  1944 pts/0    S+   12:11   0:00 grep --color=auto omxd
```
I don't know whether this will cause problem in the future or not.

subogero, 2017-11-28:
Another thing. The mere presence of an `omxctl` file in your current directory will screw up your omxd commands, because the client tries `./omxctl` first and then `/var/run/omxctl`.

OMID-313, 2017-11-28:
But after deleting everything, again I had problem sending commands.
Re-installing omxd after `super su` solved the problem.

OMID-313, 2017-12-03:
One more question @subogero 

Is the following normal ?
```
pi@raspberrypi:~ $  ps aux | grep omxd
root       686  0.0  0.5   1844  1172 ?        Ss   07:50   0:00 omxd
pi       10658  0.0  0.7   4276  1840 pts/0    S+   07:53   0:00 grep --color=auto omxd
```
How can I prevent being two omxd instances here ?

OMID-313, 2017-12-03:
Ok, I got it.
The second one shows up just because of running the command itself.
Sorry for such a silly question!

subogero, 2017-12-03:
It's not that silly and the solution is quite tricky too.

    ps aux | grep omxd | grep -v grep

oqibidipo, 2017-12-09:
Here's a neater trick:
```
ps aux | grep '[o]mxd'
```
`grep` still searches for 'omxd', but the command does not contain the string 'omxd'.

subogero, 2017-12-10:
I'm quite amazed that omxd now hosts a minor shell scripting meetup. :-D

closed 2017-12-30 by OMID-313