subogero/omxd #77: make install error "Failed to stop omxd.service"

harpss1ngh, 2018-01-22:
I'm following the install guide and I've run make install but I get this in the output:

./preinst
Running omxd preinst
Failed to stop omxd.service: Unit omxd.service not loaded.
Unknown operation 'omxd'.
cp omxd /usr/bin
cp: cannot create regular file ‘/usr/bin/omxd’: Text file busy
Makefile:18: recipe for target 'install' failed
make: *** [install] Error 1

Any ideas?

harpss1ngh, 2018-01-22:
Edit: Nevermind, not sure how but omxd was running. Did a killall omxd and it worked.

harpss1ngh, 2018-01-22:
I know why it doesn't work. The postinst recipe in the make install doesn't work. I think the makeinstall fails after the preinst script. The systemd service and other files aren't created and I have to manually copy them and create the symlinks

subogero, 2018-01-23:
Are you running these steps manually? All you should do is

    apt install omxd

subogero, 2018-01-23:
Next question. Are you running those comands as root?

closed 2018-04-13 by subogero