subogero/omxd #78: Feature Request: Allow network streams

harpss1ngh, 2018-01-22:
Thanks for the work you have put into this so far. I run omxplayer in a cron job with a rtsp stream from a webcam, is it possible to use omxd to play a stream?

e.g. I run something like this in a shell with omxplayer:

/usr/bin/omxplayer -b -y -w -r rtsp://@111.222.333.444 "$1"

subogero, 2018-01-23:
Anything that can be played with `omxplayer`, can be played with `omxd`. Sometimes more, especially with older versions of `omxplayer` that could not handle http cookies. `omxd` streams those via `curl` and a FIFO file to `omxplayer` and `curl` handles to cookies.

subogero, 2018-01-23:
If you must use the options `-b -y -w -r`, you can also tell `omxd` to apply them to playlist items. See command `O`.

What is `"$1"`?

closed 2018-04-13 by subogero