subogero/omxd #80: Short black frame between videos even in gapless mode

infantilo, 2018-04-13:
We're getting a few black frames between videos even in gapless mode.
Is there any way to start the second omxplayer instance before the first one dies?

infantilo, 2018-04-13:
> I prefer to call it low-gap mode, as it's never truly gapless.
>
> This is how it works:
>
>    - does not work if not in low-gap mode, or with audio over HDMI
>    - otherwise there are always 2 omxplayer instances running:
>       - current track
>       - next track paused at the very beginning
>    - omxplayer for current track exits
>    - omxd receives SIGCHLD signal, the rest is done in omxd's signal
>    handler
>    - wait for the dead child to clean up the zombie process, returns
>    immediately
>    - check if the dead child's PID is that of the current track
>    - if so send a p to the pipe to the STDIN of the omxplayer
>    representing the next track
>    - this starts the playback
>    - start omxplayer for the new next track, pause it immediately
>
> Please check if you play the audio over HDMI, in that case the low-gap
> mode is disabled, due to firmware issues.
>
> I just realized we could shorten the gap even further, by waiting for the
> dead child after starting the paused next track. But it would only bring
> micro- or milliseconds.

As i've seen, ffmpg > omxplayer > omxd  reports starttime , current
position and duration of current playing file. Woud it be possible not wait
until omxplayer died but rather use an precission timer within omxd and
start the next omx player by calculating starttime+duration? This way an
offset (+/- ms) to the calculated endtime could be used

subogero, 2018-04-15:
The length reported by `omxplayer` is not very precise. `omxd` has a watchdog the shut down stale `omxplayer` instances a few seconds after the calculated end time. In case of very long videos, sometimes the watchdog shuts them down before the end.