subogero/omxd #81: no audio problem

infantilo, 2018-04-15:
playing the same file with omxplayer directly gives sound on HDMI and USB audio.
Playing it with omxd there's no audio.
As we've tracked it down, omxd has --local by default (hardcoded?). How could we disable this option?

subogero, 2018-04-15:
Go to https://github.com/subogero/omxd

Hit Ctrl-F

Type HDMI

Hit Enter

subogero, 2018-04-15:
Alternative, if `omxd` is installed:

    man omxd
    /HDMI <Enter>
    n

infantilo, 2018-04-16:
I'm sorry for my terrible english...
you've misunderstood me.

We need to playback audio on both, HDMI and ALSA.
Oxmplayer is giving sound as wee need with:
--oalsa --hdmi
Omxplayer does not give audio output with:
--oalsa --hdmi --olocal
But the option --olocal is always sent to omxplayer from omxd.
May question was, how to disable the --olocal option?

  608 ?        Ss     0:00 omxd
 8508 ?        S      0:00 /bin/bash /usr/bin/omxplayer -I --vol 0 --no-osd -oalsa -h -olocal /mnt/rpistudio-server/URP_ORFeins_STATION_ID.mp4
 8515 ?        S      0:00 /bin/bash /usr/bin/omxplayer -I --vol 0 --no-osd -oalsa -h -olocal /mnt/rpistudio-server/WGEN_466.mp4
 8534 ?        R      0:00 /usr/bin/omxplayer.bin -I --vol 0 --no-osd -oalsa -h -olocal /mnt/rpistudio-server/URP_ORFeins_STATION_ID.mp4
 8539 ?        R      0:00 /usr/bin/omxplayer.bin -I --vol 0 --no-osd -oalsa -h -olocal /mnt/rpistudio-server/WGEN_466.mp4

subogero, 2018-04-17:
Where does the `--oalsa` come from?

infantilo, 2018-04-18:
Thats what i'm asking for?
It apperas the moment i clear the playlist with X
But i did not find anything related in omxd, so where could it come from
otherwise?
If i start omx from shell, there's no --olocal

subogero, 2018-04-18:
Sorry, it's actually 2 `omxplayer` wrapper scripts and 2 `omxplayer.bin` binaries, so that's OK.

What do you mean by `start omx from shell`?

And the most important question, again: where does `-oalsa` come from?

infantilo, 2018-04-19:
we could not track it down. 
the only option comming from us is --no-osd.
So where is -I --vol 0 -olocal comming from?
Do you have any idea???

Here is everything step by step:
`rpistudio-cl1:~ $ sudo /etc/init.d/omxd.sh start
[ ok ] Starting omxd playlist daemon... done.
pi@rpistudio-cl1:~ $
pi@rpistudio-cl1:~ $ sudo omxd A /mnt/rpistudio-server/Trailer_Im_Zentrum_1504.mp4
A /mnt/rpistudio-server/Trailer_Im_Zentrum_1504.mp4
pi@rpistudio-cl1:~ $
pi@rpistudio-cl1:~ $
pi@rpistudio-cl1:~ $ sudo ps ax |grep omx
12067 ?        Ss     0:00 omxd
12088 ?        S      0:00 /bin/bash /usr/bin/omxplayer -I --vol 0 -olocal /mnt/rpistudio-server/Trailer_Im_Zentrum_1504.mp4
12096 ?        Sl     0:00 /usr/bin/omxplayer.bin -I --vol 0 -olocal /mnt/rpistudio-server/Trailer_Im_Zentrum_1504.mp4
12108 pts/1    S+     0:00 grep --color=auto omx
`

subogero, 2018-04-19:
Please forget `-olocal` and try to answer this simple question:

# Where is `-oalsa` coming from?

closed 2018-07-25 by subogero

seeba8, 2020-07-08:
Sorry for reviving this old bug, but I had the same issue and found a workaround. By putting it here, maybe other people with the same issue can find a solution.
My problem is that I'm neither using the jack nor HDMI but have an extra audioHAT on the pi. When I start omxplayer as `omxplayer -olocal song.mp3`, I don't hear anything, the same with `omxplayer -ohdmi song.mp3`. I need to start it as `omxplayer -oalsa song.mp3`.

When I call `omxd O -oalsa`, htop shows that omxplayer is started as `omxplayer -oalsa -olocal ...` and in that case, the latter argument seems to have precedence since no sound comes.
My ugly workaround (since I unfortunately don't speak C) is to just change `olocal` to `oalsa` in `omxd.c` and compile that -- see 6ebeac1. A better solution would be to either:
1. Have a separate parameter in omxd to allow output to alsa, or
2. Check the params handed over using the `O` argument and if it is `-o.*`, don't add `-olocal`. 

If that feature could be added, that would be amazing!

Anyway, thank you for making omxd, I like it a lot :-)