subogero/omxd #82: duplicate omxplayer processes

clarkewd, 2018-08-22:
I'm not having much luck getting `omxd` to work with a playlist. If the playlist contains a single file It seems to work correctly, but adding a second file launches a second `omxplayer` processes, and causes the audio / playback to stop.

My setup is a `Raspberry Pi B+` with `Raspbian GNU/Linux 9 (stretch)`, kernel info is:

    Linux pi18.f8i.org 4.14.62-v7+ #1134 SMP Tue Aug 14 17:10:10 BST 2018 armv7l GNU/Linux

`omxd 2.0.1` installed via apt as [documented here](http://linux.subogero.com/tag/omxd/).

`omxplayer` info is:

	omxplayer - Commandline multimedia player for the Raspberry Pi
	        Build date: Wed, 05 Jul 2017 16:40:39 +0100
	        Version   : 5a25a57 [debian]
	        Repository: git@github.com:XECDesign/omxplayer.git

### Steps to reproduce

Boot system. Check `ps aux | grep -e [o]mx` and find that only the `omxd` daemon is running:

    root      1065  0.0  0.1   1844  1016 ?        Ss   16:19   0:00 /usr/bin/omxd

Next use the `echo A /mp3/1.mp3 > /var/run/omxctl` to add an mp3 file to the previously empty playlist. Checking the playlist with `cat /var/local/omxplay` shows just one file:

    /mp3/1.mp3

Everything is good at this point. Audio is playing. Checking `ps aux | grep -e [o]mx` shows `omxd` is still running and so is the `omxplayer` - looks normal:

	root      1065  0.0  0.1   1844  1016 ?        Ss   16:19   0:00 /usr/bin/omxd
	root      1862  0.0  0.2   3420  2628 ?        S    16:32   0:00 /bin/bash /usr/bin/omxplayer -I --vol 600 -olocal /mp3/1.mp3
	root      1870  6.7  2.8 170676 25284 ?        Sl   16:32   0:02 /usr/bin/omxplayer.bin -I --vol 600 -olocal /mp3/1.mp3

Trouble starts when adding a second file to the playlist using `echo A /mp3/2.mp3 > /var/run/omxctl` - I tried other variants besides `A` including `a` and `i` but all without luck. This causes the playback to stop. Checking the playlist file looks ok, `cat /var/local/omxplay` shows:

    /mp3/1.mp3
    /mp3/2.mp3

But checking `ps aux | grep -e [o]mx` shows multiple players:

	root      1065  0.0  0.1   1844  1016 ?        Ss   16:19   0:00 /usr/bin/omxd
	root      1862  0.0  0.2   3420  2628 ?        S    16:32   0:00 /bin/bash /usr/bin/omxplayer -I --vol 600 -olocal /mp3/1.mp3
	root      1870  4.6  2.8 170716 25284 ?        Sl   16:32   0:04 /usr/bin/omxplayer.bin -I --vol 600 -olocal /mp3/1.mp3
	root      1883  0.0  0.2   3420  2568 ?        S    16:33   0:00 /bin/bash /usr/bin/omxplayer -I --vol 600 -olocal /mp3/2.mp3
	root      1891  9.3  2.8 170528 25472 ?        Sl   16:33   0:01 /usr/bin/omxplayer.bin -I --vol 600 -olocal /mp3/2.mp3

I also tried setting up the playlist file before starting the `omxd` service, but sending the next command via `echo n > /var/run/omxctl` also causes a second process to launch and I lose audio.

Is this a bug or am I doing something wrong?

subogero, 2018-08-23:
Hi,

The 2 omxplayer instances are OK, it's called the low-gap playback mode of omxd. While the current track is playing, the omxplayer instance for the next one is already running, paused at the very beginning.

There is a known problem though. It does not work with sound over HDMI. Are you playing the sound over HDMI? If so, could you try switching to the Jack connector? Just to test.

Please see the `j h J H` commands in ther README.

If playing sound over the Jack is not an option for you, you can run omxd in `gap` mode. Only one `omxplayer` instance at a time. See the OPTIONS in the README, you need to start omxd with the -g option, or enable same option in the config file.

clarkewd, 2018-08-23:
Thank you! I believe I was actually already running it in jack mode. Doesn't the `-olocal` in the `ps` above confirm that?

I found that if I append the track (or tell it next track), then wait for a moment, then pause, then unpause, that works. Sort of a hack but here's a script that seems to work for "next track":

    #!/bin/sh
    echo n > /var/run/omxctl
    sleep .5
    echo p > /var/run/omxctl
    sleep .5
    echo p > /var/run/omxctl

To try gap mode I ran `service omxd stop` then edited `/usr/share/doc/omxd/omxd.service` and added the `-g`:

	[Unit]
	Description=omxd media player daemon
	After=network.target

	[Service]
	ExecStart=/usr/bin/omxd -g
	Type=forking

	[Install]
	WantedBy=multi-user.target
	Alias=omxd

Then ran `systemctl daemon-reload` and then `service omxd start` and confirmed with `ps`:

    root     21547  0.0  0.1   1844  1080 ?        Ss   13:18   0:00 /usr/bin/omxd -g

Appending and skipping tracks now works well. 

I was also having an issue with the volume being very low / quiet and getting popping and cracking / crackling, but running in `gap` mode seems to also have fixed this issue.

You mentioned it's a bug - not sure if that's with `omxplayer` or `omxd` but will leave it up to you whether to close this issue. Thanks so much.

subogero, 2018-08-23:
Yes, `-olocal` confirms you were in Jack mode.

Unfortunately it's an omxplayer / RPi firmware bug.

I'm very unhappy to hear that now it also affects `-olocal` as well. With my older Raspbian version it only affected `-ohdmi` audio. :-(

subogero, 2018-08-23:
Could you file a bug to omxplayer about this? popcornmix is aware of omxd's low-gap playback mode with 2 omxplayer instances. I think it was him who advised that audio only works with `-olocal` in case of 2 instances. Please let him know that it's broken now.

subogero, 2018-08-23:
BTW, I think instead of changing the systemd config, it's easier to create a `/etc/omxd.conf` file with this text:

    gap=1

Unfortunately, I just noticed that the config file options are not documented well in the README. Let me fix it...

clarkewd, 2018-08-24:
@popcornmix suggested adding `audio_pwm_mode=0` to the raspberry pi's `/boot/config.txt` and it seems to fix the issue with audio and the need for gap mode. Maybe worth adding a note about that to the README

closed 2018-08-24 by clarkewd