subogero/omxd #83: Cannot figure out how it is supposed to work.

ArosPrince, 2018-08-29:
Hello,
the deamon looks promising but I just cannot figure out how it is supposed to work.

First of all, the deamon is not started on startup (after reboot) as the README claims.

And then, I just cannot seem to understand how to control the playback.

Even this simple example:

I start the deamon myself:
`omxd`

Then add first file:
`omxd a ~/1.mp4`

The file starts playing... So far so good.

`omxd a ~/2.mp4`

OK, now the file should be appended after the current file.

`omxd n`

This I believe should now jump to the second file. However what really happens is that the current file (1.mp4) stops playing and closes the window and then it shows PAUSE 0:0:0 / 0:3:10 (which is the length of 1.mp4, not the next one) on the screen and nothing happens.

Do I get it wrong is there something broken?

I use RPi 3 B+ with the latest Raspbian Stretch.

subogero, 2018-08-29:
How did you install omxd? Debian package? `make install`?

ArosPrince, 2018-08-29:
As written in INSTALL:

```
git clone https://github.com/subogero/omxd.git
 cd omxd
 make
sudo make install
```

subogero, 2018-08-29:
Some ideas fo troubleshooting:

* After each `omxd` command, also do `omxd S all` as well. It prints the playlist and highlights the current track
* If you're in the `omxd` repo folder, also try `make ps`. It prints `omxd` related processes.
* Have you started the daemon as root or a simple user?

ArosPrince, 2018-08-30:
Hello thanks for the tips...

At this point is is difficult to test it because it stopped working altogether. No video is displayed at all... I am not sure what changed. omxd is running and when I do `omxd S all`, it prints the playlist and says Stopped 0/0. If I run `omxd p`, nothing changes.  Will probably have to reinstall from the scratch as I really don't see what could have broken it. I also tried normal Pi 3, it behaves the same...

As for the root / common user. I was running it as a normal unprivileged user. If I try it with sudo now, I cannot write to the FIFO using `omxd anything`.

subogero, 2018-08-30:
If you change from a normal user to root, you have to clean up the `omxctl` and other runtime files from the current folder.

Also, please check your `omxplayer`.

subogero, 2018-08-30:
Are you using the `-g` option? It's for "gap" mode, not gap-less mode.