subogero/omxd #85: Failed to start omxd.service: Unit omxd.service not found.

wuestkamp, 2018-09-15:
Hi,

when running `service omxd start` I'm getting:

Failed to start omxd.service: Unit omxd.service not found.

Any ideas?

Thanks,
Kim

wuestkamp, 2018-09-15:
Running this on PI 3 B+

When running `omxd a ~/1.mp4` I'm getting:

Can't open omxctl or /var/run/omxctl

I did install like this:

```
 git clone https://github.com/subogero/omxd.git
 cd omxd
 make
 sudo make install
```

Here is the output:

```
pi@raspberrypi:~/tmp/omxd $ sudo make install
./preinst
Running omxd preinst
Failed to stop omxd.service: Unit omxd.service not loaded.
Unknown operation omxd.
cp omxd /usr/bin
cp omxd.1 /usr/share/man/man1/
mkdir -p /usr/share/doc/omxd
cp init /usr/share/doc/omxd/
cp omxd.service /usr/share/doc/omxd/
cp logrotate /usr/share/doc/omxd/
cp omxwd /usr/bin
perl -lne 'print unless /^omxd$/' -i /etc/rc.local # Auto migrate from rc.local
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_CTYPE = "UTF-8",
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
```

dantaeyoung, 2019-04-21:
I had this problem: running `make start`, then `sudo make install` was necessary.

See: https://github.com/subogero/omxd/issues/77#issuecomment-359612081

robinhoodis, 2020-06-23:
I also ran a "make start" after running "make install" and it created the correct startup files and worked fine on a reboot.